FastLED/FastLED #1752: LED brightness

fernando74233, 2024-10-11:
I'm having problems with my project. When I turn on a part of the LED, let's say about 50 LEDs, they light up with a certain intensity, when I turn on 100 LEDs the brightness of the first 50 decreases to be the same as the other 50 that just turned on. I've already checked the power voltage, but there's no voltage drop. Or is this the library's fault?

marmilicious, 2024-10-15:
Hello @fernando74233
We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: 

https://www.reddit.com/r/FastLED

When you post there please tell us what power supply you are using and include a link to your code on pastebin.com or gist.github.com

zackees, 2024-10-30:
Please

1. Try the new 3.9.0 release

And if it's still broken, please post your sketch and the board you are using

closed 2024-11-01 by zackees