FastLED/FastLED #1754: ATTiny (1616) compiling issue

groteboem, 2024-10-12:
I am using the latest FastLED library (3.7.8) with ATTiny support. I am using it in combination with megaTinyCore.
I just have a simple blink sketch for a WS2812B led, from the sketch examples.
This is the error I get:
```
In file included from c:\Users\User\Documents\Arduino\libraries\FastLED\src/FastLED.h:81:0,
                 from C:\Users\User\AppData\Local\Temp\.arduinoIDE-unsaved2024912-7488-1kur20z.9gee\Blink\Blink.ino:3:
c:\Users\User\Documents\Arduino\libraries\FastLED\src/fastspi.h:163:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\User\AppData\Local\Temp\ccWeqksa.ltrans0.ltrans.o: In function `L_4762':
<artificial>:(.text+0xaf2): undefined reference to `timer_millis'
<artificial>:(.text+0xaf6): undefined reference to `timer_millis'
<artificial>:(.text+0xafa): undefined reference to `timer_millis'
<artificial>:(.text+0xafe): undefined reference to `timer_millis'
<artificial>:(.text+0xb0e): undefined reference to `timer_millis'
C:\Users\User\AppData\Local\Temp\ccWeqksa.ltrans0.ltrans.o:<artificial>:(.text+0xb12): more undefined references to `timer_millis' follow
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

Does anyone know how I can resolve this issue? I have made a pcb with an ATTiny1616 and WS2812 leds so I can't change the microcontroller. Also, a blank sketch does upload to the microcontroller.

pheoxs, 2024-10-15:
It's a bug in the current release AFAIK. Try adding this directly below the #include <Fastled.h>

`#include <FastLED.h>
volatile unsigned long timer_millis = 0;

void update_millis() {
    static unsigned long last_micros = 0;
    unsigned long current_micros = micros();
    if (current_micros - last_micros >= 1000) {
        timer_millis++;
        last_micros = current_micros;
    }
}`

zackees, 2024-10-29:
This should be fixed in the latest 3.9.0 release.

closed 2024-10-29 by zackees