FastLED/FastLED #1759: Using version 3.9.0 with bblanchon/ArduinoJson results in compilation errors

KrzysztofKowaczek, 2024-10-28:
Just like in the title. Tested out a lot of ArduinoJson versions, but all attempts resulted in hundreds of errors like "redefinition of class ...", "... is ambiguous" etc. (the errors change from version to version). Reverting to previous versions of FastLED solves the issue.

tueddy, 2024-10-28:
I use FastLED 3.9.0 together with ArduinoJSON 7 and can compile without problems. Just getting a lot of warnings like this:

`
.pio/libdeps/lolin_d32_pro_sdmmc_pe/FastLED/src/crgb.h:334:56: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
  334 |     constexpr explicit operator const uint32_t() const noexcept
`

I use pioarduino and these compiler flags in platform.ini:

`build_flags =
;    -DCORE_DEBUG_LEVEL=6
    -std=c++17
    -std=gnu++17
    -Wall
    -Wextra
    -Wunreachable-code

build_unflags =
    -std=gnu++11
    -Werror=all
`
I see some flickering with the new version but have to test a bit more..

KrzysztofKowaczek, 2024-10-28:
I forgot to mention I use VSCode with PlatformIO. Tested out the build flags posted by @tueddy, but still the same errors. Used ArduinoJson@7.2.0

Maybe the issue is more related with PlatformIO, which includes everything everywhere under the hood. Maybe I will be able to find a workaround.

nomakewan, 2024-10-28:
Perhaps duplicate of #1757 ?

zackees, 2024-10-29:
I just submitted a fix for the ArduinoJson stuff. Pin to master branch until the bug fix release is out.

> That seems like just begging for conflicts

It is if the library is not in it's own namespace. But this has been fixed.

>  Why does FastLED need JSON?

You'll see why in 4.0.

> And a custom internal version at that?

Because internal libraries don't get access to user modules. If FastLED needs to use ArduinoJson then we have to include our own version of it.

closed 2024-10-29 by zackees

nomakewan, 2024-10-29:
> Because internal libraries don't get access to user modules. If FastLED needs to use ArduinoJson then we have to include our own version of it.

This seems odd to me. For example, here is ElegantOTA offering the ability to access user modules ESPAsyncTCP and ESPAsyncWebServer: [ElegantOTA.h](https://github.com/ayushsharma82/ElegantOTA/blob/master/src/ElegantOTA.h)

Entirely possible I'm misunderstanding the issue, though.

zackees, 2024-10-30:
I discovered this previously when I was trying to do sdcard stuff. There's this gcc macro called __has_include(...) that can be used to test whether a certain header exists. For me I was trying to test if SD.h was installed. It just simply didn't work for the cpp files and even though the header was included by the build system FastLED.h wasn't detecting it.

It's possible only a minority of boards do this or I'm doing something wrong.

zackees, 2024-10-30:
@KrzysztofKowaczek master now has this fix in it. A release has been pushed to the ArduinoIDE servers and you should see FastLED 3.9.1 pop up in the next 12 hours or so.