FastLED/FastLED #1760: ESP32-S3 Compile Error (<avr/pgmspace.h>)

moretti20, 2024-10-29:
I am using platform io. 
platform.ini:
`[env:ureleaseEnv]
build_flags =
	-D RELEASE
	-DARDUINO_ARCH_ESP32
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_protocol = esptool
monitor_filters = 
	default
	esp32_exception_decoder
lib_deps = ...`

I am using
FastLED @ 3.9.0+sha.b57c39b
bblanchon/ArduinoJson@7.2.0

When Compiling i get  this error
".pio/libdeps/releaseEnv/FastLED/src/third_party/arduinojson/json.hpp:1189:10: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>"

nomakewan, 2024-10-29:
Looks like a buggered preprocessor definition:

https://github.com/FastLED/FastLED/blob/master/src/third_party/arduinojson/json.hpp#L1185

`avr/pgmspace.h` is Arduino-specific, so it would appear those lines belong _before_ the `#else` statement and not after it.

That said most of this code is unreadable to me and I'm not entirely sure why an addressable LED library needs JSON...

zackees, 2024-10-29:
Actually, i just needed to disable ARDUINOJSON_ENABLE_PROGMEM

zackees, 2024-10-29:
Okay. I added proper include guards if your sketch also includes ArduinoJson. I've also changed the namespace from ArduinoJson to FLArduinoJSon. This should all work now.

If someone could test the master branch and let me know if this fixes you then I'll mint a new release specifically for this fix.

zackees, 2024-10-29:
> Looks like a buggered preprocessor definition:
>
> https://github.com/FastLED/FastLED/blob/master/src/third_party/arduinojson/json.hpp#L1185
>
> `avr/pgmspace.h` is Arduino-specific, so it would appear [...]

When you see the unannounced features in 4.0 then you'll see why I'm using JSON.

Looki2000, 2024-10-29:
> Okay. I added proper include guards if your sketch also includes ArduinoJson. I've also changed the namespace from ArduinoJson to FLArduinoJSon for our internal usage.
> 
> So this should all work now.
> 
> If someone could test the master branch and let me know if this fixes you then I'll mint a new release specifically for this fix.

I tried building with master, but now it spits out very long error. Still builds fine without including ArduinoJson.

nomakewan, 2024-10-29:
> If someone could test the master branch and let me know if this fixes you then I'll mint a new release specifically for this fix.

Just tested on an ESP32 Dev Module that uses ArduinoJson 7.2.0. Seems there are still a lot of errors related to ambiguous calls causing compilation to fail. Verbose log is attached.

[fastled_main_esp32.txt](https://github.com/user-attachments/files/17563281/fastled_main_esp32.txt)

zackees, 2024-10-29:
@nomakewan - I'm looking at your error output.

Is it true that you are putting nearly all of your code into header files?

My recommendation is to put your code into cpp files and selectively include only what they need to compile. These global namespace conflicts in C++ are common when you include the-whole-world in your compile. For example, it's clear to me that my global use of the name Ptr, conflicts with ArduinoJson's use of it. That sucks. I'll have to namespace it.

nomakewan, 2024-10-29:
> @nomakewan - I'm looking at your error output.
>
> Is it true that you are putting nearly all of your code into header files?
>
> My recommendation is to put your code into cpp files and selectively [...]

It is true, and recommendation understood. I will pass this recommendation along to the maintainer.

zackees, 2024-10-30:
I've spent the last 5 hours fixing FastLED's namespace issue. It's now running in tests and the wasm build.

If you sync to master you should be able to force the namespace, thus fixing all the conflicts.

You can enable this by setting the following define.

```
FASTLED_FORCE_NAMESPACE
```

This is submitted to master and not released yet. But I'll probably issue a bug fix in the next couple of days.

zackees, 2024-10-30:
Actually, I just went ahead and renamed our Ptr<T> to Ref<T>. This will solve the ArduinoJson Ptr<T> global namespace issue.

Hopefully no other libraries are declaring Ref<T>

zackees, 2024-10-30:
@nomakewan I see that you are also breaking on the fs.h.

I just submitted a fix but that's not going in the 3.9.1. You can sync to master. I'll try to get this fix out in the next few days.

closed 2024-10-30 by zackees