FastLED/FastLED #1761: ESP32-WROOM-DA - Unexpected LED Color Behavior With New FastLED 3.9.0 Version

extremerotary, 2024-10-29:
Good afternoon,
I updated the FastLED version from 3.7.8 to 3.9.0. Now, when running some functions, I am getting rogue red LEDs illuminating in the strand and other odd/unexpected behavior. I reverted the version back to 3.7.8 and without any code changes, I recompiled and uploaded to the ESP32. I no longer have any rogue LEDs. An example function is below.

```
void emtFlashingAnimation(int numStrands, CRGB* leds[], int numLEDs[], uint8_t gHue, int brightness, int speed) {
    static uint32_t lastUpdate = 0;
    static int count = 0;
    static bool flashBlue = true; // To alternate blue/white flashing in the second stage

    // Check if enough time has passed based on speed
    if (millis() - lastUpdate < (speed * 4)) return;
    lastUpdate = millis();

    if (count < 20) {
        // Stage 1: Blue/Black Flashing Stripes
        for (int i = 0; i < numStrands; i++) {
            // Start by turning off all LEDs in the current strand
            fill_solid(leds[i], numLEDs[i], CRGB::Black);

            if (count % 2 == 0) { // Only apply blue on even counts to create flashing effect
                int start = random(0, 8); // Random starting point between 0 and 7
                
                // Apply blue and black pattern in segments of 8
                for (int j = start; j < numLEDs[i]; j += 16) { 
                    for (int k = 0; k < 8 && (j + k) < numLEDs[i]; k++) { // Use < instead of <=
                        leds[i][j + k] = CRGB::Blue;
                    }
                }
            }
        }
        FastLED.show();
        count++; // Increment the count after each update of blue/black pattern

    } else if (count < 28) {
        // Stage 2: Full-Strand Flashing in Blue or White
        CRGB color = flashBlue ? CRGB::Blue : CRGB::White; // Toggle between blue and white
        
        // Set all LEDs in all strands to the selected color
        for (int i = 0; i < numStrands; i++) {
            fill_solid(leds[i], numLEDs[i], color);
        }
        FastLED.show();

        flashBlue = !flashBlue; // Toggle color for the next flash
        count++; // Increment the count after each flash

    } else {
        // Reset count to restart the blue/black pattern
        count = 0;
    }
}
```

zackees, 2024-10-29:
Can you provide a full sketch? I will try and reproduce. I might have an ESP32-WROOM lying around to test this.

extremerotary, 2024-10-29:
Sure. Give me a few hours. I'll try to reproduce with just that function running. My code is rather extensive with WiFi & BLE running as well as a bunch of other physical PIN inputs and such. I'll try to slim it down to focus on just that and see if I can reproduce it.

extremerotary, 2024-10-30:
@zackees 
I am not able to reproduce this just using the function I provided. My code is proprietary. Is there a way to share a BIN file with you? I feel like, since you've changed the hardware use, that it is likely related to all the WiFi and BLE timers and other hardware constraints required for those parts of the code. I could share directly, but not on this forum. Let me know! For now, the older build works and I'll test the variable definition you provided tomorrow morning. I'm outta gas tonight.

zackees, 2024-10-30:
You can reach out to me on reddit as ZachVorhies.

If you give me a repro case then I'll have to do the old hack aroo to create a minimal test case. I'd rather you do that and provide a minimal test case that I can test.

extremerotary, 2024-10-30:
@zackees 
I will continue to work on a sketch that demonstrates the issue. Once I have something, I'll update this thread. 
Regarding your workaround - In my sketch, line 5, before any include statements, I have added ` #define FASTLED_RMT5 0 `
When compiling using version 3.9.0, I receive many errors listed below. If I comment out that line, it compiles normally. Do you have any other thoughts for a workaround?

```
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o:(.literal._ZN8CFastLED7addLedsI7WS2812BLh33EL6EOrder66EEER14CLEDControllerP4CRGBii[_ZN8CFastLED7addLedsI7WS2812BLh33EL6EOrder66EEER14CLEDControllerP4CRGBii]+0x14): undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o:(.literal._ZN19ClocklessControllerILi33ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EEC5Ev[_ZN19ClocklessControllerILi33ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EEC5Ev]+0x4): undefined reference to `RmtController::RmtController(int, int, int, int, int, bool)'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o:(.literal._ZN19ClocklessControllerILi33ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE[_ZN19ClocklessControllerILi33ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EE10showPixelsER15PixelControllerILS0_66ELi1ELm4294967295EE]+0x0): undefined reference to `RmtController::showPixels(PixelIterator&)'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<33, 60, 150, 90, (EOrder)66, 0, false, 5>::~ClocklessController()':
c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<26, 60, 150, 90, (EOrder)66, 0, false, 5>::~ClocklessController()':
c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<14, 60, 150, 90, (EOrder)66, 0, false, 5>::~ClocklessController()':
c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<27, 60, 150, 90, (EOrder)66, 0, false, 5>::~ClocklessController()':
c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<19, 60, 150, 90, (EOrder)66, 0, false, 5>::~ClocklessController()':
c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: undefined reference to `RmtController::~RmtController()'
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o:c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:74: more undefined references to `RmtController::~RmtController()' follow
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o: in function `ClocklessController<33, 60, 150, 90, (E [... cut ...]
c:/users/extre/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\extre\AppData\Local\Temp\arduino\sketches\EC6C484B0A38A75A1FEAD1D9F3E57D9B\sketch\car_lights_arduino.ino.cpp.o:c:\Users\extre\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/idf4_clockless_rmt_esp32.h:105: more undefined references to `RmtController::showPixels(PixelIterator&)' follow
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

zackees, 2024-10-30:
Put the define as a build setting. It won't work as just a header define.

extremerotary, 2024-10-30:
@zackees 
Adding a build setting is something I have not done before. If i'm going about it incorrectly, please correct me. 
On my PC. I navigated to AppData > Local > Arduino15 > packages > esp32 > hardware >esp32 > 3.0.4
I edited the platform.txt file at line 102 where build.extra_flags was defined. At the end of the line I added "-DFASTLED_RMT5=0".
I then recompiled the code and am having the same issue. So i'm unclear if I defined the parameter incorrectly or if it isn't having the desired effect.

zackees, 2024-10-31:
https://arduino-esp8266.readthedocs.io/en/latest/faq/a06-global-build-options.html

Will explain how to make a special file to turn on build settings.

Also, I want to stress that PlatformIO is far superior to the arduino ide. It's a real build system and VSCode is a real ide. You'll get things like auto complete, jump to symbol, and co pilot. You also define your board in the platform.ini file so you never have to install / set your board type. Once it's set once in your platformio sketch directory it will stick and not interfere with other projects that you have. Also, platformio will find your port automatically. The ArduinoIDE for some reason can't do this despite it being trivial to implement, heck I've done it.

To help users like you migrate I created an example repo showing how it's done:

https://github.com/FastLED/PlatformIO-Starter

If you were using platformio you can just use put this under build_flags and it's straight forward. ArduinoIDE requires you to insert a specially named file and maybe turn off aggressive global caching. Such a mess.

extremerotary, 2024-10-31:
@zackees 
I agree with you 10000%. I use PlatformIO for all of my other Arduino projects. This particular one was started in ArduinoIDE and when I tried to port it over to PlatformIO, I ran into some issues with the BLE libraries. To save time, I just continued to develop in the ArduinoIDE. Having run into this issue and requiring a new build option, I may just need to take the time to port it over to PlatformIO and retest. Updates to follow.

zackees, 2024-11-01:
You can consider finding the BLE library that ArduinoIDE has and just copying that into your project folder. I have done this before and it works great.

extremerotary, 2024-11-01:
@zackees 
I sincerely appreciate your helpfulness. Ill give 'er a go in PlatformIO. In the meantime, I have created a sketch with BLE and WiFi on and with arduinoFFT, and it still behaves correctly, but mine is still faulty in 3.9.0. I will likely reach out to you on reddit directly and share my code for your debugging. What you guys do is too low level and too far above my head to even begin troubleshooting.

extremerotary, 2024-11-05:
@zackees 
Alrighty, so good and bad news. 
Good news - I was able to successfully port my code over to PlatformIO. The majority of my updates required during the porting were for BLE and using standard string class instead of the built-in Arduino String class. So many areas of my code that manipulated pCharacteristic data and such needed to be updated. 
Bad news - or is it? Once I ported over to PlatformIO, even with the newest FastLED library, I cannot replicate the issue. Of course, take the same code (pre-porting of course) and upload it via ArduinoIDE, and the issue resurfaces. At this point i'm not sure what to do. Since it appears that my use-case must be something on the edge, I don't know if you want to close this issue out or if you want to try to investigate it farther. I'm open to either. Having moved over to PlatformIO without display issues, my need for any updates is relieved.

pyahmed, 2024-11-06:
Seeing a similar issue on ESP32-WROOM. Code below works as expected on 3.7.8, but on 3.9.x the following happens:
- RED
- GREEN
- GREEN + RED
- GREEN
- GREEN + RED
- etc.

```
#include "FastLED.h"

#define NUM_LEDS 1
#define DATA_PIN 23
 
CRGB leds[NUM_LEDS];
 
void setup() {
    FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical
}
 
void loop() {
  leds[0] = CRGB(64,0,0);
  FastLED.show();
  delay(1000);
  leds[0] = CRGB(0,0,0);
  FastLED.show();
  delay(1000);
}
```

notsolowki, 2024-11-08:
i have similar issue, in my project that uses ble the 3.7 version is working great, 3.9.0and up results in artifacts everywhere if i enable the wifi.

zackees, 2024-11-09:
So looking at the comments here, it appears that there is some corruption for the WS2812 led protocol on some ESP32 boards with the beta RMT5 driver. This did test fine on me, with a 22x22 matrix on an S3 no problem with two outputs. However, I will admit the second pin did not go to a second led panel, and was left unconnected. So maybe this isn't that surprising.

All I can say at this point is that if this is blocking you, please go back to RMT4 driver via FASTLED_RMT5. That driver is older but is battle tested more than this beta RMT5 driver.

I'm working on a big core feature right now and probably won't be able to get into this for a couple of days.

I'm just going to drop instructions on compiling ESP32 so that if anyone feels inclined they can drop into the code base and see what's going on with their setup. I've really structured the library to be to develop on. You basically clone and compile it with platformio. I think the S3 is already setup.

## Development

[image]

If you want to make changes to FastLED then please

  * [Fork](https://github.com/FastLED/FastLED/fork) the https://github.com/FastLED/FastLED repo into your github account.
  * Open up the folder with VSCode.
    * Make sure VSCode had the platformio extension.
  * Once FastLED is loading with platformio, give it some time to download the dependencies (esp32-s3 (default) has a 1+GB download!)
  * Click the platformio compile
    * Then upload to your device
  * See [dev/dev.ino](dev/dev.ino).
[image]

When changes are made then push to your fork to your repo and git will give you a url to trigger a pull request into the master repo.

# Info about the implementation

Basically the RMT driver stuff is the ESP32 led_strip component ported to C++ and is patched to call the async draw.

I'm working on a core feature right now, so this driver issue is medium priority for me, since there is a work around.

However, this issue will likely block the 4.0.0 release until it's fixed.

zackees, 2024-11-09:
Here's the entrypoint:

https://github.com/FastLED/FastLED/blob/master/src/platforms/esp/32/clockless_rmt_esp32.h

Abasz, 2024-12-09:
Hi I hit the same issue after updating from 3.7.3 to 3.9.*

Board firebeetle 2 the workaround of adding `-DFASTLED_RMT5=0` worked for the built in RGB led.

zackees, 2024-12-10:
Yeah, seems to be a common problem. If someone would gift me a board to test out that would be awesome. You can connect with me on reddit.com/u/ZachVorhies if anyone wants to help me out. I can probably knock out the bug in a day if I have one in my hand. I just can't reproduce it with the boards I have in my lab.

JKSniper1, 2024-12-13:
OMG I have same problem! So many hours wasted until I've found this. The latest (3.9.4) works with ESP32-C3 but not on ESP-WROOM-32. I have flickering and different color problems. I've thought I was stupid as some effects are working fine, as clean colors are woking fine and some not. When I've downgraded to 3.7.8 it works fine on both controllers. 

With  "#define FASTLED_RMT5 0" added in Arduino IDE before Fastled.h, it won't compile, giving tons of errors. Only way to fix is to go back to 3.7.8 for WROOM. 

On video you can see same code being overwritten from 3.9.4 to 3.7.8. In the fist "shooting star effect" you can see color problems and and after library downgrade, it works fine. 

https://github.com/user-attachments/assets/1b25fb81-d2b1-4ca0-b87d-2719e34a451d

zackees, 2024-12-13:
The RMT5 define is a build define, not an include define. Every single build system supports these EXCEPT arduino.

platformio supports these as first class citizens.

Feel free to browse our platformio starter kit. Just copy the files one by one onto the src directory.

It's backwards compatible with arduino, so you lose nothing.

zackees, 2024-12-15:
Okay I still can't reproduce this bug with a ESP32Dev/Wroom/S3/C3/C6.

This is the first attempt I'm making to fix it while "flying blind".

The code is located on the `fix-rmt` branch, which disables the RMT recycling.

https://github.com/FastLED/FastLED/tree/fix-rmt

You can download this here as a zip file and load it up in your arduino ide.

https://github.com/FastLED/FastLED/archive/refs/heads/fix-rmt.zip

You can also use platformio and set the FastLED dependency to the branch like this:

```
lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

Here's an example:

```
[env:dev]
platform = espressif32
framework = arduino
board = esp32dev

monitor_speed = 115200
upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

JKSniper1, 2024-12-15:
Wow thanks!  To be clear, on C3 it's working but on ESP32-WROOM-DA (HW-394) was not. I'll test out this branch the moment I can get around this strange virus detection... which is not detected on main release but it is on this branch. wtf? 
[image]

Abasz, 2024-12-16:
Thanks for this! Just for more info l: The wroom 32-E I have issues with (Dual core).

Also I use espressif Arduino 3.0.7 and IDF 5.1.4+ with pioarduino fork of platformio (https://github.com/pioarduino/platform-espressif32).

I will try to test shortly the changes.

Abasz, 2024-12-16:
I have done some testing, in which I reverted my code to espressif arduino 2.0.7 and IDF 3. In that case the issue was not present. RGB worked perfect.

When I went back to espressif arduino 3.0.7 (and also run a quick test on 3.1.0 which has been released a few hours ago) and IDF5.1 issue resurfaced. So there must have been some changes in esp arduino core 2.x to 3.x that affects RMT.

zackees, 2024-12-16:
> Wow thanks! To be clear, on C3 it's working but on ESP32-WROOM-DA (HW-394) was not. I'll test out this branch the moment I can get around this strange virus detection... which is not detected on [...]

This is very strange.

However, we do have some weird binaries in the extras/ directory and the person who knew anything about it is no longer around. Can you get through it?

[image]

zackees, 2024-12-16:
> I have done some testing, in which I reverted my code to espressif arduino 2.0.7 and IDF 3. In that case the issue was not present. RGB worked perfect.
>
> When I went back to espressif arduino 3.0.7 [...]

Yeah, they deprecated RMT 4.x and forced it on everyone with the Arduino Core 3. No warning.

Here's my favorite part of RMT5 driver: espressif forces a boot crash if you MERELY LINK IN THE 4.x DRIVER IN THE SAME BINARY AS 5.X. It doesn't matter if you actually use the driver or not.

Espressif inserted a weak symbol with RMT5 that collides with one of the RMT4 symbols. If RMT4 isn't linked in then the weak symbol stays null. If RMT4 is linked in, it goes to non null and they force a BOOT CRASH.

So keep that in mind. You'll have to upgrade sooner or later.

Abasz, 2024-12-17:
Tested the current `fix-rmt` branch you provided. The issue seems to be fixed there with0out the FASTLED_RMT5=0 flag.

zackees, 2024-12-18:
BOOM!

Okay, this is great. I'll have a fix for everyone pushed out tonight. This should be fixed for everyone tomorrow with the new release.

zackees, 2024-12-18:
Okay, bug fix is in and is getting pushed out.

Let me know if it doesn't work for you. Happy coding!

Abasz, 2024-12-18:
I am utterly confused by the situation: 

from the release notes I understand I should include `#define FASTLED_RMT5_RECYCLE=0` that should solve the issue.

But, with 3.9.5 without adding this flag led works but I get 
```ssh
E (83905) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
E (83905) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
```

If I add the define the RGB does not work but I dont get the above error message.

Abasz, 2024-12-18:
3.9.6 does not work for me regardless of including `#define FASTLED_RMT5_RECYCLE=0`

To be clear I tested adding this to .ini = '-D FASTLED_RMT5_RECYCLE=0'

And adding `#define FASTLED_RMT5_RECYCLE 0` to the first header include.

The behaviour now is the same as in 3.9.x i.e. led does not blink (it does not go black) but rather stays green continuously (there might be some slight dimming if I look carful but very subtle) .

zackees, 2024-12-18:
okay, i'm going to diff these files and investigate.

Do you have any ability to clone and compile fastled repo?

You just clone it, then load it up in vscode, and click compile, you'll need the platformio extension. There's already a sketch in there.

Abasz, 2024-12-18:
> Do you have any ability to clone and compile fastled repo?

Sure, I can compile my own project, so I suppose I can compile FastLED stand alone. 

> You just clone it, then load it up in vscode, and click compile, you'll need the platformio extension. There's already a sketch in there.

Ok, what is the aim here? Shall I try to run that sketch on the device? Which branch?

Abasz, 2024-12-18:
Using the rmt-fix branch and using the "dev" target, building and flashing gives me a lot of this error

```
E (79137) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
E (79137) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
```

I had to change the board of course and I did not change anything else (I cleaned pio packages and platforms folders so everything is fresh.)

zackees, 2024-12-18:
What happens with the newest release that just went out.

Abasz, 2024-12-19:
The reason I was asking is the following:

When I try use the latest master branch and dev target (changing the board to correct firebeetle and setting platform to `platform = https://github.com/pioarduino/platform-espressif32.git#develop`) I am not able to start compilation as I get script errors:

```
rocessing espdev (platform: https://github.com/pioarduino/platform-espressif32.git#develop; framework: arduino; board: dfrobot_firebeetle2_esp32e)
------------------------------------------------------------------------------------------------------------
JSONDecodeError: Traceback (most recent call last):
  File "/root/.platformio/penv/lib/python3.12/site-packages/requests/models.py", line 974, in json
    return complexjson.loads(self.text, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 4 (char 3)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/.platformio/penv/lib/python3.12/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/.platformio/penv/lib/python3.12/site-packages/platformio/run/cli.py", line 145, in cli
    process_env(
  File "/root/.platformio/penv/lib/python3.12/site-packages/platformio/run/cli.py", line 208, in process_env
    ).process()
```

This is my target in platformio.ini:

```ini
[env:espdev]
platform = https://github.com/pioarduino/platform-espressif32.git#develop
framework = arduino
board = dfrobot_firebeetle2_esp32e
upload_protocol = esptool

monitor_filters = 
    default
    esp32_exception_decoder

lib_deps =
    FastLED=symlink://./

build_type = debug

build_flags =
    -DDEBUG
    -g
    -Og
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE
    -DFASTLED_ESP32_SPI_BULK_TRANSFER=1
    -DENABLE_ESP32_I2S_YVES_DRIVER=1
```
If I move the platform to the latest stable (3.1.0) i.e. remove the `#develop` I can compile.

The reason I was asking about the aim is whether I should be testing your changes (and I am fine with using stable espressif arduino core 3.1.0) or rather any potential upstream changes with your code (i.e. I should use `#develop` branch for platform) which give script errors.

Abasz, 2024-12-19:
I have tested the most recent master with the following sketch:

```cpp
/// @file    Blink.ino
/// @brief   Blink the first LED of an LED strip
/// @example Blink.ino

#include <FastLED.h>

#define NUM_LEDS 1
#define DATA_PIN 5

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
  Serial.begin(115200);

    FastLED.addLeds<WS2812, DATA_PIN, GRB>(leds, NUM_LEDS);  // GRB ordering is typical
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Green;
  FastLED.show();
  delay(1000);
Serial.printf("led: %d\n", leds[0]);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(1000);
Serial.printf("led: %d\n", leds[0]);

  leds[0] = CRGB::Red;
  FastLED.show();
  delay(1000);

  leds[0] = CRGB::Black;
  FastLED.show();
  delay(1000);
Serial.printf("led: %d\n", leds[0]);

leds[0] = CRGB::Blue;
  FastLED.show();
  delay(1000);

    leds[0] = CRGB::Black;
  FastLED.show();
  delay(1000);
Serial.printf("led: %d\n", leds[0]);
}
```
The fix does not seem to be working as I cannot switch the LED off it remains green. Hence the color accuracy is also wrong. Here is a video:

https://github.com/user-attachments/assets/aa8dabcc-5f30-48c1-93e1-c2af7759448c

zackees, 2024-12-20:
Okay, 3.9.6 had a mistake in it where the RMT recycle value was inverted. I've fixed it in 3.9.7 which will end up in the Arduino IDE repo shortly. Feel free to install it now from the zip file to try it immediatly.

Abasz, 2024-12-20:
I think your code still not correct in v3.9.7:

You inverted the if..else and for mRecycle == false you run no_recycle. This is ok. 

```cpp
        if (mRecycle) {
            mLedStrip = create_rmt_led_strip(t0h, t0l, t1h, t1l, TRESET, mPin, pixels.size(), is_rgbw);
        } else {
            mLedStrip = create_rmt_led_strip_no_recyle(t0h, t0l, t1h, t1l, TRESET, mPin, pixels.size(), is_rgbw);
        }
```

But in the release note as well as in the comment in the code you state that the old behaviour comes with setting the define to `1` which is true. Also you initialise recycel to false if not defined, so you are running create_rmt_led_strip_no_recyle. But in the release notes you say: `To get the old behavior back use #define FASTLED_RMT5_RECYCLE=1.`

```cpp
// It turns out that RMT5 recycling causes a lot of problems with
// the first led. A bug has been filed with Espressif about this.
// Therefore we use the alternative implementation that does not
// reycle. To get the old behavior, set FASTLED_RMT5_RECYCLE to 1.
#ifndef FASTLED_RMT5_RECYCLE
#define FASTLED_RMT5_RECYCLE 0
#endif
```

It is confusing what you call old behaviour, especially when you state: `The new behavior may become the default if it turns out this is more stable.` 

But in your code the default is now no_recycle I think. Is this what your intention was?

From my perspective this is good  because this means I dont need to add any flag to potentially have a working LED :) (I will test once new release is on platformio repo as that is much easier for me if that is ok with you.)

zackees, 2024-12-20:
> But in your code the default is now no_recycle I think. Is this what your intention was?

Yes.

zackees, 2024-12-21:
Okay, ArduinoIDE now has FastLED 3.9.7, so closing this bug. Feel free to re-open if not fixed.

Abasz, 2024-12-21:
I confirm 3.9.7 works with the defaults (i.e. no recycling).

Abasz, 2024-12-21:
I was probably too quick. The current default (i.e. no recycling) works on the Firebeetle2 without any issues. 

But on the esp32-S3 (loling-s3-mini board) there is an error every time the FastLED.show() is called (though the RBG works):

```
E (79137) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
E (79137) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
```

With `FASTLED_RMT5_RECYCLE=1` everything works.

zackees, 2024-12-22:
how many strips?

zackees, 2024-12-23:
Bad news: Disabling RMT recycle broke our multi strip case on ESP32.

Good news: Espressif got back to me and said that this bug is a manifestation of the gpio state being reset after the rmt disable.

See: https://github.com/espressif/esp-idf/issues/15049

They recommended that I set the INPUT_PULLDOWN on the pin right after the rmt_disable call, to prevent the floating HIGH state on the pin. I've added this and also a micro delay of 10 microseconds to make sure that the RMT blip, if it's still happening, will now only be momentary and also the 10 microsecond delay will prevent the now shortened blip from being interpreted as a 1 or a 0.

## What I'm going to do:

  * I'm applying the espressif fix
  * I'm re-enabling RMT recycling for the next release.

## What I need from you

If nothing is done, the rmt recycling will be re-enabled on the next release. I *hope* this won't break you, but there's a small possibility that you'll get a regression back to the green led being stuck on.

However, if someone tests the updated branch rmt-fix https://github.com/FastLED/FastLED/tree/fix-rmt and let me know that it continues to work for you, we can avoid this inconvenience for you. This branch has both the espressif fix and also RMT recycling re-enabled.

If...
  * The green light is stuck on: then I will let espressif know and revert the change.
  * If the green light continues to be fixed: Then a release will be minted within 24 hours.
  * If no feedback is given, then the release will be minted by the 28th, and the chips will fall where they lay. If this breaks you then you will have to apply the workaround until the next release.
  
I'm flying blind here since I can't reproduce your issue locally. So thanks in advance, for your help.

Abasz, 2024-12-23:
> how many strips?

On single led. This is the onboard led.

I will do testing tonight:)

> I've added this and also a micro delay of 10 microseconds to make sure that the RMT blip, if it's still happening, will now only be momentary and also the 10 microsecond delay will prevent the now shortened blip from being interpreted as a 1 or a 0

Does this mean that there is a 10us delay on every show() call?

zackees, 2024-12-23:
Correct, but I'll add that there is a 10us delay *per strip*. If you have 4 strips the total delay will be 40uS.

If you are willing to continue helping me then we can try and bring that down to 0.

Abasz, 2024-12-23:
Sure, if you link the related code part, I can try changing it and
experiment what works and what not.

Abasz, 2024-12-23:
I see that you have change the default `FASTLED_RMT5_RECYCLE` to 1. So this now works on the S3, of course nothing new here. This is regardless of the wait time.

On the firebeetle2 defaults do not work. I tried to increase the delay `FASTLED_RMT5_EXTRA_WAIT_MICROS` to 100us and it does not make a difference.

The firebeetle2 only does proper blinking with `FASTLED_RMT5_RECYCLE=0` but then it give the below error:
```
E (14810) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
E (14810) led_strip_rmt: led_strip_rmt_wait_refresh_done(86): disable RMT channel failed
```

This is somewhat strange becuase 3.9.7 did not give this error with the default (where the default was FASTLED_RMT5_RECYCLE=0)

The error pops up on the S3 with `FASTLED_RMT5_RECYCLE=0`. But there is no change in this respect.

So for me the espressif fix does not seem to work.

zackees, 2024-12-23:
Thank you, looks like I have to continue going down this no recycle path :(

wouterf11, 2024-12-24:
To avoid printing the error message:
```
E (14810) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
E (14810) led_strip_rmt: led_strip_rmt_wait_refresh_done(86): disable RMT channel failed
```
perhaps the return from rmt_disable can be checked for ESP_ERR_INVALID_STATE without throwing the error (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html#_CPPv411rmt_disable20rmt_channel_handle_t). 
I could not find a method to get the rmt channel state, which could have been used to avoid calling `led_strip_rmt_wait_refresh_done` when the rmt channel has already been disabled.

zackees, 2024-12-24:
espressif is asking if there is an external pullup resistor. Can you share your schematic or take a picture?

Abasz, 2024-12-24:
Added to that thread directly

zackees, 2024-12-24:
Thanks.

Do you buy chance have a platformio.ini you can share?

Abasz, 2024-12-24:
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html
[platformio]
default_envs =  generic, kayakfirst, kayakfirst-debug, concept2-debug, concept2, kayakfirstblue, kayakfirstblue-s3, kayakfirstblue-debug

[env:generic]
monitor_speed = 115200
platform = https://github.com/pioarduino/platform-espressif32.git#53.03.10
board = dfrobot_firebeetle2_esp32e
board_build.filesystem = littlefs
framework = arduino
monitor_filters = esp32_exception_decoder
lib_deps = 
	h2zero/NimBLE-Arduino@2.1.2
	thijse/ArduinoLog@1.1.1
	fastled/FastLED@3.9.7
	greiman/SdFat@2.2.3
build_flags = 
	-O1
	'-D CONFIG_BT_NIMBLE_PINNED_TO_CORE=0'
	'-D CONFIG_BT_NIMBLE_ROLE_CENTRAL_DISABLED'
	'-D CONFIG_BT_NIMBLE_ROLE_OBSERVER_DISABLED'
build_unflags = 
	-ggdb
	-Os
	-std=gnu++17

[env:custom]
extends = env:generic
build_flags = 
 ${env:generic.build_flags}
 '-D USE_CUSTOM_SETTINGS'

[env:kayakfirst]
extends = env:custom
monitor_speed = 1500000
build_flags =
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/devkit.board-profile.h"'
 '-D ROWER_PROFILE="profiles/kayakfirst.rower-profile.h"'

[env:kayakfirstblue]
extends = env:custom
monitor_speed = 1500000
board = esp32doit-devkit-v1
build_flags =
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/devkit.board-profile.h"'
 '-D ROWER_PROFILE="profiles/kayakfirstBlue.rower-profile.h"'

[env:kayakfirstblue-s3]
extends = env:custom
board = lolin_s3_mini
monitor_speed = 1500000
build_flags =
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/lolinS3-mini.board-profile.h"'
 '-D ROWER_PROFILE="profiles/kayakfirstBlue.rower-profile.h"'
 '-D ARDUINO_USB_CDC_ON_BOOT=1'
 '-D FASTLED_RMT5_RECYCLE=1'

[env:concept2]
extends = env:custom
monitor_speed = 1500000
build_flags = 
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/firebeetle.board-profile.h"'
 '-D ROWER_PROFILE="profiles/generic.rower-profile.h"'

[env:concept2-s3]
extends = env:custom
board = lolin_s3_mini
monitor_speed = 1500000
build_flags = 
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/lolinS3-mini.board-profile.h"'
 '-D ROWER_PROFILE="profiles/generic.rower-profile.h"'
 '-D ARDUINO_USB_CDC_ON_BOOT=1'
 '-D FASTLED_RMT5_RECYCLE=1'

[env:concept2-devkit]
extends = env:custom
board = esp32doit-devkit-v1
monitor_speed = 1500000
build_flags = 
 ${env:custom.build_flags}
 '-D BOARD_PROFILE="profiles/devkit.board-profile.h"'
 '-D ROWER_PROFILE="profiles/generic.rower-profile.h"'

[env:kayakfirst-debug]
extends = env:kayakfirst
build_type = debug

[env:kayakfirstblue-debug]
extends = env:kayakfirstblue-s3
build_type = debug

[env:concept2-debug]
extends = env:concept2
build_type = debug
```

Dont mind the specific version of fast led I change for the testing of course. Same for the no recycling flag.

zackees, 2024-12-24:
That's this will allow me to get the defines

zackees, 2024-12-29:
Should be fixed in the 3.9.8 release. Let me know if there are problems.

closed 2024-12-29 by zackees