FastLED/FastLED #1763: Including ESPAsyncWebServer.h causes compilation errors

Looki2000, 2024-10-30:
New 3.9.1 fixes problem with ArduinoJson, but I'm still having problem with ESP Async WebServer. I'm not sure if it is related to this.

Hardware
ESP32-WROOM-32D

Libraries
- FastLED @ 3.9.1
- ESP Async WebServer @ 1.2.4

[FastLED ESPAsyncWebServer.txt](https://github.com/user-attachments/files/17572498/FastLED.ESPAsyncWebServer.txt)

notsolowki, 2024-10-31:
I have the same issue, 3.9.2 causes compilation errors when including ESPAsyncWebServer. downgrading to 3.9.1 and the issue goes away.

zackees, 2024-10-31:
> I have the same issue, 3.9.2 causes compilation errors when including ESPAsyncWebServer. downgrading to 3.9.1 and the issue goes away.

I think you have that wrong. 3.9.1 has the issue with compilation. 3.9.2 which hasn't been released yet will fix it.

notsolowki, 2024-10-31:
> > I have the same issue, 3.9.2 causes compilation errors when including ESPAsyncWebServer. downgrading to 3.9.1 and the issue goes away.
> 
> I think you have that wrong. 3.9.1 has the issue with compilation. 3.9.2 which hasn't been released yet will fix it.
> 
> The fix will be officially released later today.

indeed 3.9.1, sorry i got the versions corrupted in my head dealing with all the library upgrades/downgrades.   When i downgraded from 3.9.1 to 3.9.0 i had arduino json issues. now im using fastled 3.7.8 and i have no compiler warnings.

zackees, 2024-11-01:
3.9.2 with these fixes is now released to the public. I created that aggregate test and it works great. I'll hook it into the test system so we don't have this problem again.

closed 2024-11-01 by zackees