FastLED/FastLED #1767: DMA/RMT issue on ESP32 66

gmint, 2024-11-04:
Although I can get FasLed 3.9.2. (and prior) to build, I'm getting a crash on my ESP32C6 with the following messages. From my understanding there should be support for the C6, but I've looked for a very long time and I'm unable to determine if there's something else I can do to make this work, or if this just isn't working correctly. My apologies in advance if I'm being an idiot here, I've tried hard not to be...

22:16:05.565 -> E (11510) rmt: rmt_new_tx_channel(225): DMA not supported
22:16:05.597 -> E (11511) led_strip_rmt: led_strip_new_rmt_device_with_buffer(259): create RMT channel failed
22:16:05.597 -> E (11514) rtm_strip.cpp: construct_led_strip failed because of unexpected error, is DMA not supported on this device?: ESP_ERR_NOT_SUPPORTED
22:16:05.597 -> ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x420058ec
22:16:05.597 -> file: "/Users/qualleyiv/Arduino/libraries/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp" line 85
22:16:05.631 -> func: void fastled_rmt51_strip::RmtLedStrip::acquire_rmt()
22:16:05.631 -> expression: err
22:16:05.631 -> 
22:16:05.662 -> abort() was called at PC 0x40807c37 on core 0

zackees, 2024-11-06:
Fix is in master

tikiss, 2024-11-13:
I am also using ESP32 C6 zero from Waveshare:
[https://www.waveshare.com/wiki/ESP32-C6-Zero](https://www.waveshare.com/wiki/ESP32-C6-Zero)

Experiencing the same problem, I downloaded the master branch, but I cannot see any difference, it is still crashing:

```
E (161) rmt: rmt_new_tx_channel(225): DMA not supported
E (161) led_strip_rmt: led_strip_new_rmt_device_with_buffer(259): create RMT channel failed
E (163) rtm_strip.cpp: construct_led_strip failed because of unexpected error, is DMA not supported on this device?: ESP_ERR_NOT_SUPPORTED
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x42001efc
file: "/Users/tibi/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp" line 85
func: void fastled_rmt51_strip::RmtLedStrip::acquire_rmt()
expression: err
```

I am uisng Arduino IDE 2.3.3.

zackees, 2024-11-13:
Double check your install of master branch. If it doesn't work then send me your board info. I'm using the esp32 defines so there's no reason it shouldn't work. I'll mint a new release today.

Fyi i have a c6 board and it works great after the fix.

gmint, 2024-11-13:
I can confirm that the fix in Master worked for my C6.

zackees, 2024-11-14:
3.9.3 with a fix for this issue has been pushed to Arduino and will be released at any time. Closing this issue.

closed 2024-11-14 by zackees

tikiss, 2024-11-14:
I installed it again, and it seems to work now, thank you!