FastLED/FastLED #1768: RMT5 issue ESP32-S3

freezak, 2024-11-08:
For some reason I can't get RMT5 driver work on ESP32-S3. Tried with different FastLED versions (3.9.0, 3.9.1, 3.9.2 and master), tried another ESP32-S3 board (YD-ESP32-23 and Waveshare ESP32-S3 Pico). Problem always the same. On the ESP32-WROOM-32  based board (normal ESP32) RMT 5 works, but I have many glitches, and it isn't a hardware problem, because when I disabled RMT it works normally on the same hardware without any glitches, but it is too slow...

**Serial output:**
```
E (9) rmt: rmt_tx_register_to_group(132): no free tx channels
E (10) rmt: rmt_new_tx_channel(239): register channel failed
FPS: 40
FPS: 40
FPS: 40
```

**main.cpp:**
```c++
#include <Arduino.h>
#include <FastLED.h>
#include "matrix.h"

#define NUM_LEDS 1536
CRGBArray<NUM_LEDS> leds;

void setup() {
  Serial.begin(115200);

  FastLED.addLeds<WS2812B, 15, GRB>(leds, MATRIX_FRAGMENT_SIZE*5, MATRIX_FRAGMENT_SIZE);
  FastLED.addLeds<WS2812B, 16, GRB>(leds, MATRIX_FRAGMENT_SIZE*2, MATRIX_FRAGMENT_SIZE);
  FastLED.addLeds<WS2812B, 17, GRB>(leds, MATRIX_FRAGMENT_SIZE*4, MATRIX_FRAGMENT_SIZE);
  FastLED.addLeds<WS2812B, 18, GRB>(leds, MATRIX_FRAGMENT_SIZE*1, MATRIX_FRAGMENT_SIZE);
  FastLED.addLeds<WS2812B, 33, GRB>(leds, MATRIX_FRAGMENT_SIZE*3, MATRIX_FRAGMENT_SIZE);
  FastLED.addLeds<WS2812B, 34, GRB>(leds, MATRIX_FRAGMENT_SIZE*0, MATRIX_FRAGMENT_SIZE);
  FastLED.setBrightness(64);
}

uint64_t prev_millis = 0;
uint64_t current_millis = 0;

void loop() {
  static uint8_t pos = 0;
  pos++;
  if (pos == MATRIX_SIZE_Y) pos == 0;
  for (uint8_t y = 0; y < MATRIX_SIZE_Y; y++) {
    for (uint8_t x = 0; x < MATRIX_SIZE_X; x++) {
      leds[XY(x, y)] = CHSV((float)(x+y) / (float)(MATRIX_SIZE_X+MATRIX_SIZE_Y) * 255.0f + pos, 255, 255);
    }
  }
  FastLED.show();

  FastLED.countFPS(10);

  current_millis = millis();
  if (current_millis > prev_millis + 1000) {
    prev_millis = current_millis;
    Serial.print("FPS: ");
    Serial.println(FastLED.getFPS());
  }
}
```

**platformio.ini:**
```ini
[env:esp32-s3]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = esp32-s3-devkitc-1
framework = arduino
monitor_filters =
    default
    esp32_exception_decoder
build_flags = -D FASTLED_RMT_BUILTIN_DRIVER=true
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
lib_deps = 
    https://github.com/FastLED/FastLED#master
monitor_speed = 115200
monitor_port = COM8
```

**Screenshot that shows that I have arduino code with IDF > 5.0.0 (then it has rmt5):**
[image]

zackees, 2024-11-09:
Wow. I got to say. Dude, you are like some wizard that is actually pushing this to the limit.

Fucking bravo.

Let me think about this. I do want to share with you instructions on how to dive in yourself, if you feel so inclined:

## Development

[image]

If you want to make changes to FastLED then please

  * [Fork](https://github.com/FastLED/FastLED/fork) the https://github.com/FastLED/FastLED repo into your github account.
  * Open up the folder with VSCode.
    * Make sure VSCode had the platformio extension.
  * Once FastLED is loading with platformio, give it some time to download the dependencies (esp32-s3 (default) has a 1+GB download!)
  * Click the platformio compile
    * Then upload to your device
  * See [dev/dev.ino](dev/dev.ino).
[image]

When changes are made then push to your fork to your repo and git will give you a url to trigger a pull request into the master repo.

ledereyes, 2024-11-15:
I am experiencing the same issue and error message on an ESP32 S3 DevKit C1 (by WaveShare).
Tried most recent FastLED versions with and without -D FASTLED_RMT_BUILTIN_DRIVER but seems to make no difference.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x109c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb50
load:0x403cc700,len:0x2fe4
entry 0x403c98ac
E (68) rmt: rmt_tx_register_to_group(132): no free tx channels
E (69) rmt: rmt_new_tx_channel(239): register channel failed
```

With 6 strips and total 420 leds i am getting just 50 fps while the same sketch on an RP2040 runs at 400 fps even though its clocked less. Could the non-working RMT be the reason for this? Are there any workarounds yet?

Thanks!

samguyer, 2024-11-15:
I found some information on the ESP-IDF discussion forum. Apparently, this is not unique to fastLED:

https://www.esp32.com/viewtopic.php?t=42301

freezak, 2024-11-15:
@ledereyes did you found a solution? There is the part of code with rmt channel creation.
[image]

ledereyes, 2024-11-15:
@freezak thanks for pointing to the code, I tried to take the config which is mentioned in the thread that @samguyer linked, but no success yet.

Could it be that one strip already occupies 4 RMT channels, so with 2 or more strips we are already over the limit of its 6 channels?

Is there a way to run strips in some "classic" mode without RMT on the S3? To me it seems that `-D FASTLED_RMT_BUILTIN_DRIVER=true/false` makes no difference at all.

freezak, 2024-11-15:
> @freezak thanks for pointing to the code, I tried to take the config which is mentioned in the thread that @samguyer linked, but no success yet.For me the same, no succes..
> Could it be that one strip already occupies 4 RMT channels, so with 2 or more strips we are already over the limit of its 6 channels?
Idk.. In some places I can see information that on ESP32-S3 is 8 channels, in others that there are only 4 channels.. More I dig, the more confusion I get about it.. Btw I can't imagine witch way one strip can use 4 channels. I need to read more about RMT in ESP-IDF.
> Is there a way to run strips in some "classic" mode without RMT on the S3? To me it seems that `-D FASTLED_RMT_BUILTIN_DRIVER=true/false` makes no difference at all.
As I see you can run only HSPI, I2S and RMT on ESP32 series. But for me when I switched to I2S driver it won't compile.
[image]

freezak, 2024-11-15:
`@freezak thanks for pointing to the code, I tried to take the config which is mentioned in the thread that @samguyer linked, but no success yet.For me the same, no succes..`
Same for me, no success yet.

`Could it be that one strip already occupies 4 RMT channels, so with 2 or more strips we are already over the limit of its 6 channels?`
Idk.. In some places I can see information that on ESP32-S3 is 8 channels, in others that there are only 4 channels.. More I dig, the more confusion I get about it.. Btw I can't imagine witch way one strip can use 4 channels. I need to read more about RMT in ESP-IDF.

`Is there a way to run strips in some "classic" mode without RMT on the S3? To me it seems that -D FASTLED_RMT_BUILTIN_DRIVER=true/false makes no difference at all.`
As I see you can run only HSPI, I2S and RMT on ESP32 series. But for me when I switched to I2S driver it won't compile.
[image]

freezak, 2024-11-15:
Datasheet says:
[image]
But RMT docs says:
[image]

ledereyes, 2024-11-16:
Someone on reddit just got ESP32 S3 working with some custom I2S driver:
https://www.reddit.com/r/FastLED/comments/1gslqo4/latest_project_macropad_with_8x8_panels_in_the/

zackees, 2024-11-16:
> @freezak thanks for pointing to the code, I tried to take the config which is mentioned in the thread that @samguyer linked, but no success yet.
>
>
>
> Could it be that one strip already occupies 4 RMT [...]

RMT built in driver is not implemented in our RMT 5 port. That's why it does nothing.

zackees, 2024-11-16:
> Someone on reddit just got ESP32 S3 working with some custom I2S driver:
> 
> https://www.reddit.com/r/FastLED/comments/1gslqo4/latest_project_macropad_with_8x8_panels_in_the/ 

Yeah I2C driver works. The issue is that it's not flexible. However it is very very robust against network interrupts and can drive upto strips through multiplexing.

Everytime i try to implement this I get discouraged because the entire setup requires specifying everything up front with the led blocks and that doesn't map well to our driver system which assumes everything is independent.

There is another driver option which is SPI. The led_strip lib from espressif has this but i stripped it out to make the port to C++ work with async draw, which the original library did not have.

There is only one main contributor right now. Any help here would be greatly appreciated.

zackees, 2024-11-16:
> For some reason I can't get RMT5 driver work on ESP32-S3. Tried with different FastLED versions (3.9.0, 3.9.1, 3.9.2 and master), tried another ESP32-S3 board (YD-ESP32-23 and Waveshare ESP32-S3 Pico). Problem always the same. On the ESP32-WROOM-32 based board (normal ESP32) RMT 5 works, but I have many glitches, and it isn't a hardware problem, because when I disabled RMT it works normally on the same hardware without any glitches, but it is too slow...
> 
> **Serial output:**
> 
> ```
> E (9) rmt: rmt_tx_register_to_group(132): no free tx channels
> E (10) rmt: rmt_new_tx_channel(239): register channel failed
> FPS: 40
> FPS: 40
> FPS: 40
> ```
> 
> **main.cpp:**
> 
> ```c++
> #include <Arduino.h>
> #include <FastLED.h>
> #include "matrix.h"
> 
> #define NUM_LEDS 1536
> CRGBArray<NUM_LEDS> leds;
> 
> void setup() {
>   Serial.begin(115200);
> 
>   FastLED.addLeds<WS2812B, 15, GRB>(leds, MATRIX_FRAGMENT_SIZE*5, MATRIX_FRAGMENT_SIZE);
>   FastLED.addLeds<WS2812B, 16, GRB>(leds, MATRIX_FRAGMENT_SIZE*2, MATRIX_FRAGMENT_SIZE);
>   FastLED.addLeds<WS2812B, 17, GRB>(leds, MATRIX_FRAGMENT_SIZE*4, MATRIX_FRAGMENT_SIZE);
>   FastLED.addLeds<WS2812B, 18, GRB>(leds, MATRIX_FRAGMENT_SIZE*1, MATRIX_FRAGMENT_SIZE);
>   FastLED.addLeds<WS2812B, 33, GRB>(leds, MATRIX_FRAGMENT_SIZE*3, MATRIX_FRAGMENT_SIZE);
>   FastLED.addLeds<WS2812B, 34, GRB>(leds, MATRIX_FRAGMENT_SIZE*0, MATRIX_FRAGMENT_SIZE);
>   FastLED.setBrightness(64);
> }
> 
> uint64_t prev_millis = 0;
> uint64_t current_millis = 0;
> 
> void loop() {
>   static uint8_t pos = 0;
>   pos++;
>   if (pos == MATRIX_SIZE_Y) pos == 0;
>   for (uint8_t y = 0; y < MATRIX_SIZE_Y; y++) {
>     for (uint8_t x = 0; x < MATRIX_SIZE_X; x++) {
>       leds[XY(x, y)] = CHSV((float)(x+y) / (float)(MATRIX_SIZE_X+MATRIX_SIZE_Y) * 255.0f + pos, 255, 255);
>     }
>   }
>   FastLED.show();
> 
>   current_millis = millis();
>   if (current_millis > prev_millis + 1000) {
>     prev_millis = current_millis;
>     Serial.print("FPS: ");
>     Serial.println(FastLED.getFPS());
>   }
> }
> ```
> 
> **platformio.ini:**
> 
> ```ini
> [env:esp32-s3]
> platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
> board = esp32-s3-devkitc-1
> framework = arduino
> monitor_filters =
>     default
>     esp32_exception_decoder
> build_flags = -D FASTLED_RMT_BUILTIN_DRIVER=true
> board_build.mcu = esp32s3
> board_build.f_cpu = 240000000L
[... 85 lines cut ...]

board_build.mcu = esp32s3
board_build.f_cpu = 240000000L

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

; Symlink in the FastLED library so that changes to the library are reflected in the project
; build immediatly.
lib_deps =
  FastLED=symlink://./

build_type = debug

build_flags =
	-DDEBUG
    -g
    -Og
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE

    
check_tool = clangtidy
```

ledereyes, 2024-11-17:
Thanks for trying this out @zackees . Well in my case its a WaveShare ESP32-S3-DEV-KIT-N8R8 and with the above sketch I get the above mentioned RMT error and also around 50fps and non working LEDs on the strips. If I remove all but one strip the error disappears and the LEDs on the strip work fine at 350fps. 
With two or more strips the LEDs in most cases do not react at all. Sometimes they weirdly blink after a few seconds or so.

The same sketch and setup with 6 strips on an RP2040 work fine. But the RP2040 has a few other problems by itself (not FastLED related) that's why I would prefer to use the ESP32S3. ;)

freezak, 2024-11-18:
> > For some reason I can't get RMT5 driver work on ESP32-S3. Tried with different FastLED versions (3.9.0, 3.9.1, 3.9.2 and master), tried another ESP32-S3 board (YD-ESP32-23 and Waveshare ESP32-S3 Pico). Problem always the same. On the ESP32-WROOM-32 based board (normal ESP32) RMT 5 works, but I have many glitches, and it isn't a hardware problem, because when I disabled RMT it works normally on the same hardware without any glitches, but it is too slow...
> > **Serial output:**
> > ```
> > E (9) rmt: rmt_tx_register_to_group(132): no free tx channels
> > E (10) rmt: rmt_new_tx_channel(239): register channel failed
> > FPS: 40
> > FPS: 40
> > FPS: 40
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > **main.cpp:**
> > ```c++
> > #include <Arduino.h>
> > #include <FastLED.h>
> > #include "matrix.h"
> > 
> > #define NUM_LEDS 1536
> > CRGBArray<NUM_LEDS> leds;
> > 
> > void setup() {
> >   Serial.begin(115200);
> > 
> >   FastLED.addLeds<WS2812B, 15, GRB>(leds, MATRIX_FRAGMENT_SIZE*5, MATRIX_FRAGMENT_SIZE);
> >   FastLED.addLeds<WS2812B, 16, GRB>(leds, MATRIX_FRAGMENT_SIZE*2, MATRIX_FRAGMENT_SIZE);
> >   FastLED.addLeds<WS2812B, 17, GRB>(leds, MATRIX_FRAGMENT_SIZE*4, MATRIX_FRAGMENT_SIZE);
> >   FastLED.addLeds<WS2812B, 18, GRB>(leds, MATRIX_FRAGMENT_SIZE*1, MATRIX_FRAGMENT_SIZE);
> >   FastLED.addLeds<WS2812B, 33, GRB>(leds, MATRIX_FRAGMENT_SIZE*3, MATRIX_FRAGMENT_SIZE);
> >   FastLED.addLeds<WS2812B, 34, GRB>(leds, MATRIX_FRAGMENT_SIZE*0, MATRIX_FRAGMENT_SIZE);
> >   FastLED.setBrightness(64);
> > }
> > 
> > uint64_t prev_millis = 0;
> > uint64_t current_millis = 0;
> > 
> > void loop() {
> >   static uint8_t pos = 0;
> >   pos++;
> >   if (pos == MATRIX_SIZE_Y) pos == 0;
> >   for (uint8_t y = 0; y < MATRIX_SIZE_Y; y++) {
> >     for (uint8_t x = 0; x < MATRIX_SIZE_X; x++) {
> >       leds[XY(x, y)] = CHSV((float)(x+y) / (float)(MATRIX_SIZE_X+MATRIX_SIZE_Y) * 255.0f + pos, 255, 255);
> >     }
> >   }
> >   FastLED.show();
> > 
> >   current_millis = millis();
> >   if (current_millis > prev_millis + 1000) {
> >     prev_millis = current_millis;
> >     Serial.print("FPS: ");
> >     Serial.println(FastLED.getFPS());
> >   }
> > }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
[... 111 lines cut ...]
> board_build.f_cpu = 240000000L
> 
> monitor_filters = 
> 	default
> 	esp32_exception_decoder  ; Decode exceptions so that they are human readable.
> 
> ; Symlink in the FastLED library so that changes to the library are reflected in the project
> ; build immediatly.
> lib_deps =
>   FastLED=symlink://./
> 
> build_type = debug
> 
> build_flags =
> 	-DDEBUG
>     -g
>     -Og
>     -DCORE_DEBUG_LEVEL=5
>     -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE
> 
>     
> check_tool = clangtidy
> ```

50 fps is too small for parallel output of strips 256 pixels each. In some scenarios (platformio.ini configurations) I didn't get an rmt error in the serial port too, but the problem is still here, frame rate is the same as when I getting error.

zackees, 2024-11-19:
You're right.

Good catch.

Now i have something to benchmark against.

sutaburosu, 2024-12-21:
With the recent v3.9.7 release I get these results on a ESP32-S3-WROOM-1.

- `#define FASTLED_RMT5_RECYCLE 0`
  - Single Controller
    - ✅ LEDs working fine
    - ✅ Refresh rate close to LEDs' spec
    - 🐛  Each call to show() prints:
      `E (583) rmt: rmt_tx_disable(835): channel can't be disabled in state 1`
      `E (583) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed`
  - Multiple Controllers
    - ❗ LEDs receive nothing
    - ❗ show() seems to block forever
    - 🐛 Constant stream of:
      `E (583) rmt: rmt_tx_disable(835): channel can't be disabled in state 1`
      `E (583) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed`
      `E (586) rmt: rmt_tx_register_to_group(152): no free tx channels`
      `E (591) rmt: rmt_new_tx_channel(288): register channel failed`
- `#define FASTLED_RMT5_RECYCLE 1`
  - Single Controller
    - ✅ LEDs working fine
    - ✅ at expected refresh rates
    - ✅ No errors printed
  - Multiple Controllers
    - ✅ LEDs working fine
    - 🐛 Low refresh rates, equivalent to non-parallel rates
    - 🐛 Only the first call to show prints:
      `E (583) rmt: rmt_tx_register_to_group(152): no free tx channels`
      `E (583) rmt: rmt_new_tx_channel(288): register channel failed`

zackees, 2024-12-21:
Thanks for the update.

How many strips are you using in your multi strip setup?

sutaburosu, 2024-12-21:
I've tested between 1 and 5 pins, with LEDs attached to 2 of them at a time. 2-5 pins all show exactly the same results.

zackees, 2024-12-22:
Thanks, this helps a lot.

zackees, 2024-12-30:
Fixed in 3.9.8. Feel free to re-open if this is not the case.

sutaburosu, 2024-12-31:
With v3.9.8 on a ESP32-S3-WROOM-1 I see:

- Out of the box (i.e. `FASTLED_RMT5_RECYCLE` is 0)
  - Single pin (256 LEDs)
    - ✅ LEDs working fine, async working well
    - ✅ 125 FPS
    - ✅ No RMT error messages
  - Two pins (256 + 1 LEDs)
    - ❗ Panic in show():
      ```
      ============ After Setup End =============
      E (522) rmt: rmt_tx_register_to_group(132): no free tx channels
      E (523) rmt: rmt_new_tx_channel(239): register channel failed
      E (523) led_strip: led_strip_refresh_async(42): invalid argument
      ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42006a75
        #0  0x42006a75 in fastled_rmt51_strip::RmtLedStripNoRecycle::draw_async() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp:269
            (inlined by) fastled_rmt51_strip::RmtLedStripNoRecycle::draw() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp:275
      
      file: ".pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp" line 269
      func: void fastled_rmt51_strip::RmtLedStripNoRecycle::draw_async()
      expression: led_strip_refresh_async(mLedStrip)
      
      abort() was called at PC 0x4037bad7 on core 1

      Backtrace: 0x40377abe:0x3fcebbf0 0x4037bae1:0x3fcebc10 0x40380f85:0x3fcebc30 0x4037bad7:0x3fcebcb0 0x42006a75:0x3fcebce0 0x42006d8e:0x3fcebd00 0x42006fd5:0x3fcebd20 0x42002686:0x3fcebd40 0x4200e2e1:0x3fcebd60 0x42011e7c:0x3fcebdb0 0x4037dfd6:0x3fcebdd0
        #0  0x40377abe in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
        #1  0x4037bae1 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
        #2  0x40380f85 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:38
        #3  0x4037bad7 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:50
        #4  0x42006a75 in fastled_rmt51_strip::RmtLedStripNoRecycle::draw_async() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp:269
            (inlined by) fastled_rmt51_strip::RmtLedStripNoRecycle::draw() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/led_strip/rmt_strip.cpp:275
        #5  0x42006d8e in RmtController5::showPixels() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/rmt/idf5_rmt.cpp:110
        #6  0x42006fd5 in ClocklessController<48, 250, 625, 375, (EOrder)66, 0, false, 5>::endShowLeds(void*) at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/platforms/esp/32/rmt/idf5_clockless_rmt_esp32.h:60
        #7  0x42002686 in CFastLED::show(unsigned char) at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/FastLED.cpp:118
        #8  0x4200e2e1 in CFastLED::show() at .pio/libdeps/esp32-s3-devkitm-1/FastLED/src/FastLED.h:743
            (inlined by) loop() at src/main.cpp:145
        #9  0x42011e7c in loopTask(void*) at ~/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74
        #10 0x4037dfd6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
        ```
- `#define FASTLED_RMT5_RECYCLE 1`
  - Single pin (256 LEDs)
    - ✅ LEDs working fine, async working well
    - 🐛 94 FPS down from 125
    - ✅ No RMT error messages
  - Two pins (256 + 1 LEDs)
    - ✅ LEDs working fine, async working well
    - 🐛 69 FPS, where 125 should be possible.
    - 🐛 Printed on first show():
      ```
      E (522) rmt: rmt_tx_register_to_group(132): no free tx channels
      E (523) rmt: rmt_new_tx_channel(239): register channel failed
      ```
  - Four pins (4 * 256 LEDs)
    - ✅ LEDs working fine, async working well
    - 🐛 25 FPS where >60 should be possible
    - 🐛 Printed on first show():
      ```
      E (522) rmt: rmt_tx_register_to_group(132): no free tx channels
      E (523) rmt: rmt_new_tx_channel(239): register channel failed
      ```

sutaburosu, 2024-12-31:
You mean with historic FastLED versions? I don't know; I only got the S3 very recently. If it's helpful I can do this. Would testing ESP32 core 2.x.x too be of any value?

zackees, 2024-12-31:
Thank you for this. The ESP32 core 2.x.x is not a high priority for the S3, it is for the 8266.

sutaburosu, 2024-12-31:
Aha! Going back to 3.7.8 gives 62 FPS for 5 pins: 4 * 256 LEDs + 1 LED. 3.9.0 and later give 24 FPS.
There are 1,582 commits between those releases, so it looks like I have some bisecting and testing to do. That will have to wait until tomorrow.

zackees, 2024-12-31:
It probably happened on the RMT4 switch. Can you just disable RMT5 via FASTLED_RMT5=1 as a build define and test? I think that will restore the performance.

sutaburosu, 2024-12-31:
That is much easier. With `-DFASTLED_RMT5=0` 3.9.1 - 3.9.4 give expected FPS.

3.9.5 and onwards give:
```
E (101) rmt(legacy): CONFLICT! driver_ng is not allowed to be used with the legacy driver

abort() was called at PC 0x42008377 on core 0

Backtrace: 0x40377ae6:0x3fceb200 0x4037bb61:0x3fceb220 0x40381005:0x3fceb240 0x42008377:0x3fceb2c0 0x4200d86a:0x3fceb2e0 0x403772ed:0x3fceb310 0x403cdb0e:0x3fceb340 0x403cdea5:0x3fceb380 0x403c9919:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x40377ae6 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x4037bb61 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x40381005 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:38
  #3  0x42008377 in check_rmt_legacy_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/deprecated/rmt_legacy.c:1386 (discriminator 3)
  #4  0x4200d86a in do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:205
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:486
  #5  0x403772ed in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:724
```

zackees, 2025-01-04:
Okay try the latest release: 3.9.9. I removed the code to disable the RMT device between draw calls. I benchmarked 4 parallel strips at 6.5 ms for 256 leds on the Xiao S3, which is the expected maximum speed. It should all work now and run at the fastest speed.

Let me know if this isn't the case and I'll re-open.

closed 2025-01-04 by zackees

dcniemann, 2025-01-08:
This still appears to be the case on an original Espressif and an XTVTX ESP32-S3-WROOM-1 model.
I'm using Arduino IDE 2.3.2 with the Espressif esp32 package on version 3.1.1 and FastLED 3.9.9.

My code:
```c++
#include <FastLED.h>

#define LED_PIN_1 15
#define LED_PIN_2 16

#define NUM_INFO_LEDS 4
#define NUM_BOTTOM_LEDS 64

CRGB* _bottomleds = new CRGB[NUM_BOTTOM_LEDS];
CRGB* _infoleds = new CRGB[NUM_INFO_LEDS];

void setup() {
  Serial.begin(115200);

  FastLED.addLeds<WS2812, LED_PIN_1, GRB>(_infoleds, NUM_INFO_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.addLeds<WS2812, LED_PIN_2, GRB>(_bottomleds, NUM_BOTTOM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(64);

  Serial.println("Setup Done");
}

void loop() {
  for (int i = 0; i < 256; i++) {
    fill_solid(_infoleds, NUM_INFO_LEDS, blend(CRGB::Black, CRGB::White, i));
    fill_solid(_bottomleds, NUM_BOTTOM_LEDS, blend(CRGB::Black, CRGB::White, i));
    FastLED.show();
    delay(1);
  }
}
```

The error(s):
```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
Setup Done
E (6) rmt: rmt_tx_register_to_group(152): no free tx channels
E (48) rmt: rmt_new_tx_channel(288): register channel failed
E (49) rmt: rmt_tx_register_to_group(152): no free tx channels
E (49) rmt: rmt_new_tx_channel(288): register channel failed
E (55) rmt: rmt_tx_register_to_group(152): no free tx channels
E (59) rmt: rmt_new_tx_channel(288): register channel failed
E (66) rmt: rmt_tx_register_to_group(152): no free tx channels
E (70) rmt: rmt_new_tx_channel(288): register channel failed
E (77) rmt: rmt_tx_register_to_group(152): no free tx channels
E (81) rmt: rmt_new_tx_channel(288): register channel failed
E (87) rmt: rmt_tx_register_to_group(152): no free tx channels
E (92) rmt: rmt_new_tx_channel(288): register channel failed
E (98) rmt: rmt_tx_register_to_group(152): no free tx channels
```

This occurs as soon as I use more than 2 channels, 1 is fine and works as expected.
Any suggestions?

zackees, 2025-01-09:
Fixed in 3.9.10, which is pending release. Feel free to manually install the newest FastLED to get it now.