FastLED/FastLED #1774: FireBeetle esp32-e built in RGB green always on

Fluidisator, 2024-11-14:
Hello,
I tried the blink example on the built in RGB led of a FireBeetle esp32-e v1.0 (DFR0654). I'm using the arduino IDE and the v3.9.3 of the FastLed library. The green dot is always on, whatever is the set color. I tried older version, and the issue appeared since v3.9.0. It works well with v3.7.8.

nomakewan, 2024-11-14:
Perhaps related to #1761 ?

Fluidisator, 2024-11-15:
Indeed, same as reported by [pyahmed](https://github.com/FastLED/FastLED/issues/1761#issuecomment-2460242068) in the issue you mentioned.

tikiss, 2024-11-16:
I am experiencing the same problem with Waveshare ESP32-C6-Zero [Link](https://www.waveshare.com/wiki/ESP32-C6-Zero).
The first LED in chain is always full green.
I did a test with the same HW using the "Freenove_WS2812_Lib_for_ESP32" Library, it is working normally, with FastLED not.
I have checked the signal with oscilloscope, with Freenove library it is like that:
[image]
With Fast LED, it look like this:
[image]

As you can see, in case of FastLED there is a 300ms High periode, which in my opinion causing that the first LED is full green.

But I have no idea, how to fix it. May be the oscilloscope images helps to solve the issue. (in both cases the signal is measured on the ESP32 module GPIO pin)

zackees, 2024-11-18:
Thanks for the oscilliscope readings. I'm wondering if this is related to how we tear down the RMT driver every frame then re-instantiate it. It's possible there is a bug that manifests as a green dot on the first frame and you are seeing this because this RMT driver is hitting this path every time it re-instantiates.

zackees, 2024-11-18:
Can you try forcing DMA mode off? `-DFASTLED_RMT_WITH_DMA=0` as a build define. It won't work as an include define.

I'm looking on the internet and I'm seeing that there is at least one major bug with ESP32-wroom involving initialized memory and the clock selection source going to REF_TICK which is 1MHZ instead of 10MHZ that the RMT is supposed to be using.

It has something to do with DMA. Just wanted to see if this somehow fixes the green pixel.

zackees, 2024-11-18:
I can't repro this with a generic wroom board. I'm able to connect 4 strips no problem. No errors and no LED corruption.

So questions:

What IDE are you using? Arduino? PlatformIO? esp-idf?

Have you tried moving pins?

Have you tried using a different build system?

Here's my code:

**platformio.ini**
```

[platformio]
src_dir = dev ; target is ./dev/dev.ino

[env:dev]
; This is the espressif32 platform which is the 4.1 toolchain as of 2024-Aug-23rd
; platform = espressif32
; The following platform enables the espressif32 platform to use the 5.1 toolchain, simulating
; the new Arduino 2.3.1+ toolchain.

# Developement branch of the open source espressif32 platform
platform = https://github.com/pioarduino/platform-espressif32.git#develop

framework = arduino
board = firebeetle32

upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

; Symlink in the FastLED library so that changes to the library are reflected in the project
; build immediatly.
lib_deps =
  FastLED=symlink://./

build_type = debug

build_flags =
	-DDEBUG
    -g
    -Og
    -DCORE_DEBUG_LEVEL=5
    -DLOG_LOCAL_LEVEL=ESP_LOG_VERBOSE
    -DFASTLED_RMT5=1
    -DFASTLED_ESP32_SPI_BULK_TRANSFER=1
    
check_tool = clangtidy
```

**noisering ino***

```C++

/// @file    NoisePlusPalette.ino
/// @brief   Demonstrates how to mix noise generation with color palettes on a
/// 2D LED matrix
/// @example NoisePlusPalette.ino

#include <Arduino.h>

#include <stdio.h>

#include "json.h"
#include "math_macros.h"
#include "noisegen.h"
#include "screenmap.h"
#include "slice.h"
#include "ui.h"
#include "FastLED.h"
#include "pir.h"
#include "timer.h"

#define LED_PIN 27
#define BRIGHTNESS 96
#define COLOR_ORDER GRB
#define NUM_LEDS (22*22)
#define PIN_PIR 0

CRGB leds[NUM_LEDS];

Slider brightness("Brightness", 255, 0, 255);
Slider scale("Scale", 4, .1, 4, .1);
Slider timeBitshift("Time Bitshift", 5, 0, 16, 1);
Slider timescale("Time Scale", 1, .1, 10, .1);
Pir pir(PIN_PIR);

Timer timer;
float current_brightness = 0;
bool first = true;

uint8_t update_brightness(bool clicked, uint32_t now) {
    if (clicked) {
        timer.start(millis(), 1000 * 60 * 3);
    }
    if (timer.update(now)) {
        current_brightness += .6;
    } else {
        current_brightness *= .97;
    }
    uint32_t brightness = current_brightness;
    if (brightness > 255) {
        brightness = 255;
    }
    return brightness;
}

void setup() {
    ScreenMap xyMap = ScreenMap::Circle(NUM_LEDS, 2.0, 2.0);
    FastLED.addLeds<WS2811, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS)
        .setCorrection(TypicalLEDStrip)
        .setScreenMap(xyMap);
    FastLED.addLeds<WS2811, 3, COLOR_ORDER>(leds, NUM_LEDS)
        .setCorrection(TypicalLEDStrip)
        .setScreenMap(xyMap);
    FastLED.addLeds<WS2811, 4, COLOR_ORDER>(leds, NUM_LEDS)
        .setCorrection(TypicalLEDStrip)
        .setScreenMap(xyMap);
    FastLED.addLeds<WS2811, 5, COLOR_ORDER>(leds, NUM_LEDS)
        .setCorrection(TypicalLEDStrip)
        .setScreenMap(xyMap);
    FastLED.setBrightness(brightness);
}

void loop() {
    const bool detectMotion = pir.detect();
    const bool detected_motion = detectMotion || first || detectMotion;
    first = false;
    uint8_t bri = update_brightness(detected_motion, millis());
    FastLED.setBrightness(bri);
    uint32_t now = millis();
    double angle_offset = double(now) / 32000.0 * 2 * M_PI;
    //angle_offset = 0;
    now = (now << timeBitshift.as<int>()) * timescale.as<double>();
    // inoise8(x + ioffset,y + joffset,z);
    // go in circular formation and set the leds
    for (int i = 0; i < NUM_LEDS; i++) {
        float angle = i * 2 * M_PI / NUM_LEDS + angle_offset;
        float x = cos(angle);
        float y = sin(angle);
        x *= 0xffff * scale.as<double>();
        y *= 0xffff * scale.as<double>();
        uint16_t noise = inoise16(x, y, now);
        uint16_t noise2 = inoise16(x, y, 0xfff + now);
        uint16_t noise3 = inoise16(x, y, 0xffff + now);
        noise3 = noise3 >> 8;
        int16_t noise4 = map(noise3, 0, 255, -64, 255);
        if (noise4 < 0) {
            noise4 = 0;
        }

        leds[i] = CHSV(noise >> 8, MAX(128, noise2 >> 8), noise4);
        // std::swap(leds[i].b, leds[i].g);
    }
    FastLED.show();
}
```

tikiss, 2024-11-18:
> What IDE are you using? Arduino? PlatformIO? esp-idf?

I am using using the same the first poster, Arduino IDE, and 3.9.3 FastLED library.
Unfortunately I was not able to setup the project in PlatformIO, in ESP-IDF I was able to do a quick check (but not with FastLED library), and I was not experienced the problem.

> Have you tried moving pins?

Yes, I have tried, first I used the GPIO0, than moved to GPIO1, and GPIO14 and GPIO15, no change at all.

> Have you tried using a different build system?

Not really, I am not a pro in programming, I tried to use some easy development environment.

One more thing I have experienced, before that look at my test code:

```
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>

#define LED_PIN     15
#define BRIGHTNESS  60
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
#define COLOR_ORDER GRB
#define NUM_LEDS 60
CRGBArray<NUM_LEDS> leds;

void setup()
{
  Serial.begin(115200); // Enable serial output
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);
  leds.fill_solid(0x000000); // Clear all leds
  FastLED.show();
}

void loop()
{
  delay(1000);
  hue = 240;
  for (uint8_t i = 0; i < NUM_LEDS; ++i)
  {
      leds[i].setRGB(0, 0, 255);
      if (i > 1)
      {
        leds[i-1].setRGB(0, 0, 0);
      }
      FastLED.show();
      delay(300);
  }

  leds.fill_solid(0x0000ff); // Clear all leds
  FastLED.show();
}
```
So back to what I experienced, if set in the "setup" section, to fill the leds blue:
`  leds.fill_solid(0x0000ff); `
And put a 1 sec delay after the show, than I see, all the leds are blue, no green at all.
Than the 1 sec is gone, the program goes in the loop, than the first LED changes to green, and remains there forever.

Let me check your other answers, and If I can understand it, I try what you suggest.

tikiss, 2024-11-18:
> Can you try forcing DMA mode off? `-DFASTLED_RMT_WITH_DMA=0` as a build define. It won't work as an include define.
> 
> You can also copy/symlink FastLED into your Arduino path and manually set this build define in the src\platforms\esp\32\led_strip\defs.h
> 
> I'm looking on the internet and I'm seeing that there is at least one major bug with ESP32-wroom involving initialized memory and the clock selection source going to REF_TICK which is 1MHZ instead of 10MHZ that the RMT is supposed to be using.
> 
> It has something to do with DMA. Just wanted to see if this somehow fixes the green pixel.

I have put this in the third line in the def.h:
`#define FASTLED_RMT_WITH_DMA 0`
But I cannot see any change, the first LED is green.
Did I performed the test how you meant?

tikiss, 2024-11-18:
I did one more test, I put two color definition in the setup section with 1sec delay, the first color (blue) showen correctly without green, but the second color (red) is not full of red "pixels" the first is green.
```
void setup()
{
  Serial.begin(115200); // Enable serial output
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS);
  leds.fill_solid(0x0000ff);
  FastLED.show();
  delay(1000);
  leds.fill_solid(0xff0000);
  FastLED.show();
  delay(1000);
}
```

zackees, 2024-11-18:
There's a closed bug report recently of someone who followed my advice and ported his project to platform io.

The feedback he gave me was that it was easier than expected, and for some reason the bug went away.

I'm sure this bug is happening.

But here's the problem - I can't reproduce it.

There's either two possibilities: this bug is due to some bug in hardware, or in software. The big question is: what is the software context?

What combination of libraries did you point and click to get to this point? And most importantly, what version is your arduino core?

As an arduino user, you actually don't have this information handy. You actually have to engage in forensics, like a private eye, to reconstruct the build environment.

I can't reproduce your build.

However, if your project was in platformio, your platformio.ino tells me everything, and i would just copy that one file and be able to
reproduce this issue or verify that it's a hardware defect in whatever batch run you got.

And, if your like the last guy, simply putting it in platformio may just solve the bug for you - better than vegas odds infact.

Here's my guide to setting up platformio with a template project that already compiles. All you have to do is copy the files in your sketch directory into the src folder in this template.

https://github.com/FastLED/PlatformIO-Starter

> I did one more test, I put two color definition in the setup section with 1sec delay, the first color (blue) showen correctly without green, but the second color (red) is not full of red "pixels" [...]

tikiss, 2024-11-19:
With the PlatformiIO-Starter I was able to setup the project in PatformIO.
Put the same code, as I used in Arduino IDE to test my setup, and I have the same result, the first LED remains on always.
Here is my test code:
```
#include <FastLED.h>

#define LED_PIN     1
#define NUM_LEDS    60
#define BRIGHTNESS  255
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB

CRGBArray<NUM_LEDS> leds;

void setup() {
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(TypicalLEDStrip);
  FastLED.setBrightness(BRIGHTNESS);
  leds.fill_solid(0x000000); // Clear all leds
  FastLED.show();
}

void loop() {
    for (uint8_t i = 0; i < NUM_LEDS; ++i)
  {
      leds[i].setRGB(0, 0, 255);
      if (i > 1)
      {
        leds[i-1].setRGB(0, 0, 0);
      }
      FastLED.show();
      Serial.print(i);
      Serial.print("\n");
      delay(100);
  }

  leds.fill_solid(0x000000); // Clear all leds
  FastLED.show();
}
```
The platformio.ini is the same in the starter: [platformio.esp32c6.ini](https://github.com/FastLED/PlatformIO-Starter/blob/main/platforms/platformio.esp32c6.ini)

I receive a warning during build:
```
In file included from .pio/libdeps/esp32c6/FastLED/src/platforms/esp/32/fastled_esp32.h:3,
                 from .pio/libdeps/esp32c6/FastLED/src/platforms.h:44,
                 from .pio/libdeps/esp32c6/FastLED/src/FastLED.h:70,
                 from src/main.cpp:1:
.pio/libdeps/esp32c6/FastLED/src/platforms/esp/32/fastpin_esp32.h:117:2: warning: #warning Unknown ESP32 chip variant. Only pins defined by ESP-IDF will be masked. [-Wcpp]
  117 | #warning Unknown ESP32 chip variant.  Only pins defined by ESP-IDF will be masked.
```
I do not know it has nothing to with this problem, but for me is strange why the ESP is unkown, it is a ESP32-C6.

nomakewan, 2024-11-19:
I notice zackees is using the WS2811 while tikiss is using the WS2812B. Perhaps an issue specific to the WS2812B chipset code?

zackees, 2024-11-19:
You can safely ignore the error. It's saying its doesn't know which of the five or so pins are used for internal flash programming for a new sketch.

boards hide those pins anyway.

This is fixed in master. If you are eager for the fix you can replace FastLED in your libs_deal with our github at https://github.com/fastled/fastled

This will give you the latest always. You can also pin to a commit. Ask AI how to do this it will know.

> There's a closed bug report recently of someone who followed my advice and ported his project to platform io.
> 
> The feedback he gave me was that it was easier than expected, and for some reason the bug went away.
> 
> But here's the problem - I can't reproduce your problem without knowing the state of your environment.
> 
> There's either two possibilities: this bug is due to some bug in hardware, or in software. The big question is: what is the software context?
> 
> What combination of libraries did you point and click to get to this point? And most importantly, what version is your arduino core?
> 
> As an arduino user, you actually don't have this information handy. You actually have to engage in forensics, like a private eye, to reconstruct the build environment.
> 
> I can't reproduce your build.
> 
> However, if your project was in platformio, your platformio.ini tells me everything, and i would just copy that one file and be able to
> reproduce this issue or verify that it's a hardware defect.
> 
> Here's my guide to setting up platformio with a template project that already compiles. All you have to do is copy the files in your sketch directory into the src folder in this template.
> 
> https://github.com/FastLED/PlatformIO-Starter
> 
> 
> 
> > I did one more test, I put two color definition in the setup section with 1sec delay, the first color (blue) showen correctly without green, but the second color (red) is not full of red "pixels" the first is green.
> > 
> > ```
> > 
> > void setup()
> > 
> > {
> > 
> >   Serial.begin(115200); // Enable serial output
> > 
> >   FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS);
> > 
> >   FastLED.setBrightness(BRIGHTNESS);
> > 
> >   leds.fill_solid(0x0000ff);
> > 
> >   FastLED.show();
> > 
> >   delay(1000);
> > 
> >   leds.fill_solid(0xff0000);
> > 
> >   FastLED.show();
> > 
> >   delay(1000);
> > 
> > }
> > 
> > ```
> 
>

tikiss, 2024-11-19:
> I notice zackees is using the WS2811 while tikiss is using the WS2812B. Perhaps an issue specific to the WS2812B chipset code?

I have checked the LED type, and the datasheet, could you please someone clarify what are the meaning of T1, T2 and T3 times in library?
```
#ifndef FASTLED_WS2812_T1
#define FASTLED_WS2812_T1 250
#endif

#ifndef FASTLED_WS2812_T2
#define FASTLED_WS2812_T2 625
#endif

#ifndef FASTLED_WS2812_T3
#define FASTLED_WS2812_T3 375
#endif

// WS2812 - 250ns, 625ns, 375ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = GRB>
class WS2812Controller800Khz : public ClocklessController<
	DATA_PIN,
	C_NS_WS2812(FASTLED_WS2812_T1),
	C_NS_WS2812(FASTLED_WS2812_T2),
	C_NS_WS2812(FASTLED_WS2812_T3),
	RGB_ORDER> {};
```
Because if I open the following WS2812B datasheet: [https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf](https://cdn-shop.adafruit.com/datasheets/WS2812B.pdf)
I see different T0, T1 and reset times.
And if I open the LED datasheet what I use on my board: [https://www.lcsc.com/datasheet/lcsc_datasheet_2012110135_Worldsemi-WS2812B-2020_C965555.pdf](https://www.lcsc.com/datasheet/lcsc_datasheet_2012110135_Worldsemi-WS2812B-2020_C965555.pdf)
It has also different values.
So could be that the LED timing is not okay?

tikiss, 2024-11-19:
Okay, I found the answer to my question:
```
// At T=0        : the line is raised hi to start a bit
// At T=T1       : the line is dropped low to transmit a zero bit
// At T=T1+T2    : the line is dropped low to transmit a one bit
// At T=T1+T2+T3 : the cycle is concluded (next bit can be sent)
//
// Python script to calculate the values for T1, T2, and T3 for FastLED:
//
//  print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
//  T0H = int(input("  T0H: "))
//  T0L = int(input("  T0L: "))
//  T1H = int(input("  T1H: "))
//  T1L = int(input("  T1L: "))
//  
//  duration = max(T0H + T0L, T1H + T1L)
//  
//  print("The max duration of the signal is: ", duration)
//  
//  T1 = T0H
//  T2 = T1H
//  T3 = duration - T0H - T0L
//  
//  print("T1: ", T1)
//  print("T2: ", T2)
//  print("T3: ", T3)
```
In my opinion the timing should be okay.
I am pretty sure that the problem is caused by the 300us high period beginning of the pixel transfer, as I showed above.
Here is an other oscilloscope picture, this was recoded during the start-up (switch on), as you can see the first transfer has no high period before, and there there also no green "pixel", buf after the second and all other are starting with that high period.
[image]

But no idea why it is starting with 300us high period.

tikiss, 2024-11-19:
I am now using PlatformIO, used the guide what you provided.
Here is the file content, which is almost the same as in the guide:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c6] ; the esp32c6 part can be renamed to anything you want.
; Note that this special platform override is used to bring in the
; idf5.1 toolchain that is now included in Arduino 2.3.1+, and may
; not be necessary in the near future.
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
;platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc3/platform-espressif32.zip
framework = arduino
; This is a popular ESP32-S3 board.
board = esp32-c6-devkitc-1

upload_protocol = esptool

upload_port = /dev/cu.usbmodem1401
monitor_port = /dev/cu.usbmodem1401
monitor_speed = 115200

; The exception decoder is a useful tool for debugging that is difficult
; to enable in the Arduino IDE. But in platformio it will decode the
; 0x0f4852c0 addreesses into the actual function names.
monitor_filters = 
	default
	esp32_exception_decoder

lib_deps =
	FastLED

build_flags =
                -DARDUINO_USB_CDC_ON_BOOT=1
                -DARDUINO_USB_MODE=1
```

zackees, 2024-11-20:
Yes, add this

```
build_flags =
    -DFASTLED_RMT5=0
```

This will use the RMT4 driver.

zackees, 2024-11-20:
If this fixes you then I'll force RMT4 to WROOM boards until a proper cause can be found.

tikiss, 2024-11-20:
> Yes, add this
> 
> ```
> build_flags =
>     -DFASTLED_RMT5=0
> ```
> 
> This will use the RMT4 driver.

Compiled and flashed successfuly but it is not working at all :( all LEDs are black.

wouterf11, 2024-12-07:
First of all, many thanks for developing and maintaining this brilliant library!

I'm using a ESP32-C6-Devkit-C1 and am trying to run the Blink example on the built-in LED (DATA_PIN 8, NUM_LEDS 1, WS2812B) using FastLED v3.9.4 in Arduino IDE 2.3.4. Unfortunately in this example the built-in LED green is also always on.
Since I'm using a ESP32-C6 reverting to RMT4 is probably not an option for me.

The blink example from espressif (https://components.espressif.com/components/espressif/led_strip/versions/3.0.0/examples/led_strip_rmt_ws2812?language=en) does work as expected when flashed using vscode using the esp-idf extension.

I've tried adding 
`memset(rmt_strip->pixel_buf, 0, rmt_strip->strip_len * rmt_strip->bytes_per_pixel); `
to led_strip_rmt_refresh_async in led_strip_rmt_dev.cpp for debugging purposes, but even then the LED is always green using FastLED.

Would you have any pointers how I could continue debugging?

zackees, 2024-12-08:
It's weird because i've never seen this bug.

Does it only happen on the built in led.

zackees, 2024-12-09:
Would you gift me a this board. I'll provide the fix if i can simply reproduce it. Without being able to reproduce this my hands are tied.

I suspect this is due to the rmt driver being reinitialized every frame in order to enable writing to more strips than rmt channels through rmt recycling.

RMT recycling causes staggered effects anyway so it might be a good idea to acquire the resource exactly once and never allow it to be released.

tikiss, 2024-12-10:
I can give you a Waveshare ESP32-C6-Zero [Link](https://www.waveshare.com/wiki/ESP32-C6-Zero). As I see, you are not from EU. So I do know can I send it via normal mail or not

wouterf11, 2024-12-11:
> It's weird because i've never seen this bug.
> 
> Does it only happen on the built in led?

When I connect a stip, it seems only the first led is lit green, the others are fine. Some browsing on the web led me to https://www.reddit.com/r/FastLED/comments/wdbdia/comment/iijpgsi/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button
This workaround applies to my case as well. At least for the strip, the built-in led remains green for some mysterious reason.

zackees, 2024-12-15:
Okay I still can't reproduce this bug with a ESP32Dev/Wroom/S3/C3/C6.

This is the first attempt I'm making to fix it while "flying blind".

The code is located on the `fix-rmt` branch, which disables the RMT recycling.

https://github.com/FastLED/FastLED/tree/fix-rmt

You can download this here as a zip file and load it up in your arduino sketch.

https://github.com/FastLED/FastLED/archive/refs/heads/fix-rmt.zip

You can also use platformio and set the FastLED dependency to the branch like this:

```
lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

Here's an example:

```
[env:dev]
platform = espressif32
framework = arduino
board = esp32dev

monitor_speed = 115200
upload_protocol = esptool

monitor_filters = 
	default
	esp32_exception_decoder  ; Decode exceptions so that they are human readable.

lib_deps = FastLED@https://github.com/FastLED/FastLED.git#fix-rmt
```

zackees, 2024-12-18:
Okay I've made this even easier and this will work with ArduinoIDE. You need to update to the latest release that was just published, 3.9.5.

Do the following:

```C++
#define FASTLED_RMT5_RECYCLE=0
#include "FastLED.h"
```

zackees, 2024-12-18:
I don't think you have that right. The new behavior it RMT_RECYCLE=0.

Another user reported that disabling the RMT recycling is what worked for them, so I just pushed this out.

I'll follow up with some refinements.

How many strips are you using btw?

wouterf11, 2024-12-18:
I'm using a single strip

wouterf11, 2024-12-18:
[image]

zackees, 2024-12-18:
Can we schedule a call and walk through this together

zackees, 2024-12-20:
Okay, 3.9.6 had a mistake in it where the RMT recycle value was inverted. I've fixed it in 3.9.7 which will end up in the Arduino IDE repo shortly. Feel free to install it now from the zip file to try it immediatly.

wouterf11, 2024-12-21:
Apologies for my limited availability. v3.9.7 works great! I'm curious to see what https://github.com/espressif/esp-idf/issues/15049 will reveal. Happy holidays!

ccarmatic, 2024-12-21:
> Okay, 3.9.6 had a mistake in it where the RMT recycle value was inverted. I've fixed it in 3.9.7 which will end up in the Arduino IDE repo shortly. Feel free to install it now from the zip file to try it immediatly.

Hello I am using 3.9.7, as per installed by Arduino IDE (the author is listed as Daniel Garcia),  and I am getting these errors
```
E (1181787) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
E (1181802) rmt: rmt_tx_disable(774): channel can't be disabled in state 1
```
I am using an ESP32-WROOM32 and trying to use DMX Artnet with Fastled , following this example https://learn.sparkfun.com/tutorials/using-artnet-dmx-and-the-esp32-to-drive-pixels/creating-an-artnet-node-on-the-esp32

zackees, 2024-12-22:
How many strips are you using?

ccarmatic, 2024-12-23:
> How many strips are you using?

One strip, 330 ws2812 LEDs ... the LED pin is Pin 25, if that makes any difference

zackees, 2024-12-23:
Bad news: Disabling RMT recycle broke our multi strip case on ESP32.

Good news: Espressif got back to me and said that this bug is a manifestation of the gpio state being reset after the rmt disable.

See: https://github.com/espressif/esp-idf/issues/15049

They recommended that I set the INPUT_PULLDOWN on the pin right after the rmt_disable call, to prevent the floating HIGH state on the pin. I've added this and also a micro delay of 10 microseconds to make sure that the RMT blip, if it's still happening, will now only be momentary and also the 10 microsecond delay will prevent the now shortened blip from being interpreted as a 1 or a 0.

## What I'm going to do:

  * I'm applying the espressif fix
  * I'm re-enabling RMT recycling for the next release.

## What I need from you

If nothing is done, the rmt recycling will be re-enabled on the next release. I *hope* this won't break you, but there's a small possibility that you'll get a regression back to the green led being stuck on.

However, if someone tests the updated branch rmt-fix https://github.com/FastLED/FastLED/tree/fix-rmt and let me know that it continues to work for you, we can avoid this inconvenience for you. This branch has both the espressif fix and also RMT recycling re-enabled.

If...
  * The green light is stuck on: then I will let espressif know and revert the change.
  * If the green light continues to be fixed: Then a release will be minted within 24 hours.
  * If no feedback is given, then the release will be minted by the 28th, and the chips will fall where they lay. If this breaks you then you will have to apply the workaround until the next release.
  
I'm flying blind here since I can't reproduce your issue locally. So thanks in advance, for your help.

sutaburosu, 2025-01-01:
As another data point, [there was a report](https://www.reddit.com/r/FastLED/comments/1hq7b6m/data_pin_goes_high_impedance_before_transmit/) that adding an external 10k pull-down resistor can also fix the greenish first LED.

zackees, 2025-01-01:
Thanks, cross posted to espressif

zackees, 2025-01-04:
Okay, I went deeper and prevented the RMT channels from disabling at all. This should keep the pin states stable now. Upgrade to 3.9.9 for the latest fix.

closed 2025-01-04 by zackees