FastLED/FastLED #1775: Class name conflicts

pschatzmann, 2024-11-17:
The FastLED has a class naming conflict with the AudioTools library for the Str class.
It would be easy to resolve if we could activate namespace support in the fastled_config.h

zackees, 2024-11-17:
Please post exact output from the tool.

If it's an include path conflict then str.h could become fl/str.h

If it's a symbol collision then maybe namespaces make sense.

pschatzmann, 2024-11-17:
It might be potentially also your mentioned first point but I don't get it because I use Linux with a case sensitive file system:  So potentially there might be another issue e.g. in Windows.

In Linux it is a symbol collition: AudioTools uses it's own namespace but unfortunately FastLED does not.
```
Generating function prototypes...
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -c @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="linux" -DARDUINO_FQBN="esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/defines -I/tmp/.arduinoIDE-unsaved20241017-528943-1wbxl6m.rs9d/streams-audiokit-fft-led -iprefix /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/include/ @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/includes -I/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/qio_qspi/include -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32 -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/esp32 -I/home/pschatzmann/Development/Arduino/libraries/arduino-audio-tools/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/Network/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/NetworkClientSecure/src -I/home/pschatzmann/Development/Arduino/libraries/arduino-audio-driver/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/SPI/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/Wire/src -I/home/pschatzmann/Development/Arduino/libraries/FastLED/src @/tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/build_opt.h @/tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/file_opts /tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/sketch/streams-audiokit-fft-led.ino.cpp -o /tmp/1481549484/sketch_merged.cpp
/home/pschatzmann/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/1481549484/sketch_merged.cpp
Compiling sketch...
/home/pschatzmann/.arduino15/packages/esp32/tools/esp-x32/2302/bin/xtensa-esp32-elf-g++ -MMD -c @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/cpp_flags -Os -Werror=return-type -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/defines -I/tmp/.arduinoIDE-unsaved20241017-528943-1wbxl6m.rs9d/streams-audiokit-fft-led -iprefix /home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/include/ @/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/flags/includes -I/home/pschatzmann/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32/qio_qspi/include -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32 -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/esp32 -I/home/pschatzmann/Development/Arduino/libraries/arduino-audio-tools/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/WiFi/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/Network/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/NetworkClientSecure/src -I/home/pschatzmann/Development/Arduino/libraries/arduino-audio-driver/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/SPI/src -I/home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/Wire/src -I/home/pschatzmann/Development/Arduino/libraries/FastLED/src @/tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/build_opt.h @/tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/file_opts /tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/sketch/streams-audiokit-fft-led.ino.cpp -o /tmp/arduino/sketches/5B1CFBB6EA4C4CF21DFE74BC2518AC3A/sketch/streams-audiokit-fft-led.ino.cpp.o
[... cut ...]
Using library audio-driver at version 0.1.0 in folder: /home/pschatzmann/Development/Arduino/libraries/arduino-audio-driver 
Using library SPI at version 3.0.7 in folder: /home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/SPI 
Using library Wire at version 3.0.7 in folder: /home/pschatzmann/.arduino15/packages/esp32/hardware/esp32/3.0.7/libraries/Wire 
Using library FastLED at version 3.9.3 in folder: /home/pschatzmann/Development/Arduino/libraries/FastLED 
exit status 1

Compilation error: exit status 1
```

pschatzmann, 2024-11-17:
Here is how I usually design my Arduino libraries to avoid conflics
- I am putting all my classes in a namespace
- I  do not put any classes directly under src: this is asking for trouble
- I am adding a (configurable) using namespace so that the Arduino Sketches do not need to specify a namespece but can specify the namespace when there is a conflict.

zackees, 2024-11-18:
By any chance, does your *.ino file contain a header include for all your dependencies?

A lot of this template stuff I fixed this last week. `gcc` is getting confused because of it's usage of two-phase lookup for templates which causes it to include invalid templates for consideration even though the types don't match. I had to force two phase lookup in one pass via that `template_magic.h` header.

This will go into the next release. But it's available on master right now if you want to drop the fastled library into your project.

Namespaces kinda help, but the real issue I suspect is that you haven't put your project into cpp files and therefore are including the whole world in the header includes. If updating the FastLED library continues to give you issues then I suggest you put the audio portion of your project into cpp files and minimize the includes in the associated header. Alternatively you can put FastLED into a cpp file and only expose crgb.h into your ino file in order to define the led block.

For example, I typically put fastled related stuff into `gfx.h`  which will have `gfx_setup()` and `gfx_draw()` which is a function that will invoke `FastLED.show()`.

The unfortunate thing about FastLED is that it once was a header only library and so it wants to pull in the whole world when it compiles and you get these typical namespace collisions.

closed 2024-11-18 by zackees

pschatzmann, 2024-11-18:
Thanks, I also found the work-around that I can prevent the issue by including FastLED first in the sketch.