FastLED/FastLED #1777: Number of Controllers limit of 8

JohnSnorkel, 2024-11-17:
Version 3.9.0 on will only support a maximum of 8 controllers.  I have 12 controllers and with  release 3.9.x only any 8 of the twelve work.  It compiles but does not drive above 8 arduino digital i/o pins.  This is true on a Uno, Leonardo, Nano & Mega 2560 processor boards.

zackees, 2024-11-18:
The number of controllers is bumped up to 16 for Leonardo and Mega. I'm not going to increase it for the Uno or the nano because these devices are very memory constrained and we have users that are at the absolute limits of their memory capacity.

You can either download the fastled lib directly into your sketch or hack the defines through global project settings to get this defines into the build system. These are not include level defines. If you try to use them as such they will be ignored.

The define you will override will be `MAX_CLED_CONTROLLERS`, which defaults to 8 on these platforms.

The build system defined will look like this `-DMAX_CLED_CONTROLLERS=16`

This change will go into the next release.

```C++
#ifndef MAX_CLED_CONTROLLERS
#ifdef __AVR__
// if mega or leonardo, allow more controllers
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega32U4__)
#define MAX_CLED_CONTROLLERS 16
#else
#define MAX_CLED_CONTROLLERS 8
#endif
#else
#define MAX_CLED_CONTROLLERS 64
#endif  // __AVR__
#endif  // MAX_CLED_CONTROLLERS
```

JohnSnorkel, 2024-11-18:
Great.  I agree with you re not for the uni or nano.  I only used these to
test with minimal leds define per controller.
Thanks