FastLED/FastLED #1780: FastLED generating ESP32 code but compiling for Mega2560

kbastronomics, 2024-11-23:
I'm having an issue compiling with FastLED and a target of mega2560
I am using platformIO as my build environment.

this is my platformio.ini contents
[platformio]
default_envs = Mega2560-Release

[env:Mega2560-Release]
platform = atmelavr
board = ATmega2560
framework = arduino
monitor_speed = 250000
monitor_echo = yes

i have FastLED as a directory in the lib path not managed via platformio (but get same results if I do)
FastLED Version is 3.9.3

when I compile I get the following

```
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\js.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\timer.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\button.cpp.o
Assembler messages:
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\checkbox.cpp.o
Fatal error: can't create .pio\build\Mega2560-Release\libcda\FastLED\platforms\esp\32\experimental\s3_clockless_and_clocked_driver.cpp.o: No such file or directory
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\description.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\events.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\setup_and_loop.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\slider.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\title.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\ui_internal.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\ui_manager.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\platforms\wasm\ui\ui_number_field.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\power_mgt.cpp.o
Compiling .pio\build\Mega2560-Release\libcda\FastLED\ref.cpp.o
*** [.pio\build\Mega2560-Release\libcda\FastLED\platforms\esp\32\experimental\s3_clockless_and_clocked_driver.cpp.o] Error 1
============================================================================================ [FAILED] Took 4.69 seconds ============================================================================================
```

I have no references to wasm or esp anywhere.
i've done full clean of my build enviroment and no help.

If I change my board/platform to say a Pro-micro is compiles with no issues
only the Mega2560 is experiencing this

nomakewan, 2024-11-28:
Hey there,

We use FastLED with a Mega 2560 and cannot replicate your bug report. Here is one of our platformio.ini files:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_deps =
    gpstar81/GPStar Audio Serial Library@^1.2.0
    bakercp/CRC32@^2.0.0
    fastled/FastLED@^3.9.3
    avandalen/Switch@^1.2.7
    powerbroker2/SafeString@^4.1.35
    powerbroker2/SerialTransfer@^3.1.3
    arduinogetstarted/ezButton@^1.0.6
    arminjo/digitalWriteFast@^1.2.1
    siteswapjuggler/Ramp@^0.6.2
    flav1972/ArduinoINA219@^1.1.1
monitor_speed=9600
monitor_filters =
    time     ; Add timestamp with milliseconds for each new line
    default  ; Remove typical terminal control codes from input
upload_speed=115200
upload_port=/dev/cu.usbserial-A*
```

Maybe something useful in there for you to try?

kbastronomics, 2024-11-28:
The only real difference I see is the board name you use.  I'll try that and see if it helps

kbastronomics, 2024-11-28:
ok, just tried that one change and that fixed it.
I'm not sure what the difference off hand is between megaatmega2560 and atmega2560 on the board but it's compiling clean now,  thank you

closed 2024-11-28 by kbastronomics