FastLED/FastLED #1781: ESP8266 D1 Mini Crash while wifi Activate

dovevgo, 2024-11-25:
From version 3.9.0 and up, 
When using ESP8266WiFi.h with FastLED.h, after few seconds from connection, the device crushes.

3:44:26.949 -> *wm:Starting Web Portal 
13:44:29.608 -> 
13:44:29.608 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
13:44:29.639 -> 
13:44:29.639 -> Soft WDT reset
13:44:29.639 -> 
13:44:29.639 -> Exception (4):
13:44:29.639 -> epc1=0x4020f6b0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
13:44:29.639 -> 
13:44:29.639 -> >>>stack>>>
13:44:29.639 -> 
13:44:29.639 -> ctx: cont

zackees, 2024-12-10:
What happens when you take out the FastLED include and disable all the FastLED code in your sketch?

The `wdt` stands for "watch dog timer" and if your CPU locks up then the watch dog timer will detect that and issue an MCU reset. So something is locking up your device. There is like one other bug report about this, so I'm not sure this is a FastLED issue or something that you are doing wrong.

https://github.com/FastLED/FastLED/issues/762

zackees, 2024-12-10:
Yeah, this looks like a bug in the esp firmware. Other users report this happens with incorrect wifi credentials. I don't think this is a FastLED issue per se so I'm closing it. Feel free to re-open if you have further information about it.

closed 2024-12-18 by zackees