FastLED/FastLED #1782: Compilation failure when FASTLED_ALL_PINS_HARDWARE_SPI is used with ESP32-S2

DavidHaitch, 2024-11-25:
Compiling a FastLED project with the FASTLED_ALL_PINS_HARDWARE_SPI flag enabled while targeting an ESP32-S2 board results in the following compile error:
`
.pio/libdeps/adafruit_qtpy_esp32s2/FastLED/src/platforms/esp/32/fastspi_esp32.h:75:32: error: 'VSPI' was not declared in this scope`
`
The relevant code in fastspi_esp32.h is as follows:
`
#else // Configuration for other ESP32 variants
	#ifndef FASTLED_ESP32_SPI_BUS
	#pragma message "Setting ESP32 SPI bus to VSPI by default"
	#define FASTLED_ESP32_SPI_BUS VSPI
	#endif
#endif
`

Sure enough, the VSPI macro is only defined for boards that have `CONFIG_IDF_TARGET_ESP32` defined. When the default SPI interface is changed from VSPI to FSPI, compilation succeeds and I see an approximate 15% improvement in performance.

I don't have access to the full range of ESP32 boards, so I can't test any fixes for this, but my first thought would be to add an additional check to that define structure to handle the -S2 case.

DavidHaitch, 2024-11-25:
Addendum: I've discovered that this compilation failure can also be averted by setting `#define FASTLED_ESP32_SPI_BUS FSPI` before including FastLED. This does immediately unblock me, but I still think it would be preferable if the library defaulted to a safe SPI interface for the given hardware.

zackees, 2024-12-08:
Thank you, going in on the next revision.

closed 2024-12-08 by zackees