FastLED/FastLED #1783: Doesa not compile in ESPHome - Missing header - hal/gdma_types.h:

PaulSchulz, 2024-11-25:
Greetings,

I get a compile error when using ESPHome (a a custom component), essentially arduino/platformio, but when I use the Github main branch of FastLED directly, rather than the 'released; version.

The error:

FastLED/src/third_party/yves/I2SClocklessVirtualLedDriver/detail/env_s3.h:24:10: fatal error: hal/gdma_types.h: No such file or directory

Broken
Dependency Graph
|-- SPI @ 2.0.0
|-- Ticker @ 2.0.0
|-- SX126x-Arduino @ 2.0.27
|-- FastLED @ 3.9.4+sha.c4987fb

Fixed
Dependency Graph
|-- SPI @ 2.0.0
|-- Ticker @ 2.0.0
|-- SX126x-Arduino @ 2.0.27
|-- FastLED @ 3.9.4

I think ESPHome gets it's libraries via platformio. The difference in my compilation is to use:
  # broken
  libraries:
    - "SPI"
    - "Ticker"
    - "SX126x-Arduino"
    - "FastLED=FastLED=https://github.com/FastLED/FastLED"

  # working
  libraries:
    - "SPI"
    - "Ticker"
    - "SX126x-Arduino"
    - "FastLED"

PaulSchulz, 2024-11-25:
The compile error occurs between the following commits:
- [x] 10dca0518415dccae45bdfafb26dd9acc94edd2b - N
- [x] 01005e48a186cc2780af6bac70d0814686282a89 - Y

https://github.com/FastLED/FastLED/commit/10dca0518415dccae45bdfafb26dd9acc94edd2b

zackees, 2024-12-08:
Thanks, I'll take care of it now.

closed 2024-12-08 by zackees