FastLED/FastLED #1784: Multiple controller example not working - Wrong Interface

NikBomb, 2024-11-27:
The multiple controller [examples](https://github.com/fastled/fastled/wiki/multiple-controller-examples) suggest to use the `showLeds` method. 

The interface implements only a `showLedsInternal` .   

[Here ](https://wokwi.com/projects/415534419404525569) a code that shows the with the issue.

[Here ](https://www.reddit.com/r/FastLED/comments/1h0br5r/comment/lz4l3zs/) the reddit thread that sparked this issue.

PS: I am happy to contribute to the project, and help solve this :)

nomakewan, 2024-11-28:
Had a quick peek and can confirm that the function suggested in the example was at some point renamed to `void showLedsInternal(uint8_t brightness)`.

Unfortunately there is no way to make a PR against the wiki; it must be manually updated by someone with write access to this repo.

NikBomb, 2024-11-28:
Is this only a documentation problem?

If the methods have been renamed and the API has changed this should become a new version of FastLED according to semantic versioning rules.

Also the suffix `Internal` suggests that these methods should not be used by users?

nomakewan, 2024-11-28:
It is a documentation problem. The methods were renamed in #1733 which became a part of FastLED 3.9.0 (the release prior was 3.7.8).

The suffix `internal` in this case appears to refer to the fact that you are controlling the internal CLEDController `show` rather than using the global version. However, I did not write this code, so I do not know the actual reasoning for the change in name.

dmadison, 2024-12-03:
Changing a public function name is an API change and should only be done with a corresponding major version release. This is not a documentation problem, it breaks a bunch of code that *should* still work with version 3.x of the library.

I don't understand why the names of public and often-used functions are being changed on a whim and without discussion.

dmadison, 2024-12-03:
It looks like there's a functional change in #1733 too: `CLEDController::clearLeds(int)` no longer clears the LEDs, just zeroes the data.

dmadison, 2024-12-03:
This isn't simple to revert either, because the protected function signatures with the same names were changed as well, along with the associated `CPixelLEDController` functions and how data is passed between them. Many of the commits contain multiple logical changes spread across several files. Someone is going to have to untangle the signature changes from the functional rewrites and then check that behavior is the same as previous versions.

Why are these changes on the master branch? Any changes that affect the API or change existing functionality should be on a development branch and deferred until the next major release.

zackees, 2024-12-05:
> Changing a public function name is an API change and should only be done with a corresponding major version release. This is not a documentation problem, it breaks a bunch of code that _should_ [...]

3.9 is the alpha version of 4.0. We've been announcing this on the reddit page for quite some time. Yes, we have some minor API-breaking changes in this alpha stage.

We had to break showLeds() on the individual controller strips because we are making FastLED highly parallel. Unfortunately the 3.8.X contract was that `showLeds()` would complete the entire draw before it returned. This made sense when everything was software bit banged out via the CPU circa 2012. That's all changed now with the higher end chips which feature multiple independent hardware bit banging backed by a DMA controller. What this means is that `showLeds()` has been transformed from "draw this right now" to "acquire the resources and prepare for an impending async draw command".

When that command is issued, all strip controllers start banging out the colors at the same time.

As such, `showLeds()` on a given controller should not be used. What it actually does varies from platform to platform. On the UNO and other avr platforms it will do the same thing as always. On ESP32 the draw command won't actually be issued until `endDraw()` is invoked and will wait to complete once `beginDraw()` is called on the next frame. This pattern will be replicated for the Teensy40/41 series in the next release, or the one after.

As far as code stability, stick to the releases. Don't point to the master branch. We chosen to do this this way because of the sheer number of builders we have to ensure code quality and compatibility and I don't want to exceed our limit with github. Keeping everything on the master branch also means we can get rapid feedback from the users the want the latest always, and it also makes merging PR's easier since it's always one branch people are forking from.

closed 2024-12-08 by zackees

nomakewan, 2024-12-08:
@zackees I see you updated the wiki, but you only corrected the function names for 3 of the 4 devices. `controllers[3]` is still using the incorrect `showLeds` method. Just a heads up.

dmadison, 2024-12-08:
>3.9 is the alpha version of 4.0. We've been announcing this on the reddit page for quite some time.

Respectfully Zach, that's not how that works. It's a violation of semantic versioning to make breaking API changes, no matter how minor, without a corresponding major release. 3.9 is now objectively not a 3.x version of the library, as code written for previous 3.x versions is no longer compatible. That is confusing for both users and developers.

Either the API changes should have been merged to a development branch and deferred until the next major release, or a major release should have been done. Now we're in a strange liminal space where we're no longer on 3.x still not on 4.x, but still calling it 3.x. And a significant amount of user code and 3rd party libraries (including my own) are broken, with seemingly no plan to fix compatibility with the (still current) API?

>As such, showLeds() on a given controller should not be used

That is not a minor breaking change. Controllers are part of the public API, and there is a lot of user code that relies on managing individual strips via their controllers, *not* through the `CFastLED` instance. There are plenty of situations when you don't want to update all strips, regardless of performance. That's why controller references are accessible by the user. If the function still exists then the existing serial / blocking behavior of that function should be maintained and the new parallel functionality should be implemented alongside it.

zackees, 2024-12-13:
What's broken? This is the first bug I've seen from you.

3.9.x doubled the frame rate for cpu intensive scenes with a combination of the async draw on esp32 and also WS2812 over clocking. I also fixed RMT4 boot crash that affected every single led driver in this space.

I chose this path deliberately. The community has responded so far by doubling the velocity of stars on our repo.

And what you are suggestion is actually harmful.

Case in point:

Espressif followed the methodology you are suggesting and absolutely unleashed hell via IDF v5.1 that was forced on everyone with the arduino 3.0 core update.

Every single led driver in the open source community was broken by it. 

Every.

Single.

One.

I absolutely refuse to do that to the users of fastled. They deserve better than to have all the breaking changes silod away and them dumped on them in one update.

dmadison, 2024-12-13:
>They deserve better than to have all the breaking changes silod away and them dumped on them in one update.

Zach, you are not a novice developer. I find it hard to believe you don't understand how semantic versioning works.

The purpose of a major release is not to "dump" changes on users, it's to create a clean break of compatibility. Users of the old API can continue using the old API, while other users can choose to migrate to the new API and get all of the new features therein. No one is "forced" to update the library. You can select what version of the library you want from within the IDE.

The entire *purpose* of semantic versioning is so that users can understand, at a glance, what changed. A patch increment is a bugfix. A minor version is a new feature. A major version is a significant rewrite, causing the public API to no longer be compatible with older user code.

Ideally you would add new functionality, mark the previous functions as deprecated, allow users to integrate the new functions at their own pace, and then remove them after a significantly long grace period (at which point you would then push a new major release).

I do not understand why you would choose to deliberately remove and/or change the old functionality and break user code. Why not create a major release? Or a major release candidate? Why not defer the API changes *for* a major release? Why not add the new features *alongside* the old functions, maintaining compatibility with existing user code?

>And what you are suggestion is actually harmful.

Maintaining compatibility with user code is harmful? What?

dmadison, 2024-12-13:
>And what exactly is broken? This is the first bug report Ive seen you in the conversation and you aren't even the op.

From my previous comments in this issue:
* The public function `CLEDController::showLeds()` has been changed to `CLEDController::showLedsInternal()`, and the signatures of the related "show LEDs" controller functions have also changed
* `CLEDController::clearLeds(int)` no longer sends a 'show' update to the LEDs
* The protected function signatures of `CPixelLEDController::showColor()` and `CPixelLEDController::show()` have changed  

The first is OP's issue, the others are ones I discovered while trying to fix OP's issue. There very well may be more changes to the API which break compatibility with version 3 user code, these are just what I noticed at a glance.

Per your comment earlier in the thread, it is apparently not possible to refactor user code that references `showLeds()` to use `showLedsInternal()`, as the behavior of the function itself has changed. That alone renders much user code that uses controller objects directly obsolete.

I'm not sure how it's relevant that I'm not the OP, but if you'd prefer I can submit my own issue on this topic.

zackees, 2024-12-13:
I'll look into the clear operation. From what I recall from two months ago is that it invoked the templated show() function and a massive amount of code was being injected into the binary that was showing up in the profiler. I changed it during a binary reduction phase for 3.9 because the 3.8.X series started breaking product makers using the attiny platform with sketches that were at the absolute limits of the platform's flash capacity.

3.9 is ~200 bytes less that 3.8. Fun fact, putting a virtual destructor on CLEDController will transitive pull in malloc and free and add about 1k of binary to the flash and blow out the avr people.

The show internal will have to stay as is because of the parallel nature of the driver and the use of async draw now.

I understand you want to selectively disable controllers. After our conversation last week I went ahead and implemented CLEDController::setEnabled(bool) which will do what you want I believe. It's still just on master, and will go in on the next release. If this does what you want, could you please check out this feature in master and let me know  if this works for you.

zackees, 2024-12-13:
Yes, I understand semantic versioning.

I'm deliberately not following it for 3.9 because it doesn't work for us.

We have a gigantic number of platforms and an odd combination of power users with sub $1 micros. Releasing fast means that when I break someone somewhere, in a place I didn't expect, I can get feedback immediately and have it fixed in the next release while my mind is still fresh.

If all these little subtle breakages i was able to fix were all compacted into one event it would create a crisis much like python 2->3 or idf 5.1 did to arduino users. And the cherry on top is that without fast feedback i can venture down a path that ossifies and then I CANT revert.

So going off strict semantic versioning for 3.9 was a conscious decision 

Fixing show() on individual controllers actually can be done in a way to restore the issue above. An inline function which calls beginDraw() showInternal() endDraw(). But the draw will not be async, like the previous behavior.

zackees, 2024-12-14:
The reason I took out show() from clearLeds() is that it adds ~200 bytes to the binary. It looks like if the function is simply not virtual then we maintain the same size and can call showLeds().

```
no virtual / no showLeds() : 10441
virtual / no showLeds() : 10621
no virtual / showLeds() : 10441
virtual / showLeds() : 10838
```

I've created a macro called `VIRTUAL_IF_NOT_AVR` which will add 200 bytes to all the other platforms, but will keep avr the same.

nomakewan, 2024-12-14:
I think what he's getting at, just to be perfectly clear, is that people who were using the function called `CLEDController::clearLeds(int)` in their code and had written code specifically understanding that it has always resulted in the physical LEDs attached to that controller actually being cleared in real life are now being blindsided by a completely undocumented change to the API that switches `CLEDController::clearLeds(int)` from physically clearly the LEDs to merely clearing the LED color data in memory, which is a massive change in output functionality.

It's not about memory size difference, it's about the fact that a function with an identical name to the previous function now performs a completely different operation in the real world.

zackees, 2024-12-15:
The bug is not the issue. I'm willing to fix a bug and did so.

The issue is his lecturing me on not going with strict semantic versioning for the 3.9.X series, as if I don't know what I'm doing and he does.

dmadison, 2024-12-15:
>After our conversation last week I went ahead and implemented CLEDController::setEnabled(bool) which will do what you want I believe.

That sounds like a great idea for parallel strips. Does the legacy blocking `CLEDController:showLeds()` work independent of that setting?

>Yes, I understand semantic versioning. I'm deliberately not following it for 3.9 because it doesn't work for us... So going off strict semantic versioning for 3.9 was a conscious decision

That's concerning. You can release fast and still keep with semantic versioning, they're not mutually exclusive. Calling 3.9 "4.0 rc1" would have been enough. Now the repository is in a messy state where code from 3.7 functions differently from code for 3.9, and users looking to update their library to the latest version that is compatible with their code (both now and in the future) are going to be confused.

Ironically, you releasing version 3.9 and **not** 4.0 is why I am commenting here nearly 3 months after these commits were pushed to master. I had no reason to think the API would have changed, because you implicitly told me that it hadn't.

>The bug is not the issue. I'm willing to fix the bug and did so.

The "bug" that I'm referencing **is** that the library is not following semantic versioning. All the other issues are symptoms of that. If any of those functions had changed or been removed with a major release that formally severed the connection to all previous user code then none of this would matter.

As I said, I haven't gone through the commit history to verify that the current codebase is compliant with the 3.7 API but I would not be surprised if there were other discrepancies. I don't think I'll be able to find the time before the holiday, but this would be a good target for a unit test.