FastLED/FastLED #1786: Bug: LEDs on different strips are always the same color

thomergil, 2024-12-10:
See below. There are two strips, one LED each. The code sets them to two different colors, but they are always the same color. This is on an ESP32-S3 Zero.

```
#include <FastLED.h>

#define NUM_LEDS 1

CRGB led1[NUM_LEDS];
CRGB led2[NUM_LEDS];

void setup() {
    FastLED.addLeds<WS2812B, 7, GRB>(led1, NUM_LEDS);
    FastLED.addLeds<WS2812B, 8, GRB>(led2, NUM_LEDS);

    led1[0] = CRGB::Red;
    led2[0] = CRGB::Blue;
    FastLED.show();
}

void loop() {
    // Empty - just demonstrating initial state bug
}
```

nomakewan, 2024-12-14:
Potentially directly related to #1761 and #1774 since I see you are only defining 1 LED per pin?

zackees, 2024-12-20:
Okay, 3.9.6 had a mistake in it where the RMT recycle value was inverted. I've fixed it in 3.9.7 which will end up in the Arduino IDE repo shortly. Feel free to install it now from the zip file to try it immediatly.

zackees, 2024-12-23:
Bad news: Disabling RMT recycle broke our multi strip case on ESP32.

Good news: Espressif got back to me and said that this bug is a manifestation of the gpio state being reset after the rmt disable.

See: https://github.com/espressif/esp-idf/issues/15049

They recommended that I set the INPUT_PULLDOWN on the pin right after the rmt_disable call, to prevent the floating HIGH state on the pin. I've added this and also a micro delay of 10 microseconds to make sure that the RMT blip, if it's still happening, will now only be momentary and also the 10 microsecond delay will prevent the now shortened blip from being interpreted as a 1 or a 0.

## What I'm going to do:

  * I'm applying the espressif fix
  * I'm re-enabling RMT recycling for the next release.

## What I need from you

If nothing is done, the rmt recycling will be re-enabled on the next release. I *hope* this won't break you, but there's a small possibility that you'll get a regression back to the green led being stuck on.

However, if someone tests the updated branch rmt-fix https://github.com/FastLED/FastLED/tree/fix-rmt and let me know that it continues to work for you, we can avoid this inconvenience for you. This branch has both the espressif fix and also RMT recycling re-enabled.

If...
  * The green light is stuck on: then I will let espressif know and revert the change.
  * If the green light continues to be fixed: Then a release will be minted within 24 hours.
  * If no feedback is given, then the release will be minted by the 28th, and the chips will fall where they lay. If this breaks you then you will have to apply the workaround until the next release.
  
I'm flying blind here since I can't reproduce your issue locally. So thanks in advance, for your help.

thomergil, 2024-12-24:
Will do, but in ~2 weeks. Thanks for looking into it further.

zackees, 2024-12-29:
3.9.8 is released. Should fix the issue. We are using a new non-recycling RMT driver.

closed 2024-12-29 by zackees