FastLED/FastLED #1789: Controller show() function signature changed

dmadison, 2024-12-15:
The function `void CLEDController::show(const struct CRGB *data, int nLeds, uint8_t brightness)` is now protected and abstract, where it was previously public and not abstract. This breaks compatibility with 3.x user code that used this function to drive LEDs.

zackees, 2024-12-16:
These were changed a long time ago and not by me.

Also the CRGB scale doesn't make sense anymore unless there is an additional brightness factor. 3.9.X added support for high definition color component chipsets, of which the APA102/SK9822 series now uses.

In this mode, color correction scale is separated from the brightness in order to do pseudo 13-bit color output for the APA102HD/SK9822HD controller types, which use this feature to do driver level gamma correction. In the future, other high resolution chipsets will use a similar mechanism. This can only be optimally done with these color/brightness components being seperate. Prior to 3.9.X, these two components were pre-mixed, which decimated the color resolution. This is fine of course for the RGB8 controller types like the WS2812, but anything higher than 8-bit resolution like the APA102 would result in lossy color decimation at the low end of the color brightness spectrum.

For example consider the color correction which attenuates green and red: (172, 128, 255), which is premixed with a global brightness factor of 16. This would result in a final CRGB scale of (10, 8, 16) multiplied against each color. This will result in the APA102 getting it's low order bits destroyed before it can trade per-led brightness deboosting with per-led-component boosting. With this CRGB scale and brightness being separate we get preserved CRGB scaling and defer the brightness correction to the driver level. It's more complicated than this, but that's the gist of it.

In APA102-non-HD mode, this global brightness (which is the same for every led in the strip) is used for the 5-bit brightness, with the color scale being applied to each of the components. A massive increase in color resolution on the low end spectrum (right were everyone cares the most) was the result.

Given that there has been no complaints in this part of the code base since it's release, I'm going to assume this is the new API unless someone comes to me with a really good reason for why this should be put back in. Manually handling controllers outside of FastLED.show() is becoming more deprecated by the day, especially with the introduction of the CLEDController::setEnabled(bool) function that was put in this release to shut off strips explicitly (which is what this feature does anyway) and the use of bulk strips updating in parallel. I really don't want to have additional paths that users can take to make their strips light up, outside of FastLED.show().  The more we increase parallel async drawing (the future of FastLED) the more FastLED.show() get's complicated. I don't want users to try and take control of the strips themselves. It's really a maintenance nightmare. The big features that the power users want is multiple controller types assigned to the same pins so that apps can switch between WS2812/APA102/Whatever at runtime. Right now this use case is undefined for any given platform. It may work or it may not. What we need for the controller types is the ability to safely run their destructors so that the SPI/Whatever resources can be released. I know that this is NOT the case with Arduino which will immediately crash if multiple resources share the same pin via an assert they force.

closed 2024-12-16 by zackees

dmadison, 2024-12-17:
Thank you for the explanation on the color correction scaling, that makes a lot of sense. It's a pity that the previous API can't be retained for compatibility reasons, even if the end result is worse.

>The more we increase parallel async drawing (the future of FastLED) the more FastLED.show() get's complicated. I don't want users to try and take control of the strips themselves.

This is where you lose me. There are many setups where you *can't* have parallel async drawing due to the way the hardware is configured, where trying to force parallel drawing when it's not needed will instead create massive slowdowns from redundant data.

The simplest example is with any setup that has multiple LED controllers operating on a shared bus, whether that's UART, I2C, DMX, CAN, what-have-you. If a single group is being driven at a high refresh rate, the bus may saturate from the data being pushed from the other (unchanging) groups, bringing the total framerate down. In which case being able to control one group rather than all of them is preferred.

Using `setEnabled(bool)` in that context seems backwards, because instead of driving one instance you have to micromanage disabling all of the others. If nothing else that's a violation of encapsulation, since in large projects you may not be the one coding all of the strips.

>Manually handling controllers outside of FastLED.show() is becoming more deprecated by the day, especially with the introduction of the CLEDController::setEnabled(bool)

I want to be able to fix my library and a number of other animations for future compatibility, so I want to make sure I understand this properly.

Previously, the API for sending an arbitrary array of LED data to a controller output was as follows:

```cpp
void show(CRGB* leds, int numLeds, uint8_t brightness) {
    controller.showLeds(leds, numLEDs, brightness);
}
```

The public functions for driving LEDs directly through the controller are deprecated, so we must use the `CFastLED` singleton for managing them. That means the API is now... this?

```cpp
template<class TYPE, uint8_t DATA_PIN, EOrder COLOR_ORDER>
void show(CRGB* leds, int numLeds, uint8_t brightness) {
	// there is no way to send another LED buffer to a controller,
	// so we must declare another one and pass the parameters in
	// (this requires the template params as well, unfortunately)
	CLEDController& controller = FastLED.addLeds<TYPE, DATA_PIN, COLOR_ORDER>(leds, numLeds);
	
	// we don't want to write to any other controllers, so we need to
	// buffer their 'enabled' state to avoid conflicts
	const int NumControllers = FastLED.count(); 
	bool* enableBuffer = new bool[NumControllers];
	
	for(int i = 0; i < NumControllers; ++i) {
		CLEDController& con = FastLED[i];  // so we don't have to iterate twice
		enableBuffer[i] = con.getEnabled();  // I don't think this function exists yet
		con.setEnabled(false);  // so we don't write to this group
	}
	controller.setEnabled(true);  // enable our new controller only
	
	// buffer existing brightness
	const uint8_t pBrightness = FastLED.getBrightness();
	FastLED.setBrightness(brightness);
	
	// finally show the LEDs
	FastLED.show();
	
	// reverse everything
	FastLED.setBrightness(pBrightness);
	
	for(int i = 0; i < NumControllers; ++i) {
		CLEDController& con = FastLED[i];
		con.setEnabled(enableBuffer[i]);
	}
	
	delete [] enableBuffer;
}
```

Is there something I'm missing? That seems needlessly complicated. It also requires templating every function that previously dealt with a controller pointer, which is... blech

If you're committed to deprecating the controller API, maybe something like `setExclusive(bool)` would be a good answer? Although I'd imagine we'd have to iterate the linked list twice on every update.

That also doesn't solve the issue of passing in arbitrary LED arrays, which is a big performance saver for animations. I'm not sure if there are any side effects of setting the member pointer, although again that's a store/set/reset per frame which seems needless.

zackees, 2024-12-17:
> This is where you lose me. There are many setups where you can't have parallel async drawing due to the way the hardware is configured, where trying to force parallel drawing when it's not needed will instead create massive slowdowns from redundant data.

The architecture has to accommodate parallel async. That's the constraint that make this complicated. Sync behavior won't slow down at all.

> may saturate from the data being pushed from the other (unchanging) groups, bringing the total framerate down. In which case being able to control one group rather than all of them is preferred.

That's a user problem. They should call CLEDControlller::setEnabled(false). Anyone who's proficient enough to do multiple strips can call setEnabled(false).

> Using setEnabled(bool) in that context seems backwards, because instead of driving one instance you have to micromanage disabling all of the others. If nothing else that's a violation of encapsulation, since in large projects you may not be the one coding all of the strips.

You either have flexibility or performance. Choose 1. That's always the tradeoff and my personal roadmap includes 24 way I2S ESP32 hardware bitbanging, and 42 way DMA bitbanging on Teensy40/41. The point of a good API is that the proper abstraction is selected which allows for performance when in matters and is maximally simple. Individual controllers being controlled by users willy nilly contradicts massive performance opportunities where all the pins can prepare their resources and draw together. If you want to get a scope of the complexity, go look at Yvez trick to get I2S to work. You absolutely cannot drive I2S controlled independently. All the data has to be drawn together. You can't even have different Clockless controller timings. The only way that's possible is by grouping similar chipset types into a batch.

zackees, 2024-12-17:
> Is there something I'm missing? That seems needlessly complicated. It also requires templating every function that previously dealt with a controller pointer, which is... blech

Yes, global brightness is a hack. It needs to be made individually to every single controller, which they already do during drawing. Eventually global brightness should be per strip, not the whole program.

zackees, 2024-12-17:
> That also doesn't solve the issue of passing in arbitrary LED arrays, which is a big performance saver for animations. I'm not sure if there are any side effects of setting the member pointer, although again that's a store/set/reset per frame which seems needless.

I am perfectly ok with someone setting arbitrary CRGB arrays in the LED strips. Direct memory access was a relic of older chips that didn't have much memory. I2S and RMT mangle the CRGB data in a side buffer now anyway. The future is oodles of memory. The bottle neck is the degree things can go parallel.

dmadison, 2024-12-17:
>That's a user problem. They should call CLEDControlller::setEnabled(false). Anyone who's proficient enough to do multiple strips can call setEnabled(false).

You misunderstand my point. It's not that calling the function is complicated, it's that control of writing *other* strips is delegated to someone trying to write one. It encourages poor encapsulation, because now if someone else mismanages their controllers then I can't manage mine.

>You either have flexibility or performance. Choose 1

>individual controllers being controlled by users willy nilly contradicts massive performance opportunities where all the pins can prepare their resources and draw together. If you want to get a scope of the complexity, go look at Yvez trick to get I2S to work. You absolutely cannot drive I2S controllers independently

Then why isn't the focus on I2S and DMA bitbanging being the components driven in parallel, rather than forcing everything down that path? Why not implement parallel output at the controller level (which it seems you're doing?) with a corresponding serial function to invoke it, rather than forcing all non-parallel controllers to pretend to be?

At least from where I'm standing, it would seem a much simpler and more straightforward solution to implement parallel output as its own controller class with a separate API and separate linked list, only accessible via the global singleton. You'd get the best of both worlds: massive performance bonuses for platforms that support it, *and* fine control by the user for explicitly synchronous interfaces.

>Yes, global brightness is a hack.

>I am perfectly ok with someone setting arbitrary CRGB arrays in the LED strips

I don't understand this. You agree that global brightness is a hack and you think we should allow arbitrary arrays to be passed, but you don't want to reinstate the function that does both? Do you have another implementation in mind?

>The future is oodles of memory.

Of course, but we're not in the future, we're trying to build things today. As you were quick to point out yourself, there are people building projects with ATtiny85's! Squeezing memory performance is still important for many applications.

zackees, 2024-12-18:
> . It's not that calling the function is complicated, it's that control of writing other strips is delegated to someone trying to write one. It encourages poor encapsulation, because now if someone else mismanages their controllers then I can't manage mine.

You can either do it the FastLED way which is continuously tested or go into the wild west of undefined behavior for a given platform. If there is a common use case that's not supported, we should support it, because we can test it and prevent regressions.

> Then why isn't the focus on I2S and DMA bitbanging being the components driven in parallel, rather than forcing everything down that path? Why not implement parallel output at the controller level [...]

I am focusing on I2S and DMA. I've already ported ObjectFLED and there's a test for it. I've committed a CL today to support I2S.  It's already in the code base. I don't get your use case of controlling your own controllers manually. Is this your use case? No one has ever filed a bug report since I've been here saying "we take control of our controllers manually". You are arguing for a hypothetical that's not in demand. I just fixed a bug for RMT which is massively parallel. There's a LOT of people using this now.

> At least from where I'm standing, it would seem a much simpler and more straightforward solution to implement parallel output as its own controller class with a separate API and separate linked [...]

This is the best idea you've had so far in this conversation. Say more.

> I don't understand this. You agree that global brightness is a hack and you think we should allow arbitrary arrays to be passed, but you don't want to reinstate the function that does both? Do you have another implementation in mind?

Yes, the implementation would be similar to how an RGBW object is an attribute in CLEDController. For example, show doesn't include the RGBW object, it's implied based on the attributes of the strip object itself.

> Of course, but we're not in the future, we're trying to build things today.

Yeah and it's already complicated. Hence the reason I test the shit out of it, including binary size tests that run on EVERY CL. But I'm willing to do this because it's popular. If controlling controllers manually was just as popular, then this would be an entirely different conversation.

> As you were quick to point out yourself, there are people building projects with ATtiny85's! Squeezing memory performance is still important for many applications.

Yeah, and that's a path i explicitly support because, again, its incredibly popular. And I've never seen anyone take manual control of their controllers. They all add it to the FastLED and off they go.

dmadison, 2024-12-19:
>You can either do it the FastLED way which is continuously tested or go into the wild west of undefined behavior for a given platform

The irony here is that you are removing "the FastLED way" as an option. If you are unwilling to fix this issue and restore the previous behavior, my only way forward *is* to go "into the wild west" and roll my own solution that is not officially supported and not continuously tested with the rest of the library.

>I don't get your use case of controlling your own controllers manually. Is this your use case? No one has ever filed a bug report since I've been here saying "we take control of our controllers manually". You are arguing for a hypothetical that's not in demand.

Yes, this is my use case. Yes, this is other's use cases as well. You are literally commenting on a bug report about this very issue, linked to a library that is now broken because the feature was removed, after you closed a previous issue opened by someone else also about one of these missing functions. What are you talking about?

I'm also currently working on a project that uses a function (passing arbitrary CRGB pointers to the controller) which was removed, which negatively impacts performance. It's not the end of the world, but it's needlessly slower.

As I've said elsewhere, the only reason this was raised now and not 3 months ago was because semantic versioning was not followed, and I had no reason to suspect that any features were purposefully removed.

>And I've never seen anyone take manual control of their controllers.

Have you spent much time on the subreddit? The topic comes up surprisingly often, almost always as a solution to a problem rather than something people seek out to do. Reddit search is notoriously bad, but if you look up "`CLEDController`" via Google (`site:reddit.com`) you'll find a number of hits for both /r/FastLED and /r/Arduino.

Here are the first few results for me:
* [Is it possible to have FastLED change data pins?](https://www.reddit.com/r/FastLED/comments/x9j20d/is_it_possible_to_have_fastled_change_data_pins/)
* [Redefine FastLED.addLeds After Startup](https://www.reddit.com/r/FastLED/comments/16vn2uo/redefine_fastledaddleds_after_startup/)
* [How to get different brightness for different WS2812B and WS2813 chains on ESP32?](https://www.reddit.com/r/FastLED/comments/vgx72v/how_to_get_different_brightness_for_different/)
* [Non-global show function??](https://www.reddit.com/r/FastLED/comments/1afr2mm/nonglobal_show_function/)
* [How do I dynamically change the number of LEDs being updated.?](https://www.reddit.com/r/FastLED/comments/lmy59t/how_do_i_dynamically_change_the_number_of_leds/)

Previously, direct access to the controller objects allowed you to change the strip type or color order at runtime, avoid copying when displaying frame animations / moving patterns, and take fine control of the output in order to achieve higher performance with fewer writes.

All of those features are now removed or deprecated, and for the life of me I still do not see an upside to doing it this way. There is no reason the existing behavior had to be demolished to make way for parallel output, the two can coexist.

---

At this point I think we're going in circles. You've already closed this topic and I don't imagine you'll change your mind.

I'll keep the library issue open for posterity. If the API does ever stabilize it might be worth creating a companion library to reinstate some of the lost functionality. In the meantime I suppose I'll redirect users to the older releases if need-be.

zackees, 2024-12-20:
> I'm also currently working on a project that uses a function (passing arbitrary CRGB pointers to the controller) which was removed, which negatively impacts performance. It's not the end of the world, but it's needlessly slower.

I am 100% for this use case. Who removed it?

Look, if you want this feature, feel free to do the work and issue a PR. The big issue is that with RMT and ObjectFLED and I2S bulk controllers is that everything is going to get more complicated. We have an api designed for independent controllers. But the bulk stuff needs them to work together. To make this work correctly I put in the beginShow() and endShow() functions. This keeps the api the same but makes my job harder, but that's ok.

None of this stuff had any unit tests prior to me joining this project. We don't need to unit test everything, but simply to ensure the function signature stays compatible.

Until @sutaburosu mentioned all the uses cases this API actually had, I assumed it was pretty much unused. The use cases you have is important, but you mentioned it at the tail end of the conversation.

This bug was closed automatically because I did restore the show() part of the API on the individual controllers and it was challenging because it increased the size of the binary by 200 bytes and caused the AVR binary size tests to fail until came up with innovative ways to not have this happen.

The avr builds are sacred, and so any new features have to balanced against increasing the size of these builds. For example, the RGBW feature meant I had to go around the code base converting unnecessary uint32 calculations to equivalent uint16 calculations to recover code execution size on these 8 bit micros.

So feel free to add this yourself, write the simplest unit test to ensure the API remains stable, and we'll get it in.

You can run all the unit tests by going to the project root and invoking "./test"