FastLED/FastLED #1790: Version 3.7.8 is invalid

dmadison, 2024-12-15:
Release 3.7.8 of the library is not tracked to a commit that is part of the repository. This makes it impossible to check out this version of the library via the git tag.

The relevant commit is `a9fede2`, and the tag should be updated accordingly.

zackees, 2024-12-16:
Says it's tracked to `commit de02bc1829df15a13babe88f457e2316fb3e2e39 (tag: 3.7.8)`

```
commit de02bc1829df15a13babe88f457e2316fb3e2e39 (tag: 3.7.8)
Author: zackees <z@zackees.com>
Date:   Mon Sep 23 20:40:52 2024 -0700

    update fastled to 3.8

diff --git a/docs/Doxyfile b/docs/Doxyfile
index 293cc60b..076e2708 100644
--- a/docs/Doxyfile
+++ b/docs/Doxyfile
@@ -48,7 +48,7 @@ PROJECT_NAME           = FastLED
 # could be handy for archiving the generated documentation or if some version
 # control system is used.

-PROJECT_NUMBER         = 3.7.7
+PROJECT_NUMBER         = 3.7.8

 # Using the PROJECT_BRIEF tag one can provide an optional one line description
 # for a project that appears at the top of each page and should give 
viewer a
diff --git a/library.json b/library.json
index 66231788..cc142457 100644
--- a/library.json
+++ b/library.json
@@ -38,7 +38,7 @@
         "type": "git",
         "url": "https://github.com/FastLED/FastLED.git"
     },
-    "version": "3.7.7",
+    "version": "3.7.8",
     "license": "MIT",
     "homepage": "http://fastled.io",
     "frameworks": "arduino",
diff --git a/library.properties b/library.properties
```

dmadison, 2024-12-17:
From what I can tell, de02bc1829df15a13babe88f457e2316fb3e2e39 is an orphan commit and not on the master branch. The commit a9fede2f1ce54db1b7a1d82431b0eae42d9515c9 has the same contents and meta, and *is* on the master branch.

I only discovered this because I tried to checkout 3.7.8 on my fork by fetching from upstream and failed, because git does not see that commit.

#### Clone of FastLED/FastLED:
```
git branch -a --contains de02bc1829df15a13babe88f457e2316fb3e2e39
```
The lack of a response is indicative that the commit exists, but is not on any branch.
```
git branch -a --contains a9fede2f1ce54db1b7a1d82431b0eae42d9515c9
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/asm
  remotes/origin/fix-rmt
  remotes/origin/master
```

#### Clone of dmadison/FastLED:
```
git branch -a --contains de02bc1829df15a13babe88f457e2316fb3e2e39
  error: no such commit de02bc1829df15a13babe88f457e2316fb3e2e39
```
```
git branch -a --contains a9fede2f1ce54db1b7a1d82431b0eae42d9515c9
  docs
  docs-squishy
  docs-workflow-dispatch
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/docs
  remotes/origin/docs-squishy
  remotes/origin/docs-workflow-dispatch
  remotes/origin/master
  remotes/upstream/HEAD -> upstream/master
  remotes/upstream/asm
  remotes/upstream/fix-rmt
  remotes/upstream/master
```

If you chase back through the parent commits they both eventually met up at 86343ce875d630355a107aae2b81ab5e20f9bc4b. My guess is that the master branch was rebased and force-pushed around this time, creating the discrepancy.

dmadison, 2024-12-18:
I took a look through the open PRs and found this buried at the bottom of #1548:

[image]

Well that explains that. Yikes.

sutaburosu, 2024-12-18:
If you're looking for force pushes specifically, there's [a list here](https://github.com/FastLED/FastLED/activity?activity_type=force_push).

dmadison, 2024-12-18:
Thank you @sutaburosu. I just stumbled upon that activity in the PR, but it's good to know that GitHub tracks it in one unified place.

By my count that's 67 times the history on master has been changed via force push in the past 4 months. Wow. That's... something.

closed 2025-01-13 by zackees