FastLED/FastLED #1792: LuminescentGrand MIDI library is GPL

dmadison, 2024-12-15:
The MIDI library bundled with the "LuminescentGrand" example is marked as being licensed under GPL. It should be removed and replaced with an alternative, otherwise FastLED will need to switch to the GPL license as well in order to remain compliant.

Perhaps we could add the [MIDI library](https://github.com/FortySevenEffects/arduino_midi_library) as a dependency in the `library.properties` file instead?

nomakewan, 2024-12-15:
It should be noted that the documentation on that header appears to be erroneous; the author themselves state this library has an MIT license.

dmadison, 2024-12-15:
[It looks as though the license changed back in 2015](https://github.com/FortySevenEffects/arduino_midi_library/commit/fb693e724508cb8a473fa0bf1915101134206c34). Prior to that point it was licensed under GPL v3, and the version currently in FastLED is prior to that change.

IANAL, but given that the code currently on master is from when the library was licensed under GPL I think it would still be prudent to either change to a more recent MIT-licensed version, or link to it as a dependency.

zackees, 2024-12-16:
Good catch, the newest version was drag and drop compatible so updated