FastLED/FastLED #1796: Examples are no longer good learning tools

dmadison, 2024-12-15:
Previously, examples had all of the relevant code within the example itself. This made it easy for novices to load up an example, figure out what it was doing, and modify it for their needs.

Now many of the effect-based examples have been rewritten with the effects integrated into the library as classes, and the example simply runs the effect class. This is great for simplicity and integrating multiple effects, but it is bad for people learning how to use the library.

---

By way of example, let's look at the Larson scanner from `Cylon.ino`:

### `Cylon.ino` - Dan's Version

```cpp
// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
	Serial.begin(57600);
	Serial.println("resetting");
	FastLED.addLeds<WS2812,DATA_PIN,RGB>(leds,NUM_LEDS);
	FastLED.setBrightness(84);
}

void fadeall() { for(int i = 0; i < NUM_LEDS; i++) { leds[i].nscale8(250); } }

void loop() { 
	static uint8_t hue = 0;
	Serial.print("x");
	// First slide the led in one direction
	for(int i = 0; i < NUM_LEDS; i++) {
		// Set the i'th led to red 
		leds[i] = CHSV(hue++, 255, 255);
		// Show the leds
		FastLED.show(); 
		// now that we've shown the leds, reset the i'th led to black
		// leds[i] = CRGB::Black;
		fadeall();
		// Wait a little bit before we loop around and do it again
		delay(10);
	}
	Serial.print("x");

	// Now go in the other direction.  
	for(int i = (NUM_LEDS)-1; i >= 0; i--) {
		// Set the i'th led to red 
		leds[i] = CHSV(hue++, 255, 255);
		// Show the leds
		FastLED.show();
		// now that we've shown the leds, reset the i'th led to black
		// leds[i] = CRGB::Black;
		fadeall();
		// Wait a little bit before we loop around and do it again
		delay(10);
	}
}
```

This teaches a novice how to use the library to make a simple effect:
* Setting up and pushing data to the LEDs
* Assigning a color based on hue to the LED
* Changing the hue over time
* Moving the LED down the strip
* Fading the LEDs as they move

A novice can look at that code and change the speed, the foreground color, the background color, the fading speed, and more. If they want, they can also borrow these snippets and build their own unique effect.

### `Cylon.ino` - Current Version

```cpp
// Define the array of leds
CRGB leds[NUM_LEDS];

// Create a Cylon instance
Cylon cylon(NUM_LEDS);

void setup() { 
    FastLED.addLeds<WS2812,DATA_PIN,RGB>(leds,NUM_LEDS).setRgbw();
    FastLED.setBrightness(84);
}

void loop() { 
    cylon.draw(Fx::DrawContext(millis(), leds));
    FastLED.show();
    delay(cylon.delay_ms);
}
```

For a novice, this only teaches them how to run the effect. It does not show how the effect works, and the effect itself is not modifiable except by the constructions parameters, which aren't called out.

The `.hpp` where the effect itself resides is not opened by the IDE, and if I were a novice I'd imagine this example would teach me next to nothing about how to make or customize a 'cylon' effect to my liking.

---

This is all as a way to say: I don't think this is the right approach. The examples don't exist primarily to show off effects, they exist as a way to get users' feet wet with the library. I have run into many people who built their own effects using modified versions of `Fire2012` or some of the functions in `DemoReel100`. That's not nearly as easy to do with this setup.

Don't get me wrong, as a more advanced user I think the effects classes are great and a big step in the right direction. Effects are now available in the library, run asynchronously, retain their state between renders, and can be deployed onto any strip easily. 10/10 no notes. But if you're someone who just bought an Arduino and has never written a line of code in your life, none of that helps you learn.

I think most of the effect-based examples should be reverted back to the way they were, with the new class-based versions added alongside in some capacity. If we don't want to clutter the examples folder then an "fx" directory with a handful of examples (run single effect, run multiple effect, swap on button press, yada yada) is probably the way to go.

Thoughts?

marmilicious, 2024-12-15:
I'm all for the library to move forward and improve. In ALL the ways. (_Thank you Zack and other contributors!_) However, I also agree that the examples seem less friendly to new users now. 

Having examples with everything contained in a single file makes it easier for a new user to see how the code flows and make experimental changes in one place. And when helping a new user, being able to point them to an example and suggest modifying lines x and y gets right to the point. If there's a way to retain some of this in a number of basic introductory examples I think it would be beneficial for new users. 

Maybe a number of the original examples could be renamed **Basics_FirstLight**, **Basics_Blink**, **Basics_Cylon**, **Basics_DemoRee100**, etc and the code kept closer to the old style, and then have the updated versions and additional new examples that use the new format? 

This naming could also naturally help point new users into checking out the **Basics_** examples first. There could also be something like a **Basics_Using_FXs** example which introduces the new classes/FXs and explains things further.

zackees, 2024-12-15:
My plan is to duplicate the examples so it's like the way it was before and also have the fx equivalents.

closed 2024-12-16 by zackees