FastLED/FastLED #1798: I2S broken since commit 4966fcd (with solution!?)

cheche8, 2024-12-16:
Hello,

i tried to get the latest version of FastLED (3.9.4) running using Platformio (Core 6.1.16, Home 3.4.4) using 8 parallel output over I2S.

**First issue (solved)**
The code compiles without problems, but while linking the firmware.bin i get warnings along the lines of "undefined reference to `i2s_transpose_and_encode(int, unsigned int, unsigned int volatile*)'"
Looks like somehow the flag FASTLED_ESP32_I2S (that is set before importing FastLED.h) is not respected by the i2s.cpp file. 
If i **set the FASTLED_ESP32_I2S  as a build flag** it does compile and link. So far so good.

**Second issue (solved)**
The ESP32 (Olimex-POE, WROOM-32E) panics instantely when FastLED.show() is called. I (expertely xD) traced that down, with a lot of printing to the serial port, to this line of code: https://github.com/FastLED/FastLED/blob/0c7a88d97c3737be961ae63b8a32612907e13310/src/platforms/esp/32/clockless_i2s_esp32.h#L266
The preparation of this variable can be found on line https://github.com/FastLED/FastLED/blob/0c7a88d97c3737be961ae63b8a32612907e13310/src/platforms/esp/32/clockless_i2s_esp32.h#L173
Well, the real culprit lies here https://github.com/FastLED/FastLED/blob/0c7a88d97c3737be961ae63b8a32612907e13310/src/platforms/esp/32/clockless_i2s_esp32.h#L168
I get, that the i2s interface only needs to be initialized once. But I also think that most of the other stuff in this function still should happen, even if the i2s interface is already initialized. Especially the variable "mPixels" should be allocated since we write to that later.
So my solution to this looks like:

```
void init() {

        // -- Allocate space to save the pixel controller
        //    during parallel output
        mPixels = (PixelController<RGB_ORDER> *)malloc(
            sizeof(PixelController<RGB_ORDER>));

        // -- Construct the bit patterns for ones and zeros
        if (!i2s_is_initialized()) {
            i2s_define_bit_patterns(T1, T2, T3);
            i2s_init(I2S_DEVICE);
            i2s_set_fill_buffer_callback(fillBuffer);
        }

        gControllers[gNumControllers] = this;
        int my_index = gNumControllers;
        ++gNumControllers;

        // -- Set up the pin We have to do two things: configure the
        //    actual GPIO pin, and route the output from the default
        //    pin (determined by the I2S device) to the pin we
        //    want. We compute the default pin using the index of this
        //    controller in the array. This order is crucial because
        //    the bits must go into the DMA buffer in the same order.
        mPin = gpio_num_t(DATA_PIN);
        i2s_setup_pin(DATA_PIN, my_index);
    }
```
With this change (and setting the build flag) i am able to compile, upload und use FastLED with 8 parallel outputs via the i2s interface with about 600pixel each output. 

Please let me know if you can replicate that and if my solutions are crap and how to do it right. 

My platformio.ini
> [env:esp32-poe]
> platform = espressif32
> board = esp32-poe
> framework = arduino
> monitor_speed = 115200
> lib_deps = fastled/FastLED@^3.9.4
> build_flags = 
>    -D FASTLED_ESP32_I2S

My main.cpp
```
#include <Arduino.h>

#define FASTLED_ESP32_I2S
#include <FastLED.h>

#define DATA_PIN_1 4
#define DATA_PIN_2 5 
#define DATA_PIN_3 33
#define DATA_PIN_4 32

#define DATA_PIN_5 16
#define DATA_PIN_6 15
#define DATA_PIN_7 14
#define DATA_PIN_8 13

#define LED_TYPE    WS2812
#define COLOR_ORDER RGB

#define NUM_LEDS_PER_UNIVERSE 170
#define NUM_LEDS_PER_OUTPUT 600

CRGB* leds;

void setup() {

  leds = new CRGB[NUM_LEDS_PER_UNIVERSE * 32];

  FastLED.addLeds<LED_TYPE, DATA_PIN_1>(leds, NUM_LEDS_PER_UNIVERSE*0, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_2>(leds, NUM_LEDS_PER_UNIVERSE*4, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_3>(leds, NUM_LEDS_PER_UNIVERSE*8, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_4>(leds, NUM_LEDS_PER_UNIVERSE*12, NUM_LEDS_PER_OUTPUT);

  FastLED.addLeds<LED_TYPE, DATA_PIN_5>(leds, NUM_LEDS_PER_UNIVERSE*16, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_6>(leds, NUM_LEDS_PER_UNIVERSE*20, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_7>(leds, NUM_LEDS_PER_UNIVERSE*24, NUM_LEDS_PER_OUTPUT);
  FastLED.addLeds<LED_TYPE, DATA_PIN_8>(leds, NUM_LEDS_PER_UNIVERSE*28, NUM_LEDS_PER_OUTPUT);
}

void loop() {

    size_t cpylen = NUM_LEDS_PER_UNIVERSE * 3 * 32 - 3 * 3;
    void* dst = (void*)(&leds[3]);
    void* src = (void*)(&leds[0]); 
    memmove(dst, src, cpylen);

    // Loop over all 8 outputs to set first three leds
    for (int i = 0; i < 8; i++) {
        if (random8() > 200) {
            leds[i*NUM_LEDS_PER_UNIVERSE * 4] = CRGB(255,0,0);
            leds[i*NUM_LEDS_PER_UNIVERSE * 4 + 1] = CRGB(150,0,0);
            leds[i*NUM_LEDS_PER_UNIVERSE * 4 + 2] = CRGB(100,0,0);
        } else {
            leds[i*NUM_LEDS_PER_UNIVERSE * 4] = CRGB(0,0,0);
            leds[i*NUM_LEDS_PER_UNIVERSE * 4 + 1] = CRGB(0,0,0);
            leds[i*NUM_LEDS_PER_UNIVERSE * 4 + 2] = CRGB(0,0,0);
        }                    
    }
   FastLED.show();
   delay(33);
}
```

zackees, 2024-12-18:
These are great solutions, can you put this into a pull request?

zackees, 2024-12-18:
Never mind, I just put this in.

closed 2024-12-18 by zackees