FastLED/FastLED #1803: idf4_rmt_impl: unknown ESP32 target

Jueff, 2024-12-17:
Compiling the Blink sketch fails with unknown ESP32 target 

Board: ESP32DevBoard
FastLED: 3.9.5
esp32:esp32: Espressif 1.0.4

c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:643:2: error: #error Not yet implemented for unknown ESP32 target
 #error Not yet implemented for unknown ESP32 target
  ^
c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:734:2: error: #error Not yet implemented for unknown ESP32 target
 #error Not yet implemented for unknown ESP32 target
  ^
c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:816:2: error: #error Not yet implemented for unknown ESP32 target
 #error Not yet implemented for unknown ESP32 target
  ^
In file included from C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/soc/soc/gpio_reg.h:17:0,
                 from C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/driver/driver/gpio.h:19,
                 from C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32/esp_sleep.h:19,
                 from C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/esp32/esp_system.h:21,
                 from C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/esp32-hal.h:36,
                 from c:\Users\.....\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/led_sysdefs_esp32.h:2,
                 from c:\Users\.....\Documents\Arduino\libraries\FastLED\src/led_sysdefs.h:42,
                 from c:\Users\.....\Documents\Arduino\libraries\FastLED\src/FastLED.h:59,
                 from c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:14:
c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp: In static member function 'static void ESP32RMTController::interruptHandler(void*)':
c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:822:31: error: 'tx_next_bit' was not declared in this scope
             if (intr_st & BIT(tx_next_bit))
                               ^
C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/soc/soc/soc.h:140:42: note: in definition of macro 'BIT'
 #define BIT(nr)                 (1UL << (nr))
                                          ^
c:\Users\.....\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\idf4_rmt_impl.cpp:831:35: error: 'tx_done_bit' was not declared in this scope
                 if (intr_st & BIT(tx_done_bit))
                                   ^
C:\Users\.....\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4/tools/sdk/include/soc/soc/soc.h:140:42: note: in definition of macro 'BIT'
 #define BIT(nr)                 (1UL << (nr))
                                          ^

Using library FastLED at version 3.9.5 in folder: C:\Users\Jürgen\Documents\Arduino\libraries\FastLED

zackees, 2024-12-18:
This is old code that's been around for a while. Is it suddenly giving issue now? You should have CONFIG_IDF_TARGET_ESP32 define automatically with your Arduino. Can you explain a little bit more about what's going on with your setup?

Jueff, 2024-12-18:
I guess it has something to do with code in led_sysdefs_esp32.h
```
// Handling for older versions of ESP32 Arduino core
#if !defined(ESP_IDF_VERSION)
// Older versions of ESP_IDF only supported ESP32
#define CONFIG_IDF_TARGET_ESP32 1
// Define missing version macros.  Hard code older version 3.0 since actual version is unknown
#define ESP_IDF_VERSION_VAL(major, minor, patch) ((major << 16) | (minor << 8) | (patch))
#define ESP_IDF_VERSION ESP_IDF_VERSION_VAL(3, 0, 0)
#endif
```
with introducing esp_version.h that sets the default for old platforms the define ESP_IDF_VERSION is no long undefined.

maybe a change to 
```
// Handling for older versions of ESP32 Arduino core
#if !defined(ESP_IDF_VERSION)
// Older versions of ESP_IDF only supported ESP32
#define CONFIG_IDF_TARGET_ESP32 1
// Define missing version macros.  Hard code older version 3.0 since actual version is unknown
#define ESP_IDF_VERSION_VAL(major, minor, patch) ((major << 16) | (minor << 8) | (patch))
#define ESP_IDF_VERSION ESP_IDF_VERSION_VAL(3, 0, 0)
#else
#if ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(4, 0, 0)
#define CONFIG_IDF_TARGET_ESP32 1
#endif
#endif
```
will help. For me the change helped.

closed 2024-12-18 by zackees