FastLED/FastLED #1805: namespace clash for ArduinoJson and FastLEDs wrapper for wasm

vortigont, 2024-12-17:
I can find this line `using namespace fl;` in `FastLED.h`
https://github.com/FastLED/FastLED/blob/3bd32773e52f808d9d5f986492c84e5c076fd7c5/src/FastLED.h#L87C1-L87C20

Importing namespace in a header file is so wrong! This results in definition clashing and failure to build units which includes both `FastLED.h` and real `ArduinoJson` lib's header.

i.e. like this:
```
src/embuifs.hpp:29:42: error: reference to 'JsonDocument' is ambiguous

.pio/libdeps/hdwf2_my/FastLED/src/fl/screenmap.h:23:7: note: candidates are: 'class fl::JsonDocument'
   23 | class JsonDocument;
      |       ^~~~~~~~~~~~

.pio/libdeps/hdwf2_my/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:20:7: note:                 'class ArduinoJson::V721PB22::JsonDocument'
   20 | class JsonDocument : public detail::VariantOperators<const JsonDocument&> {
```

this namespace import should be removed from `FastLED.h`

zackees, 2024-12-17:
good catch on using namespace fl.

That's a bug. I will remove it immediately.

zackees, 2024-12-17:
As for the embedding json. We have to if we are going to use it internally.

Ancient avr compilers will absolutely chock so i've opted to make it compile optionally with a stub replacement otherwise.

zackees, 2024-12-18:
Okay fixed. I've also added a test to catch this in there future. Thank you for reporting this bug.

closed 2024-12-18 by zackees

vortigont, 2024-12-18:
thanks! That was fast!

zackees, 2024-12-18:
Great! Now I can ship the bug fix.

We are using json to communicate with the web js app when using the wasm compiler.