FastLED/FastLED #1806: Calculation of T1, T2, T3 is inconsistent?

cheche8, 2024-12-18:
Hello there,

one can find a python script to calculate the T1, T2 and T3 timings based on T0H, T1H, T0L, T1L timings in the datasheet here: https://github.com/FastLED/FastLED/blob/135e53939c0e526aa57c2bb1a6834612bef70170/src/chipsets.h#L878

an example of a WS2812 Datasheet can be found here: https://cdn-shop.adafruit.com/datasheets/WS2812.pdf

The crucial information from that datasheet is in this picture:
[image]

A reference to what exactly the T1, T2, T3 timing do can be found on several places, they all seem consistent:
https://github.com/FastLED/FastLED/blob/135e53939c0e526aa57c2bb1a6834612bef70170/src/chipsets.h#L873
https://github.com/FastLED/FastLED/blob/135e53939c0e526aa57c2bb1a6834612bef70170/src/chipsets.h#L664
https://github.com/FastLED/FastLED/blob/135e53939c0e526aa57c2bb1a6834612bef70170/src/platforms/esp/32/i2s.h#L94

This is how these timings are calculated:

> // At T=0        : the line is raised hi to start a bit
> // At T=T1       : the line is dropped low to transmit a zero bit
> // At T=T1+T2    : the line is dropped low to transmit a one bit
> // At T=T1+T2+T3 : the cycle is concluded (next bit can be sent)

The Python script to calculate those timing looks like this (from chipset.h):
```
print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
T0H = int(input("  T0H: "))
T0L = int(input("  T0L: "))
T1H = int(input("  T1H: "))
T1L = int(input("  T1L: "))

duration = max(T0H + T0L, T1H + T1L)

print("The max duration of the signal is: ", duration)

T1 = T0H
T2 = T1H
T3 = duration - T0H - T0L

print("T1: ", T1)
print("T2: ", T2)
print("T3: ", T3)
```

Now bear with me:  
-👍setting `T1 = T0H` seems legit, since at T1 we drop low to transmit a 0 bit.
-❌ setting `T2 = T1H` seems wrong to me! If T1+T2=time to drop low to transmit a 1bit, then T2 should be `T2=T1H-T0H` eg. the difference between dropping low for a 0 bit and dropping low for a 1 bit
-❌ setting `T3=duration - T0H -T0L` also seems wrong to me. T3 is the time we wait after transmitting a 1 bit. So this should be `T3=duration-T2`

The timing from the datasheet:

> T0H -> 0.35us ±150ns -> min. 200ns, max. 500ns
> T1H -> 0.7us ±150ns -> min. 550ns, max. 850ns
> T0L -> 0.8us ±150ns -> min. 650ns, max. 950ns
> T1L -> 0.6us ±150ns -> min. 450ns, max. 750ns

The timings for the WS2812 Chipset are set in the FastLED source to:

> T1 = 250 
> T2 = 625
> T3 = 375

👍T1 seems possible  T0H is **200ns min** and **500ns max** 
👍T2 seems possible  T1H(min)-T0H(max)=550-500=**50ns min**, T1H(max)-T0H(min)=850-200=**650ns max**
❌T3 seems not possible!   T1L(min)=**450ns min**, T1L(max)=**750ns max**

What do you guys think? Am I getting something completely wrong or are these timing out of spec?
How were the values calculated for including in the FastLED source?

Personally I use a lot of WS2815 strips (lots of them) and never had problems with driving them on the WS2812 protocols, even if the timings for the WS2815 are way worse than WS2812 according to the datasheet. But this now let me down this rabbithole to investigate and i am getting more confused the deeper i go^^

As last, i'd like to share how i think the python script for calculating these timings should look like:
```
print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
T0H = int(input("  T0H: "))
T0L = int(input("  T0L: "))
T1H = int(input("  T1H: "))
T1L = int(input("  T1L: "))

duration = max(T0H + T0L, T1H + T1L)

print("The max duration of the signal is: ", duration)

T1 = T0H
T2 = (T1H-T0H)
T3 = duration - T2

print("T1: ", T1)
print("T2: ", T2)
print("T3: ", T3)
```

zackees, 2024-12-19:
Yeah, the mystery goes deeper.

Let me give you some background. This script was my attempt to expose these enigmatic timings to myself and to the public. This script was lifted off of the esp32 rmt4 encoder that @samguyer wrote. It's the only place in the code base I could find an algorithmic example of the timings.

Yes, the timings seem wrong to me too, but they are ROCK solid when I test them and currently they are maximally stable for overclocking. I don't have any more information than this.

A release or two ago I added a feature to override the WS2812 timings yourself and take full control of T1, T2, T3.

https://github.com/FastLED/FastLED/blob/135e53939c0e526aa57c2bb1a6834612bef70170/src/chipsets.h#L940

Is this your final version of the python script?

MagicMods, 2024-12-19:
I too have been struggling with this and WS2813b-v5 + WS2818F & WS2813b-mini-v1 + WS2818F combo on custom boards.
Using esp32-s3  (Waveshare) with touchscreen and TFT_eSpi as main controller (with a 5v level shifter - technically not necessary with those versions but works well to protect the controller (fried two controllers by hot pluging without level shifter).

https://www.lcsc.com/datasheet/lcsc_datasheet_2012110135_Worldsemi-WS2813B-V5_C965558.pdf
https://www.lcsc.com/datasheet/lcsc_datasheet_2310251544_Worldsemi-WS2813B-MINI-V1_C2832664.pdf
https://wmsc.lcsc.com/wmsc/upload/file/pdf/v2/lcsc/2406201649_Worldsemi-WS2818F_C5446693.pdf
https://www.led-stuebchen.de/download/WS2818.pdf

Both chips have the same timings so theoretically compatible on the same data line, but getting some random glitches with WS2813b-v5 + WS2818F after less than a minute on cold boot. (Could heat affect time keeping for signal reshaping amplification?)
However, WS2813b-mini-v1 + WS2818F works great since applying MAX timings, so does a classic WS2813b-v5 strip.
Using MIN or AVG timings would not work properly. 

// WS2813b-V5 & WS2818F - (Using MAX timings, T1 380ns, T2 1000ns, T3 620ns)

// T0H = 220ns ~ 250ns ~ 380ns,
// T1H = 580ns ~ 790ns ~ 1us,
// T0L = 580ns ~ 790ns ~ 1us,
// T1L = 580ns ~ 790ns ~ 1us,
// RES > 280us

Calculated using the python script :
// MIN T1 220 T2 580 T3 360
// AVG T1 250 T2 790 T3 540
// MAX T1 380 T2 1000 T3 620 

Since the setup is a bit unusual and running custom boards, 
it is difficult to single out issue having many other aspects to consider.

Working a different solution to try isolate the issue before digging deeper into fastled timing.
Currently building jigs to be able to scope properly under stress/load SI SO and power/noise.
Also looking at controller itself, between 15 to 45% cpu usage with wifi might introduce time keeping issues? (although, would have expected strip wide glitches rather than individual/groups (usually 3 or 4 sequential boards).
Making a setup with a bare esp32-s3 without TFT_eSpi, touch or wifi to see if anything interfere with it.

PS. For anyone having issues with FastLed & TFT_eSpi, try the following defines:

-D FASTLED_RMT5=1
-D ESP32S3
-D USE_HSPI_PORT
-D CONFIG_IDF_TARGET_ESP32S

Other note, most WS28xx datasheets suggests resistors to be place on data line.
Not for WS2813b-v5 and WS2813b-mini-v1 but suggested for WS2818 
The combo WS2813b-mini-v1 + WS2818F works great with 33ohms placed just before Si & Bi. Rock solid.
But had to bypass them for WS2813b-V5 & WS2818F to work at all (perhaps a clue?).

dmadison, 2024-12-19:
> Now bear with me:
> -👍setting T1 = T0H seems legit, since at T1 we drop low to transmit a 0 bit.
> -❌ setting T2 = T1H seems wrong to me! If T1+T2=time to drop low to transmit a 1bit, then T2 should be [...]

I think you're close. From how I understand it your conclusions for T1 and T2 are correct. T3 is padding to make the transmit time for each bit the same. Since the high bit takes marginally more time, we can just use that. So it is not `duration - T2` it is `duration - (T1 + T2)`, or simply `T1L` so long as the 1-bit timing is longer. That matches the comments in the source.

Here's the script with that in mind:
```python
print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
T0H = int(input("  T0H: "))
T0L = int(input("  T0L: "))
T1H = int(input("  T1H: "))
T1L = int(input("  T1L: "))

duration = max(T0H + T0L, T1H + T1L)

print("The max duration of the signal is: ", duration)

T1 = T0H
T2 = (T1H-T0H)
T3 = duration - (T1 + T2)

print("T1: ", T1)
print("T2: ", T2)
print("T3: ", T3)
```

Here's the same script in reverse (calculating the timings from T1, T2, and T3):
```python
print("Enter the values of T1, T2, and T3 in nanoseconds: ")
T1 = int(input("  T1: "))
T2 = int(input("  T2: "))
T3 = int(input("  T3: "))

print()

T0H = T1
T1H = T1 + T2
T0L = (T1 + T2 + T3) - T0H  # implicitly, T2 + T3
T1L = (T1 + T2 + T3) - T1H  # implicitly, T3

print("T0H: ", T0H)
print("T0L: ", T0L)
print("T1H: ", T1H)
print("T1L: ", T1L)

print("0-bit duration: ", T0H + T0L)
print("1-bit duration: ", T1H + T1L)
```

And here's both of them combined for validation:
```python
print("Enter the values of T0H, T0L, T1H, T1L, in nanoseconds: ")
T0H = int(input("  T0H: "))
T0L = int(input("  T0L: "))
T1H = int(input("  T1H: "))
T1L = int(input("  T1L: "))

print()

duration = max(T0H + T0L, T1H + T1L)
print("The max duration of the signal is: ", duration)

T1 = T0H
T2 = (T1H-T0H)
T3 = duration - (T1 + T2)

print("T1: ", T1)
print("T2: ", T2)
print("T3: ", T3)

NT0H = T1
NT1H = T1 + T2
NT0L = (T1 + T2 + T3) - T0H  # implicitly, T2 + T3
NT1L = (T1 + T2 + T3) - T1H  # implicitly, T3

if(T0H != NT0H): print("T0H mismatch ({} in, {} out)".format(T0H, NT0H))
if(T1H != NT1H): print("T1H mismatch ({} in, {} out)".format(T1H, NT1H))
if(T0L != NT0L): print("T0L mismatch ({} in, {} out)".format(T0L, NT0L))
if(T1L != NT1L): print("T1L mismatch ({} in, {} out)".format(T1L, NT1L))

print("0-bit duration: ", NT0H + NT0L)
print("1-bit duration: ", NT1H + NT1L)
```

NB: if the scripts in the repo are changed it might be better to reorder the script inputs / outputs to match the order in the datasheet.

---

Using the values on master:
* T1: 250
* T2: 625
* T3: 375

Results in the following timings, at 1250 ns per bit:
* :heavy_check_mark: T0H: 250 (350 nom +/- 150, 200-500)
* :x: T0L: 1000 (800 nom +/- 150, 650-950)
* :x: T1H: 875 (700 nom +/- 150, 550-850)
* :x: T1L: 375 (600 nom +/- 150, 450-750)

T0H is in tolerance, T0L and T1H are slightly out of tolerance (long), T1L is slightly out of tolerance (short).

---

Using the nominal values from the datasheet yields:
* T1: 350
* T2: 350
* T3: 600

And results in the following timings, at 1300 ns per bit:
* :heavy_check_mark: T0H: 350 (350 nom +/- 150, 200-500)
* :heavy_check_mark: T0L: 950 (800 nom +/- 150, 650-950)
* :heavy_check_mark: T1H: 700 (700 nom +/- 150, 550-850)
* :heavy_check_mark: T1L: 600 (600 nom +/- 150, 450-750)

Everything matches the inputs except for T0L, which is close to out (long). The problem here is that the TN values are in steps of 125, so these values are ideals and not practical.

---

Using the closest 125 increment:
* T1: 375
* T2: 375
* T3: 500

Results in the following timings, at 1250 ns per bit:
* :heavy_check_mark: T0H: 375 (350 nom +/- 150, 200-500)
* :heavy_check_mark: T0L: 875 (800 nom +/- 150, 650-950)
* :heavy_check_mark: T1H: 750 (700 nom +/- 150, 550-850)
* :heavy_check_mark: T1L: 500 (600 nom +/- 150, 450-750)

All of those are in-tolerance, and with the same total bit time as the current implementation. Assuming the base calculations are correct. It does, however, put all of the timings closer to each-other.

---

My assumption is that this is on purpose, because the timing difference between T0H and T1H is so close (500 ns max vs 650 ns min). With the current timings T0H errs towards the minimum timing and T1H errs towards the maximum. If there are any execution delays, it's less likely for a bit to flip.

The big question is how the real world output timing differs from the T1, T2, and T3 inputs.

dmadison, 2024-12-19:
To sum up in plain English, as I understand it:

* `T1` is the high time for a 0 bit
* `T2` is the additional high time required for a 1 bit
* `T3` is the padding for a low time afterwards

zackees, 2024-12-20:
> I too have been struggling with this and WS2813b-v5 + WS2818F & WS2813b-mini-v1 + WS2818F combo on custom boards.
>
> Using esp32-s3  (Waveshare) with touchscreen and TFT_eSpi as main controller (with [...]

Can you do me a favor and put the esp32 timings on a scope? If this python script is slightly wrong, then so is our C++ version specifically for the RMT5/4 timings.

It would be interesting to see if T1, T2, T3 are actually being expressed like the code is directing it to do.

zackees, 2025-11-05:
## ✅ Issue Confirmed and Fixed

Thank you @cheche8 for the detailed analysis! You are **100% correct** - the original Python script had **two bugs**:

### Bugs Identified
1. ❌ **T2 = T1H** (WRONG) → Should be **T2 = T1H - T0H** 
2. ❌ **T3 = duration - T0H - T0L** (WRONG) → Should be **T3 = duration - T1H**

Your proposed fix is exactly right!

closed 2025-11-05 by zackees