FastLED/FastLED #1807: macro "digitalWrite" requires 2 arguments, but only 1 given

adesouky13, 2024-12-18:
Hello,

I have a program that previously worked but now when I try to verify it (no changes to the program), I get the error listed below.
it claims the issue is in the _FastLED/src_ file but I am guessing that some link is just broken. I have tried to verify using multiple computers, as well as verifying with multiple revisions of FastLED but keep getting the same issue. Since it is pointing me to the _FastLED/src_ file, I am not sure how to go about debugging it.

The error:
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:33: error: macro "digitalWrite" requires 2 arguments, but only 1 given
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                 ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:97:32: error: macro "digitalWrite" requires 2 arguments, but only 1 given
     mImpl->digitalWrite(is_high);
                                ^
In file included from /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/Arduino.h:223,
                 from /var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:20:
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:42:65: error: 'digitalPinToGPIONumber' is not a type
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:48:10: note: in expansion of macro 'digitalRead'
     bool digitalRead() { return HIGH == ::digitalRead(mDigitalPin); }
          ^~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:86: error: variable or field 'digitalWrite' declared void
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                                                                      ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:86: error: expected ';' at end of member declaration
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                                                                      ^
                                                                                       ;
In file included from /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/Arduino.h:223,
                 from /var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:20:
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp: In member function 'bool fl::DigitalPin::high() const':
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:42:91: error: too few arguments to function 'int8_t digitalPinToGPIONumber(int8_t)'
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                                           ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:93:19: note: in expansion of macro 'digitalRead'
     return mImpl->digitalRead();
                   ^~~~~~~~~~~
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:9:8: note: declared here
 int8_t digitalPinToGPIONumber(int8_t digitalPin);
        ^~~~~~~~~~~~~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp: In member function 'void fl::DigitalPin::write(bool)':
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:97:12: error: 'class fl::DigitalPinImpl' has no member named 'digitalWrite'; did you mean 'digitalRead'?
     mImpl->digitalWrite(is_high);
            ^~~~~~~~~~~~
            digitalRead

Any ideas would be greatly appreciated
Thank you!

zackees, 2024-12-18:
digitalRead and digitalWrite are macro defines here...

closed 2024-12-18 by zackees

adesouky13, 2024-12-19:
> FastLED 3.9.4 and earlier all work for me with core 2.0.18. You are using an earlier version of the core, v2.0.13. With that version installed I had to go back to FastLED 3.9.3 for a successful compile.

I looked into the ESP32 cores and I found a difference between the Arduino Cloud IDE and the Desktop IDE.

Desktop IDE:
I was able to get a successful compile with FastLED 3.9.4 using both ESP32 core 2.0.18 or 2.0.13. Any newer version of FastLED would result in the error.

Arduino Cloud IDE:
I am not sure how I can change the ESP32 core so I had to stick with the 2.0.13 that it currently has. I was not able to get a successful compile with any version of FastLED (I tried all versions from 3.7.8 to 3.9.6) including the 3.9.3 and 3.9.4 that worked in the desktop application.
Ideally I would like to use the Cloud IDE as I have it set up for OTA updates.

zackees, 2024-12-19:
I just did two releases. 3.9.6 has fixes for arduino cloud, based on your error log output. Give it spin, it should work now.

zackees, 2024-12-19:
I just did a few fixes for arduino cloud. Can you provide some log output and I'll see if I can get it to work.

adesouky13, 2024-12-19:
> I just did a few fixes for arduino cloud. Can you provide some log output and I'll see if I can get it to work.

I just tried it again with FastLED v3.9.6 as you suggested and I still am getting the same error.

Apologies as I am kind of just repeating the error message, but is this the log output that you were referencing? Otherwise if you are looking for something else I can try to find that for you as well.

`/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:33: error: macro "digitalWrite" requires 2 arguments, but only 1 given
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                 ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:97:32: error: macro "digitalWrite" requires 2 arguments, but only 1 given
     mImpl->digitalWrite(is_high);
                                ^
In file included from /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/Arduino.h:223,
                 from /var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:20:
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:42:65: error: 'digitalPinToGPIONumber' is not a type
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                 ^~~~~~~~~~~~~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:48:10: note: in expansion of macro 'digitalRead'
     bool digitalRead() { return HIGH == ::digitalRead(mDigitalPin); }
          ^~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:86: error: variable or field 'digitalWrite' declared void
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                                                                      ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:49:86: error: expected ';' at end of member declaration
     void digitalWrite(bool value) { ::digitalWrite(mDigitalPin, value ? HIGH : LOW); }
                                                                                      ^
                                                                                       ;
In file included from /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/Arduino.h:223,
                 from /var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:20:
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp: In member function 'bool fl::DigitalPin::high() const':
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:42:91: error: too few arguments to function 'int8_t digitalPinToGPIONumber(int8_t)'
 #define digitalRead(pin)                            digitalRead(digitalPinToGPIONumber(pin))
                                                                                           ^
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:93:19: note: in expansion of macro 'digitalRead'
     return mImpl->digitalRead();
                   ^~~~~~~~~~~
/run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/cores/esp32/io_pin_remap.h:9:8: note: declared here
 int8_t digitalPinToGPIONumber(int8_t digitalPin);
        ^~~~~~~~~~~~~~~~~~~~~~
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp: In member function 'void fl::DigitalPin::write(bool)':
/var/run/arduino/directories-user/libraries/FastLED/src/sensors/digital_pin.cpp:97:12: error: 'class fl::DigitalPinImpl' has no member named 'digitalWrite'; did you mean 'digitalRead'?
     mImpl->digitalWrite(is_high);
            ^~~~~~~~~~~~
            digitalRead
Multiple libraries were found for "WiFi.h"
  Used: /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/libraries/WiFi
  Not used: /run/arduino/directories-data/internal/WiFiEspAT_1.5.0_5169a7bcde168205
  Not used: /run/arduino/directories-data/internal/indhilib_3.0.5_645ae16078e566f4
  Not used: /run/arduino/directories-data/internal/VEGA_WiFiNINA_1.0.1_ab2db6059607e9f4
  Not used: /run/arduino/directories-data/internal/BetterWiFiNINA_1.3.0_0ccca45bdf408f34
  Not used: /run/arduino/directories-data/internal/Seeed_Arduino_rpcWiFi_1.1.0_a93a349668c69c81
  Not used: /run/arduino/directories-data/internal/DA16200_Wi-Fi_Library_for_Arduino_1.1.0_9d156d307009fe30
  Not used: /run/arduino/directories-data/internal/WiFi_1.2.7_c357666d3257ae8a
  Not used: /run/arduino/directories-data/internal/NINA-Wi-Fi_1.0.1_8b5f9e1dc71eca55
  Not used: /run/arduino/directories-data/internal/WiFiNINA_1.8.14_50fd037d6f2f4b4b
Multiple libraries were found for "WiFiClientSecure.h"
  Used: /run/arduino/directories-data/packages/arduino/hardware/esp32/2.0.13/libraries/WiFiClientSecure
  Not used: /run/arduino/directories-data/internal/Seeed_Arduino_rpcWiFi_1.1.0_a93a349668c69c81`

zackees, 2024-12-19:
And the other errors are gone now right?

Okay, it will work on the next release then. I'll try and get it out soon.

adesouky13, 2024-12-19:
Nope, it is the exact same errors as the initial post: macro "digitalWrite" requires 2 arguments, but only 1 given

And thank you for looking into it and your efforts with a new release. 
I look forward to your next release

zackees, 2024-12-19:
Thanks. I hope you enjoy it.