FastLED/FastLED #1811: ESP32 I2S compile errror

Jueff, 2024-12-22:
Compiling the Blink sketch with FASTLED_ESP32_I2S and FASTLED_ESP32_I2S_NUM_DMA_BUFFERS 4 defined fails 

Board: ESP32DevBoard
FastLED: 3.9.7 (when switching back to 3.6.0 problem is gone)
esp32:esp32: Espressif 1.0.4

Code

```
#define FASTLED_ESP32_I2S
#define FASTLED_ESP32_I2S_NUM_DMA_BUFFERS 4
#include <FastLED.h>
...
...

```
To get the code compiled I also commented out the "#ifdef FASTLED_ESP32_I2S" in i2s.cpp.

Build output
```
Compiling core...
Using precompiled core: C:\Users\Tester\AppData\Local\arduino\cores\1420933ad65eba4cac1859e0a4bb7280\core.a
Linking everything together...
"C:\\Users\\Tester\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib 
...
...

C:\Users\Tester\AppData\Local\arduino\sketches\4EB0E45605664DFC0D280DCBCBFD79BC\sketch\Blink.ino.cpp.o:(.literal._ZN19ClocklessControllerILi16ELi60ELi150ELi90EL6EOrder66ELi0ELb0ELi5EE10fillBufferEv[ClocklessController<16, 60, 150, 90, (EOrder)66, 0, false, 5>::fillBuffer()]+0x1c): undefined reference to `gCntBuffer'
collect2.exe: error: ld returned 1 exit status

Using library FastLED at version 3.9.7 in folder: C:\Users\Tester\Documents\Arduino\libraries\FastLED 
exit status 1
```

the variabel gCntBuffer is decleared as extern in i2s.h, but nowhere defined.

Jueff, 2024-12-22:
looks like the code 
    #if FASTLED_ESP32_I2S_NUM_DMA_BUFFERS>2
        gCntBuffer++;
    #endif        
is missing now.

zackees, 2024-12-24:
Fixed, it will go out in the next release. Feel free to install FastLED manually for a work around now. You can also just do

`int gCntBuffer = 0;` in your sketch so that the linker finds the symbol.

closed 2024-12-24 by zackees

Jueff, 2024-12-24:
Works having FASTLED_ESP32_I2S_NUM_DMA_BUFFERS undefined.
Having FASTLED_ESP32_I2S_NUM_DMA_BUFFERS defined not pulse output to pin.

please see my first comment.