FastLED/FastLED #1812: Limiting the number of leds arrays to 8

Dusty613, 2024-12-22:
I will embarrass myself with this question I am certain. I have been writing the code for FastLED to run 12 arrays for up to 20 LED's in the array. Recently I cannot compile more that 8 arrays. Using Arduino Nano and Uno boards, WX2812B strips. Anyone else seeing a problem like this.

sutaburosu, 2024-12-22:
From the [release notes](https://github.com/FastLED/FastLED/releases) for 3.9.5:
> * AVR
>   * The Atmega family and 32u now has a maximum of 16 controllers that can be active, up from > 8, due to these models having more memory. Someone actually needed this, suprisingly.

Like you, I have used >8 pins on a Nano, so this surprised me.

[It turns out this change from 3 months ago](https://github.com/FastLED/FastLED/commit/2593b3a80dbedf938e7dc0e707bb7bcd5853f7c3#diff-c4bf79d7ed6a953c1a4e68cb36e23d1d8cf22964e23d55b82bd2c695aba9873dR9-R11) reduced it to 8 for AVR chips. Apparently simply adding `#define MAX_CLED_CONTROLLERS 12` to your sketch before `#include <FastLED.h>` WILL NOT influence this limitation. [See here](https://github.com/FastLED/FastLED/issues/1777#issuecomment-2483038524).

Please direct support request such as this to [r/FastLED](https://reddit.com/r/FastLED).

zackees, 2024-12-24:
Yeah this code is in a cpp file so an include define will not reach it. You'll need to use a build define. We limit number of controllers to make sure that the AVR compiles stay small.

But most likely, you are on an Arduino and that won't be possible, since it's the only build system in existence which does not support this feature. You can either download the FastLED library and embed it in your project and change this define, or migrate to the much better experience of using the PlatformIO, which is backwards compatible with Arduino and... well it's simply so much better it's really hard to compare.

I've created a repo for this use case. Just download it (or git clone). Swap in the proper platform.ini and start moving your source code in the `src/` directory until the demo project becomes your project. It will take you five minutes and you'll never go back. This is the feedback I get from everyone that uses this.

Start here.
https://github.com/FastLED/PlatformIO-Starter

closed 2024-12-24 by zackees

Dusty613, 2024-12-24:
Thank you so very much. 
Dan Stephenson