FastLED/FastLED #1813: Flickering with Artnet on ESP32C3

Tomatendose, 2024-12-22:
Hello,

I have a problem with the FastLED Libary via Artnet Wifi which runs on an ESP32-C3. Several ESP32-C3 are addressed. Occasionally there is flickering in different colors during operation. I have made a video in slow motion, you can see that there is a delay and then the flickering appears. This can be observed on 144 WS2812 (set in code 170). When using the NeoPixel Libary with Artnet with 64 WS2812 (in code 64) otherwise the same setting, the problem does not occur. 

The following Aretnet [example](https://github.com/rstephan/ArtnetWifi/blob/master/examples/ArtnetWifiFastLED/ArtnetWifiFastLED.ino) is used. The number of LEDs is set to 170 in the code .

https://github.com/user-attachments/assets/46b0f4ee-c812-4bd5-a04b-f7bda6c971ce

zackees, 2024-12-24:
ESP32C3 is notorious for buffer underrunning. You can mess with the settings and put in a build parameter to increase the RMT buffer size to 512 rmt symbols, the default is 128 symbols on C3 because we assume you'll have up to 4 strips. Since you only have 1, you can use the entire block just for your project.

Use this build define (not include define)

`-DLED_STRIP_RMT_DEFAULT_MEM_BLOCK_SYMBOLS=512`

But the big problem is that the ESP32C3 is a single core. So all your wifi stuff is going through the same core as the interrupts (priority 3) which is being used to re-buffer the RMT data.

You'll likely see an improvement with these settings but 100 ms delay is just simply a ton of data.

My best suggestion is to upgrade to the S3 or the C6. Both are dual core but the S3 has an DMA memory and deep transaction queues. The C6 is slightly more expensive than the C3, while the S3 is a couple bucks more.

In the future, unless you are building for mass production, always use the S3. It's the power work horse of the ESP32 family.

closed 2024-12-24 by zackees

Tomatendose, 2024-12-27:
Thanks, I will test it with the RMT symbols.

I only have to correct you in one place. the ESP32C6 is not a DualCore. The DeepSleep version has a low power core. So only an ESP32 S3 would be an option.

zackees, 2024-12-28:
Oh, thanks for the correction!!!