FastLED/FastLED #1814: Error: Template with C linkage

diabl0w, 2024-12-23:
Hi, I apologize as this is most likely noob user error, but I am trying to compile via (makefile)[https://github.com/plerup/makeEspArduino]

When I do, I get tons of errors that look like this:
```
ESP/toolchains/esp8266/cores/esp8266/libc_replacements.cpp:36:
/home/brian/src/MCUs/ESP/aquarium-make/libraries/FastLED/src/fl/ptr.h:405:5: error: template with C linkage
  405 |     template <typename T> friend class Ptr;
      |     ^~~~~~~~
```
Could you please help diagnose if I am doing something wrong?

zackees, 2024-12-24:
Why is libc_replacements.cpp trying to include our internal Ptr class?

Are you wrapping our includes in extern "C"?

Feel free to dump whatever you have into this bug report. However, we have so many compile tests I can't assume this is anything related to us, unless proven otherwise.

diabl0w, 2024-12-24:
> Why is libc_replacements.cpp trying to include our internal Ptr class?
> 
> Are you wrapping our includes in extern "C"?
> 
> Feel free to dump whatever you have into this bug report. However, we have so many compile tests I can't assume this is anything related to us, unless proven otherwise.

unfortunately I am away for my computer for a week or 2, but the other part of the log that I stupidly didn't include was exactly what you mentioned `extern C`. I don't have any such extern C code in my simple main.cpp, it is coming from here: https://github.com/esp8266/Arduino/blob/bb79e9076e8402c81c24efbabd19f1baab1c6bd1/cores/esp8266/libc_replacements.cpp#L48

 The log was just a repeat errors of `extern C` (from libc_replacements.cpp) followed by `some template` (from a fastled class).. it wasn't specific to ptr class

zackees, 2024-12-24:
Thanks for the links.

The include at line 36 is this:

`#include <sys/stat.h>`

What are the build steps?

diabl0w, 2024-12-24:
> Thanks for the links.
> 
> The include at line 36 is this:
> 
> `#include <sys/stat.h>`
> 
> Which doesn't seem to make any sense.
> 
> What are the build steps?

In my build directory, I simply have main.cpp, a libraries folder with my 3rd party libs in them (such as fastled) and I just run `make` using [this makefile](https://github.com/plerup/makeEspArduino)

diabl0w, 2024-12-24:
Also Esp.cpp:
```
/root/Espressif/new-aquarium/libraries/FastLED/src/fl/callback.h:12:1: error: template with C linkage
   12 | template<typename ...Args>
      | ^~~~~~~~
/root/Espressif/toolchains/esp8266/cores/esp8266/Esp.cpp:36:1: note: 'extern "C"' linkage started here
   36 | extern "C" {
      | ^~~~~~~~~~
```

zackees, 2024-12-24:
can you zip up your project and make it available to me?

diabl0w, 2024-12-24:
> can you zip up your project and make it available to me? Or is that too much to ask?

Not too much at all.. see attached

You must `git clone` fastled repo into libraries folder
Then you must `git clone`, `submidule update`, and `python3 repoflder/tools/get.py` [this repo](https://github.com/esp8266/Arduino)
And in the makefile update the ESP_ROOT to that folder
Then just run `make`
[project.zip](https://github.com/user-attachments/files/18241756/project.zip)

sutaburosu, 2024-12-24:
I can reproduce your exact problem here. I can also create very similar problems with nothing but an empty setup() and loop() in `main.cpp`. To my mind, this points to a problem with your build system rather than FastLED. I moved past the first error by adding an include path that is needed: `-I${ESP_ROOT}tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/10.3.0`. There were many more errors.

zackees, 2024-12-25:
Thank you @sutaburosu for the check. That's super appreciated.

closed 2024-12-25 by zackees

diabl0w, 2024-12-25:
> I can reproduce your exact problem here. I can also create very similar problems with nothing but an empty setup() and loop() in `main.cpp`. To my mind, this points to a problem with your build [...]

 I don't have issues with any other libraries , so that was my only clue to fingerpoint at FastLED. But it's too far beyond my knowledge to have any other evidence beyond that, so thank you for checking anyway

zackees, 2024-12-25:
Paste your cmake file to chat gpt and see if it can diagnose what's going wrong. It's extremely good.

diabl0w, 2024-12-26:
> Paste your cmake file to chat gpt and see if it can diagnose what's going wrong. It's extremely good.

Haha, I did try that to no avail... I'll try to make the time to make it a learning opportunity and figure it out myself. Thanks for offering the insight and sorry for any waste of time