FastLED/FastLED #1815: Support for Arduino Giga R1 Support?

k-hanover, 2024-12-23:
Are there any plans to support the Arduino Giga R1?

zackees, 2024-12-23:
Unfortunately, there's only one active developer on FastLED right now and I don't use mbed. If anyone send me a PR for this board I'll happily merge it in.

zackees, 2024-12-24:
Just compiled it for laughs. There are so many compiler errors it's not even funny.

This board doesn't have pin definitions. It doesn't have normal types for controllers. Some of these are probably long hanging fruit - by turning on certain switches that similar boards have may get us closer. However, F_CPU is not defined because this board has a variable clock rate instead of a constant clock rate. This breaks our assumptions and also the software CPU bit-banging code.

In short, it's going to take a lot of work to get this board to work at all.

closed 2024-12-24 by zackees