FastLED/FastLED #1816: Not working for ATmega8A/PU

Sinamajidi, 2024-12-25:
I recently tried using this library in PlatformIO for the ATmega8A/PU. I changed the LED count to 12 and set the LED pin to PB0. However, despite checking other ATmega pins like PC5, I encountered an issue. None of them caused the NeoPixel ring LED to light up. I also attempted to use different drivers, such as SK... and WS..., but none of them worked.

I confirmed that I was able to toggle the pin using simple commands with both Arduino.h and the DDRB/PORTB registers, and those worked correctly. However, the NeoPixel still isn't functioning. I even used an oscilloscope to check the output, and it shows that nothing is being outputted.

zackees, 2024-12-28:
This is out of my specialty. I will accept a PR that fixes this though and put it under our test framework if someone smart figures out a fix.

zackees, 2025-01-25:
Oh, it's an 8mhz board. Yeah, anything under 10mhz doesn't work with FastLED ws2812.