FastLED/FastLED #1817: RTM channel failed :: version 3.9.7

rootexpression, 2024-12-26:
May be related to https://github.com/FastLED/FastLED/issues/1768 but it is far outside my ability to diagnose.

`00:22:23.725 -> E (39946) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
00:22:23.725 -> E (39946) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
00:22:23.757 -> 1`

FastLED 3.9.6 works, 3.9.7 errors with the above

For now, I've fallen back to 3.9.6

ShinodaY, 2024-12-26:
me, too.
FastLED 3.9.6 works fine
But 3.9.7 works but the same error messages.  
I've fallen back to 3.9.6, too. Its OK.

zackees, 2024-12-27:
Yeah, i'm on it.

We have an issue with the first led being stuck green and I can't repro so unfortunately there have been a few releases where we try to fix. The next release will probably fix it.

zackees, 2024-12-28:
Fixed in 3.9.8. Feel free to re-open if this is not the case.

closed 2024-12-28 by zackees

ShinodaY, 2025-01-01:
On my arduino sketch for WS2812Bs,
* FastLED 3.9.6 works fine
* FastLED 3.9.7 works but Error message on serial monitor
  E (84374) rmt: rmt_tx_disable(835): channel can't be disabled in state 1
  E (84374) led_strip_rmt: led_strip_rmt_wait_refresh_done(85): disable RMT channel failed
* FastLED 3.9.8 works fine
Thank you for your prompt action!!