FastLED/FastLED #1818: Weird behavior of v3.9.8 on Teensy 4.x with ObjectFLED driver

lpaolini, 2024-12-28:
I've been using FastLED with the mighty Teensy 4.x platform for years for my [Striptease project](https://github.com/lpaolini/Striptease), together with [Paul Stoffregen's WS2812Serial library](https://github.com/PaulStoffregen/WS2812Serial), which implements non-blocking rendering to get the highest possible framerate, and [Paul Stoffregen's Audio library](https://github.com/PaulStoffregen/Audio), which provides professional audio processing capabilities.

When I saw the release notes for v3.9.8 I was thrilled to see this technology integrated into FastLED, so I made a quick test, replacing `#define USE_WS2812SERIAL` with `#define FASTLED_USES_OBJECTFLED`.

It kind of worked, as the fps jumped from 43 (plain FastLED) to 133 (the same I get with WS2812Serial), but there some issues, like random pixels flickering all the time (see [video](https://photos.app.goo.gl/U5HdcNE6H2MaTKgz8)).

Probably there's something in my code, but maybe not. Perhaps some interaction with the [audio shield](https://www.pjrc.com/audio-shield-for-teensy-4-0/) and [its library](https://www.pjrc.com/teensy/td_libs_Audio.html)? For now I'll stay with WS2812Serial, as it works buttery smooth with my project, and I don't need massive scalability, but if you want me to do some specific test to try debugging this issue, please let me know and I'd be happy to collaborate.

zackees, 2024-12-28:
My one request for you would be to `#if 0` out any code so that it's only running with the ObjectFLED driver.

Please let me know if it's stable then.

zackees, 2024-12-29:
Can you explain how you enable the Serial WS2812 driver in your project?

lpaolini, 2024-12-29:
> Can you explain how you enable the Serial WS2812 driver in your project?

See lines [6-8 here](https://github.com/lpaolini/Striptease/blob/e475d509d8c7edea839b931e4da59397e3fe6bd6/src/Stage.h#L6C1-L8C21) and [396-399 here](https://github.com/lpaolini/Striptease/blob/e475d509d8c7edea839b931e4da59397e3fe6bd6/examples/example1/src/MyStage.cpp#L396
)

Then I'm addressing LEDs using `CRGBSet`s.

lpaolini, 2024-12-29:
> My one request for you would be to `#if 0` out any code so that it's only running with the ObjectFLED driver.
> 
> Please let me know if it's stable then.

I'll start removing the audio library and related code.

lpaolini, 2024-12-30:
I hope this incompatibility can be resolved, as the Audio library offers immense value for LED projects based on Teensy 4.x.

zackees, 2024-12-30:
@PaulStoffregen thanks for the great audio library. Any insight here?

KurtMF, 2024-12-31:
I know the DMA code uses channels dma0-dma3; not sure if audio library has a dependency on those channels or not.  

When I was testing objectFLED, flickering like that seemed to be related to having too short a latch delay time.  Does FastLED let you increase this from default of 75 uS?  Or provide access to it's waveform timing capability?

KurtMF, 2024-12-31:
I will try to dig into the audio library, and see if it's trying to use the same dma or timer channels as OctoWS2811/ObjectFLED.

What LED chip are you using?  WS2812?

KurtMF, 2024-12-31:
> @PaulStoffregen thanks for the great audio library. Any insight here?

Yes!  My thanks as well, @PaulStoffregen.  I enjoyed learning how your code works, and after trying to get parallelism with software spi and uart, discovering your DMA solution was awesome!  

I tried to make it into a beginner-friendly package, to co-exist with FastLED.  Hopefully, this will lead others in the LED art community to discover the power of Teensy!

zackees, 2024-12-31:
> I know the DMA code uses channels dma0-dma3; not sure if audio library has a dependency on those channels or not.
>
> When I was testing objectFLED, flickering like that seemed to be related to having [...]

I'll add it now

zackees, 2024-12-31:
Okay, checked into master, enable it like this:

```C++
#define FASTLED_OBJECTFLED_LATCH_DELAY 150
#include "FastLED.h"
```

zackees, 2024-12-31:
Kurt, did you see any change in stability when the DMA buffer size was increased?

KurtMF, 2024-12-31:
I never tested ObjectFLED with other apps running DMA in the background.  Increasing DMA block sz decreased the amount of background interrupts generated by DMA end-of-block handling.  However it also increased the amount of time show() blocked before it returned.  So I made it as small as I could before interrupt activity started to degrade fps on a 4k LED object.

zackees, 2024-12-31:
Any way you can increase the DMA block size that I can pass in via the constructor?

lpaolini, 2024-12-31:
> I will try to dig into the audio library, and see if it's trying to use the same dma or timer channels as OctoWS2811/ObjectFLED.
> 
> What LED chip are you using? WS2812?

WS2812B

KurtMF, 2024-12-31:
I'm familiar with -12B datasheet, and I don't think this is a waveform latch delay problem.

I snooped into the audio library code, and it is in fact using at least 1 DMA channel.  Maybe something else you load is using additional channels too?  My best theory is that ObjectFLED is being forced to use a non-compatible DMA channel after Audio (and others?) are allocating their DMA channels.  From the Teensy hardware manual, only the first 4 DMA channels are compatible with timer trigger and GPIO register output.  ObjectFLED uses 3 DMA channels, and it looks like Audio uses 1.  

To test this theory, can you ensure in your code that ObjectFLED will run it's begin() before any of the Audio objects are started or initialized?  And since you are using OFLED via FastLED, that means running addLEDs() before initializing Audio or other objects.  This should  allow ObjectFLED to get the first 3 available hardware DMAs.

Then hopefully, both libraries can be happy together.

lpaolini, 2024-12-31:
> I have managed to reproduce the issue with the following code, by uncommenting [line 10](https://github.com/lpaolini/FastLED-1818/blob/da57fc771dc8eea6e2b46d93e1c77a7161f9fcf2/src/main.cpp#L10).
> 
> https://github.com/lpaolini/FastLED-1818

@KurtMF Please check the (minimal) sketch above reproducing the issue.
I'm not using ObjectFLED directly, only via FastLED.

KurtMF, 2024-12-31:
Right.  So can you instantiate the AudioInput I2S object after calling addLEDs()?  That constructor looks like it allocates 2 DMA channels for Teensy.  

If that's not feasible, then I will have to do some deep diving later to see how DMAChannel.h is allocating channels to multiple apps.  If this is truly a DMA conflict, then the Audio library would have been in conflict with the OctoWS2811 library all these years too.

lpaolini, 2024-12-31:
I've tried instantiating AudioInputI2S in the setup() function, before or after invoking FastLED.addLeds(). When I do so there's no flickering, but about half of the strip gets stuck at 100% white.

KurtMF, 2024-12-31:
Ok.  WS2812Serial uses only 1 DMA, that could explain the difference.

There's one more thing you can try while we wait on an answer from Paul to my question over on PJRC forum.  This is to rule out FastLED as the cause, and verify it's between ObjectFled and Audio.

First, include ObjectFLED.h.
Comment out call to fastLED().
Right after you define your CRGB draw buffer, instantiate an ObjectFLED object (reference in readme)
In setup, call begin() on the object.
In loop, comment out calls to FastLED.show() and add call to your object.show()

Also, any calls to FastLEDs brightness or color balance functions must be replaced with calls to the equivalent functions on your object.  The rest of FastLED functions should work fine.

lpaolini, 2024-12-31:
Ok, I hope I got it right. Exact same behavior as before. [see here](https://photos.app.goo.gl/1CpBnYwL2c3tTEPs9)

```
#include <FastLED.h>
#include <ObjectFLED.h>
#include <Audio.h>

#define PIN 17
#define NUM_LEDS 192

CRGB leds[NUM_LEDS];
uint8_t pinList[1] = {PIN};
ObjectFLED fled(NUM_LEDS, leds, CORDER_GRB, 1, pinList);
AudioInputI2S audioInput;

void setup() {
    fled.begin();
}

void loop() {
    fled.show();
    delay(10);
}

```

KurtMF, 2024-12-31:
Ok.  Thanks for digging into it.  Your code looks correct, and I'm assuming what I saw on your light strip is incorrect, and that you can correct it now by simply commenting out the AudioInputI2S like before.  That confirms the conflict.

Since you already have ObjectFLED activated, please try one last simple thing... replace  
fled.begin(); 
with 
fled.begin( 1.0, 150 );

That sets extra long latch delay, to rule that out also.

I will dig into this again later, and try to figure out how DMAChannel.h is allocating channels to apps.  I will also try to install Audio and duplicate what you have on my Chinese WS2812b clones.

lpaolini, 2024-12-31:
> Your code looks correct, and I'm assuming what I saw on your light strip is incorrect, and that you can correct it now by simply commenting out the AudioInputI2S like before. That confirms the conflict.

Correct.

> Since you already have ObjectFLED activated, please try one last simple thing... replace fled.begin(); with fled.begin( 1.0, 150 );
> 
> That sets extra long latch delay, to rule that out also.

Same behavior.

> I will dig into this again later, and try to figure out how DMAChannel.h is allocating channels to apps. I will also try to install Audio and duplicate what you have on my Chinese WS2812b clones.

Thanks! Any chance to get it fixed by the end of the year? Just kidding :-)

KurtMF, 2024-12-31:
I found an old post in pjrc forum about Audio and Octo libraries not working together, but that turned out to be pin conflicts with Octo on Teensy 3.x, which did not allow use of all digital pins like Teensy 4.x.  

However, the guy got it working (8 years ago) by loading Audio.h after OctoWS2811.h (or ObjectFLED or FastLED)- and you are already doing that.  But he also instantiated his AudioInputI2S object before his OctoWS2811 object.  So this is another simple change you can try to the above code... try placing your AudioInputI2S declaration above the ObjectFLED (or addLEDs) lines.

And happy new year either way!  :-)

KurtMF, 2024-12-31:
Oh.. and I just read there may be a timer conflict as well... both audio and led libraries trying to use the same hw timer as DMA trigger.

lpaolini, 2025-01-01:
> However, the guy got it working (8 years ago) by loading Audio.h after OctoWS2811.h (or ObjectFLED or FastLED)- and you are already doing that. But he also instantiated his AudioInputI2S object [...]

No matter where I instantiate AudioInputI2S inside the setup() function, I get the second half of my strip stuck to full white (same behavior as with indirect ObjectFLED via FastLED).

Changing the order of the includes and/or of the lines instantiating AudioInputI2S and ObjectFLED seems to have no effect whatsoever.

> Also, the guy did report LED troubles when using Audio I2S for both output and input... and switched to just input and it worked. So... OctoWS2811/ObjectFLED want 3 dma channels, Audio uses 1 for input and 1 for output... it's the 5th allocation that breaks things. 

I'm not using audio output at all.

> Do you have something else trying to use a 5th DMA channel? When you try FastLED and ObjectFLED... is WS2811Serial commented out (and not included)?

The code is the one you see in the stripped-down test case - just FastLED, ObjectFLED and AudioInputI2S. I'm not even trying on my real code anymore since when I managed to reproduce the issue.

Just out of curiosity, can you briefly explain why three DMAs are required?

> Happy new year either way! :-)

HNY!

lpaolini, 2025-01-01:
> Audio, OctoWS2811 and WS2812Serial are all from Paul Stoffregen, I doubt they are conflicting. For sure WS2812Serial and Audio aren't, as I've been using them together for years.

[This post](https://forum.pjrc.com/index.php?threads/octows2811-adaptor-audio-adaptor-board.26016/post-49655) on the PJRC forum seems to confirm OctoWS2811 and Audio libraries can work together.

On the other hand, [this more recent post](https://forum.pjrc.com/index.php?threads/audio-i2s-quad-octo2811-conflict.35283/) seems to confirm there are some incompatibilities between the two libraries (and thus with ObjectFLED), and they smell a lot like DMA.

Perhaps @PaulStoffregen can shed some light on this.

KurtMF, 2025-01-01:
Thanks for the forum links.  I found a few others which also show that there is limited compatibility between Octo and Audio, depending on which Audio classes are used.  I've also posted about [this issue](https://forum.pjrc.com/index.php?threads/fastled-now-has-50-pin-parallel-teensy-4-1-ws2812-support.76316/), and waiting to hear something from Paul.

Meanwhile, I plan to install Audio.h and start searching the code for exactly which timers and DMA channels it uses, to see if this can be fixed in ObjectFLED.  I was planning on doing something with sound-driven LEDs for my next project, and I think it's important for all to have these 2 libraries work together.

3 channels are used to create the 3 parts of LED waveform which, like serial/uart protocol, uses start and stop bits.  But with LEDs, each bit has a start and stop bit.  So 1 DMA channel sends start bits, ch 2 sends the data bit, ch 3 sends stop bits.  Each channel is triggered by separate timer, so the timer values can be changed when calling begin() with timing specs, to allow the library to work with a wider range of LED chips which have different waveform timing specs (some LEDs have longer/shorter start/stop bits than others).  It is possible to use a single DMA channel to feed LEDs, but for that you only get a fixed waveform where start bit, stop bit, data bit each have equal time duration.  The advantage to having control of the waveform with 3 channels is that a wider variety of LED chips can work, plus you can overclock the LEDs more when the waveform shape better matches the LED spec sheet.

There are 16 DMA channels in a teensy 4x.  Only the first 4 are usable with timer trigger and GPIO mux mapping (needed by all 3 Octo DMAs).  I hope to learn that Audio doesn't need more than one of the first 4.

lpaolini, 2025-01-01:
Thanks for the explanation, much appreciated!

Ensuring compatibility with the Audio library is definitely worth it.
If you haven't come across it yet, have a look at the dedicated [design tool](https://www.pjrc.com/teensy/gui/), to get an idea of what you can do with it.

zackees, 2025-01-02:
@lpaolini https://github.com/KurtMF/ObjectFLED/issues/2#issuecomment-2567142225

zackees, 2025-01-06:
Hey @lpaolini - FastLED at the master branch now has Object FLED 1.10 in it. Can you please do me a solid and upgrade to confirm it fixes your issues?

@KurtMF

zackees, 2025-01-10:
Okay, unless anyone has reservations, I consider the ObjectFLED driver stable enough to introduce as a default.

Please let me know if this is a mistake.

KurtMF, 2025-01-10:
I agree.  Not a mistake.  Thanks for integrating ObjecFLED!

lpaolini, 2025-01-10:
Unfortunately, I still have problems.

Switching to ObjectFLED causes some offset issues in my setup, which seem to be caused by the fact that I'm using multiple strips of different lengths. Setting the same length for all fixes the problem.

If ObjectFLED requires all the strips to have the same length, then I won't be able to use it, and I'll have to stick to WS2812Serial.

Also, with the new timings some of the strips work well, while one of them still flickers (but it doesn't with WS2812Serial).

KurtMF, 2025-01-10:
Are you defining each string in separate object?  Or putting them in some other var-length memory structure?

At present, ObjectFLED only supports var-length strings as separate objects.  @zackees mentioned this, and we discussed adding a var-length string object to ObjectFLED.  This will allow you to show all in one command.  Or better yet, will give you ability to show all and individual strings.

I'm planning to add this for ObjectFLED 2.0.  I can't commit to a timeline for this right now, but soon I will need a technical diversion from doing all the soldering involved in connecting 4,096 thru-hole LEDs into a perfect cube.

zackees, 2025-01-10:
> > Are you defining each string in separate object? Or putting them in some other var-length memory structure?
> 
> Basically, my setup looks like this, where `SIZEx` is not constant.
> 
> ```c++
> CRGBArray<SIZE1> array1;
> CRGBArray<SIZE2> array2;
> CRGBArray<SIZE3> array3;
> CRGBArray<SIZE4> array4;
> 
> FastLED.addLeds<WS2812B, PIN1, GRB>(array1, SIZE1);
> FastLED.addLeds<WS2812B, PIN2, GRB>(array2, SIZE2);
> FastLED.addLeds<WS2812B, PIN3, GRB>(array3, SIZE3);
> FastLED.addLeds<WS2812B, PIN4, GRB>(array4, SIZE4);
> ```
> 
> > At present, ObjectFLED only supports var-length strings as separate objects. @zackees mentioned this, and we discussed adding a var-length string object to ObjectFLED. This will allow you to show all in one command. Or better yet, will give you ability to show all and individual strings.
> 
> Ok, I think I'll have to wait then.
> 
> > I'm planning to add this for ObjectFLED 2.0. I can't commit to a timeline for this right now, but soon I will need a technical diversion from doing all the soldering involved in connecting 4,096 thru-hole LEDs into a perfect cube.
> 
> I was thinking of something similar, but then I changed my mind. I'm not sure it's worth it, because it's a lot of work for 16 pixels of linear resolution, and I suspect it's not enough to get something really cool. This is just my opinion, I hope you're happy with the result.

This looks like my bug. On the backend I'm creating a rectangular buffer automatically and then passing that to ObjectFLED.

I'll investigate later today.

lpaolini, 2025-01-10:
Well, I could certainly instantiate CRGBArrays using the maximum length of any string, and then define a logical length for each string (i.e. the physical length), but that would require changes in my code and be a waste of memory (not in my case though, as I'm only using about 800 leds). It'd certainly be a cleaner solution to allow individual different string lengths to be used.

KurtMF, 2025-01-10:
I agree.  I'm thinking along the lines of adding StringFLED object, same signature as ObjectFLED, but there will be a stringShow(), to show all strings, in addition to .show() on each string.

zackees, 2025-01-11:
Hi there, I think I found the bug!

The issue was that I was pushing bytes back to the rectangular leds and forgot to account for the fill bytes, if the given strip was smaller than the max strip size.

Here's the cl that fixes it. I don't have a teensy 4.0 right now to test with, but the change is simple and compiles ok.

https://github.com/FastLED/FastLED/compare/master...fix-fled

I put this into a branch for you to test out @lpaolini 
https://github.com/FastLED/FastLED/tree/fix-fled

Thanks in advance.

lpaolini, 2025-01-11:
Hmm... the offset issue seems to be still there.

Also, I get a couple of unrelated warnings during compilation:
```
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (5.0.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.159.0 (1.59) 
 - tool-teensy @ 1.159.0 (1.59) 
 - toolchain-gccarmnoneeabi-teensy @ 1.110301.0 (11.3.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 95 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Audio @ 1.3.0+sha.bd7de87
|-- FastLED @ 3.9.10+sha.f89a6f3
|-- IRMP @ 3.6.4
|-- EEPROM @ 2.0
Building in release mode
Compiling .pio/build/teensy41/lib8e1/FastLED/FastLED.cpp.o
Compiling .pio/build/teensy41/lib8e1/FastLED/bitswap.cpp.o
...
Compiling .pio/build/teensy41/lib8e1/FastLED/platforms/esp/32/clockless_i2s_esp32s3.cpp.o
Compiling .pio/build/teensy41/lib8e1/FastLED/platforms/esp/32/experimental/s3_clockless_and_clocked_driver.cpp.o
.pio/libdeps/teensy41/FastLED/src/platforms/esp/32/clockless_i2s_esp32s3.cpp:3:2: warning: #warning "esp_memory_utils.h is not available, are you on esp-idf 4? The parallel clockless i2s driver will not be available" [-Wcpp]
    3 | #warning "esp_memory_utils.h is not available, are you on esp-idf 4? The parallel clockless i2s driver will not be available"
      |  ^~~~~~~
Compiling .pio/build/teensy41/lib8e1/FastLED/platforms/esp/32/rmt_4/idf4_rmt.cpp.o
Compiling .pio/build/teensy41/lib8e1/FastLED/platforms/esp/32/rmt_4/idf4_rmt_impl.cpp.o
...
Compiling .pio/build/teensy41/lib8e1/FastLED/sensors/digital_pin.cpp.o
Compiling .pio/build/teensy41/lib8e1/FastLED/sensors/pir.cpp.o
.pio/libdeps/teensy41/FastLED/src/platforms/arm/k20/clockless_objectfled.cpp: In member function 'void fl::ObjectFled::showPixels(uint8_t, PixelIterator&)':
.pio/libdeps/teensy41/FastLED/src/platforms/arm/k20/clockless_objectfled.cpp:170:14: warning: unused variable 'totalStrips' [-Wunused-variable]
  170 |     uint16_t totalStrips = group.mObjects.size();
      |              ^~~~~~~~~~~
Compiling .pio/build/teensy41/lib8e1/FastLED/simplex.cpp.o
Compiling .pio/build/teensy41/lib8e1/FastLED/third_party/espressif/led_strip/src/led_strip_api.c.o
...
```

zackees, 2025-01-11:
Thanks, fixed the warnings and pushed a new version.

Can you take a picture of the offset issue or a video? Thank you very much.

lpaolini, 2025-01-11:
Yay! I managed to isolate and reproduce the issue, I think.
Apparently there's a problem when the longest strip is not the first one to be added to FastLED.

I've prepared a couple of tests. The top strip is 100px, the bottom is 144px.
A couple of green leds are rendered at the beginning and at the end of each strip.

1) The longest strip (bottom) is instantiated first. So far so good.
https://photos.app.goo.gl/a2swQ6d6crYsdhbm7

2) The shortest strip (top) is instantiated first. Buffer overflow!
https://photos.app.goo.gl/omLEZgDa4BfGzRED8
Please note the two adjacent green leds in the top strip, showing that the second strip (longer) overflows into the first one (shorter).

zackees, 2025-01-12:
I spent the day refactoring the rectangular buffer, putting it under test and then re-writting the bulk led controllers.

This fixed the I2S for Esp32. I think it fixes teensy40 too.

I have to run now but the branch is updated with the latest. I did not test the the issue that you had. I only tested that it again displays leds like it should with the ino test example.

lpaolini, 2025-01-12:
Awesome, the overflow issue seems to be solved!
I've tried changing the order of initialization and it's working good.

However, one of the strips is still flickering.
Isn't the latest ObjectFLED driver supposed to be using the very same timings as WS2812Serial?

zackees, 2025-01-12:
> Awesome, the overflow issue seems to be solved! I've tried changing the order of initialization and it's working good.

Merged.

> However, one of the strips is still flickering. Isn't the latest ObjectFLED driver supposed to be using the very same timings as WS2812Serial?

Try this:

```C++
#define FASTLED_OVERCLOCK 1.2
#include "FastLED.h"
```

lpaolini, 2025-01-12:
When I do that, the already working strips continue to do so, while the flickering one shows noise.

KurtMF, 2025-01-12:
> Isn't the latest ObjectFLED driver supposed to be using the very same timings as WS2812Serial?

I studied WS2812Serial a while back.  They are using 5-bit patterns for each LED pulse (each serial byte + 2 start/stop bits = 2 LED pulses).  So their 1250 nS clock is divided into 1/5 for 0H and 4/5 for 1H; T0H = 250, T1H = 1000.  The equivalent ObjectFLED setup is .begin(1250, 250, 1000, 300);  //300 uS latch delay coded into WS2812Serial.

KurtMF, 2025-01-12:
I'm a bit confused by the square buffer problem.  When var length strings use a 2D array for buffer, and row length matches longest string, you draw each row according to known sting lengths, then pass the array to a display driver, which blindly sends full rows to each string.  Those strings which receive extra data from excess row size, will blindly ignore the extra data while waiting for the latch pulse.  Why is it necessary to do any processing on these over-sized buffer rows?

lpaolini, 2025-01-12:
> > Isn't the latest ObjectFLED driver supposed to be using the very same timings as WS2812Serial?
> 
> I studied WS2812Serial a while back. They are using 5-bit patterns for each LED pulse (each serial byte + 2 start/stop bits = 2 LED pulses). So their 1250 nS clock is divided into 1/5 for 0H and 4/5 for 1H; T0H = 250, T1H = 1000. The equivalent ObjectFLED setup is .begin(1250, 250, 1000, 300); //300 uS latch delay coded into WS2812Serial.
> 
> If I'm not mistaken, your static problem was fixed when Zach updated FastLED to 1.1.0 ObjectFLED, right? Now it's broke again even with 1.2 OC factor, which also once fixed this problem?

Before the fix I was having flickering on any string. With latest version only on a couple of older ones, which I'm using in my setup. However, with WS2812Serial even those old clunkers worked good :-)

zackees, 2025-01-12:
> I'm a bit confused by the square buffer problem.  When var length strings use a 2D array for buffer, and row length matches longest string, you draw each row according to known sting lengths, then [...]

Can you elaborate

KurtMF, 2025-01-12:
Ok, I have 2 strings, one 50 LEDs one 100.  I use CRGB[2][100] array to draw both strings, then call show() for the entire array.  Driver sends 100 * 3 bytes to each string, then a latch pulse to tell the LEDs to latch their register data to the ADCs which drive the actual LED (turns the lights on).

Even though there's only 50 LEDs on the short string, during transmission of 100*3 bytes of draw buffer, the short string only stores the first 50*3 bytes into their registers.  The remaining 50*3 bytes are passed onto the next LED all the way to the end where that last LED is not connected.  The last 50*3 bytes are ignored by every LED on the string because they each already got populated with the first 50*3 bytes.  At the end of 100*3 bytes transmitted on each string, the latch pulse comes and the color data is displayed as drawn in the array.

So why does this 2D array need refactoring or special handling?  How does it cause a problem?

zackees, 2025-01-12:
> Ok, I have 2 strings, one 50 LEDs one 100. I use CRGB[2][100] array to draw both strings, then call show() for the entire array. Driver sends 100 * 3 bytes to each string, then a latch pulse to tell the LEDs to latch their register data to the DACs which drive the actual LED (turns the lights on).

This is what is being done internally.

> So why does this 2D array need refactoring or special handling? How does it cause a problem?

I got lazy and implemented it wrong. I was using vector push back and forgot to add extra fill bytes.

I put this all under test then used some typing to enforce correct usage.

zackees, 2025-01-13:
https://github.com/FastLED/FastLED/blob/master/src/fl/rectangular_draw_buffer.h

KurtMF, 2025-01-13:
Ah vectors.  So I guess you bypass ObjectFLED.genFrameBuffer() and create your own (FastLED) framebuffer in vectors?

Just looked at the header file.  Looks complicated.  :-)

zackees, 2025-01-13:
>  So I guess you bypass ObjectFLED.genFrameBuffer() and create your own (FastLED) framebuffer in vectors?

No I don't bypass it.

I just use more memory.

zackees, 2025-01-13:
Also FYI, FastLED completely ignores all of the extra modes and just does RGB mode only. RGBW mode is supported by extending the buffer, adding an extra pixel if needed, then sending all the RGBW data down the pipe as if it were RGB data. Works great and we get all the RGBW fastled functions to work.

This is how I bolted on RGBW onto the Yves S3 I2S driver.

zackees, 2025-01-13:
Okay, I minted a new release 3.9.11 with all the fixes.

Tomorrow I'm going to ask again if ObjectFLED is stable enough to be the main driver, so that we can make it the default for 3.9.12.

closed 2025-01-13 by zackees