FastLED/FastLED #1827: RMT channel not available

munkerench, 2024-12-31:
Updated to latest 3.9.8

Now getting this trying to run 2 strips 

E (6246) rmt: rmt_tx_register_to_group(132): no free tx channels
E (6247) rmt: rmt_new_tx_channel(239): register channel failed
E (6247) led_strip: led_strip_refresh_async(42): invalid argument
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42009d51
file: "c:\..\arduino\libraries\fastled\src\platforms\esp\32\led_strip\rmt_strip.cpp" line 269
func: void fastled_rmt51_strip::RmtLedStripNoRecycle::draw_async()
expression: led_strip_refresh_async(mLedStrip)

abort() was called at PC 0x4037dc2f on core 1

Backtrace: 0x40377666:0x3fcebc90 0x4037dc39:0x3fcebcb0 0x40383989:0x3fcebcd0 0x4037dc2f:0x3fcebd50 0x42009d51:0x3fcebd80 0x4200a076:0x3fcebda0 0x42003355:0x3fcebdc0 0x42008ba2:0x3fcebde0 0x42004491:0x3fcebe00 0x4200716d:0x3fcebe30 0x4201b323:0x3fcebe70 0x4038076a:0x3fcebe90

-------------------------------------------------------------------------
I tried setting up the RMT with This config 

void configureRmtChannels() {
	rmt_channel_handle_t tx_channels[3] = { NULL };
	gpio_num_t tx_gpio_number[3] = { GPIO_NUM_21, GPIO_NUM_6, GPIO_NUM_7 };  // pin numbers
	for (int i = 0; i < 3; i++) {
		Serial.printf("setting up channel %d\n", i);
		rmt_tx_channel_config_t tx_chan_config = {
		  .gpio_num = tx_gpio_number[i],
		  .clk_src = RMT_CLK_SRC_DEFAULT,
		  .resolution_hz = 10 * 1000 * 1000,  // 10MHz clock
		  // fix for problem with more than two Tx channels see
		  //  https://esp32.com/viewtopic.php?f=13&t=42301&p=139024#p139024
		  #if 0
				.mem_block_symbols = 64,
		  #endif
				.mem_block_symbols = 48,
				.trans_queue_depth = 1,
		};
		ESP_ERROR_CHECK(rmt_new_tx_channel(&tx_chan_config, &tx_channels[i]));
	}
	rmt_transmit_config_t transmitConfig = {
	  .loop_count = -1
	};

	ESP_ERROR_CHECK(rmt_enable(tx_channels[2]));
	ESP_ERROR_CHECK(rmt_enable(tx_channels[1]));
	ESP_ERROR_CHECK(rmt_enable(tx_channels[0]));

}
-------------------------------------------------------------------------
and the error changed to 

setting up channel 0
setting up channel 1
setting up channel 2
E (5750) rmt: rmt_new_tx_channel(254): intr_priority conflict
E (5751) led_strip_rmt: led_strip_new_rmt_device_with_buffer(266): create RMT channel failed
E (5757) rtm_strip.cpp: construct_led_strip failed because of unexpected error, is DMA not supported on this device?: ESP_ERR_INVALID_ARG
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42009dc7
file: "c:\users\pcopp\onedrive\documents\arduino\libraries\fastled\src\platforms\esp\32\led_strip\rmt_strip.cpp" line 219
func: void fastled_rmt51_strip::RmtLedStripNoRecycle::acquire_rmt_if_necessary()
expression: err

abort() was called at PC 0x4037dc2f on core 1

Backtrace: 0x40377666:0x3fcebc60 0x4037dc39:0x3fcebc80 0x40383989:0x3fcebca0 0x4037dc2f:0x3fcebd20 0x42009dc7:0x3fcebd50 0x42009e0e:0x3fcebd80 0x4200a14e:0x3fcebda0 0x42003385:0x3fcebdc0 0x42008c7a:0x3fcebde0 0x42004585:0x3fcebe00 0x42007241:0x3fcebe30 0x4201b3fb:0x3fcebe70 0x4038076a:0x3fcebe90

ELF file SHA256: ee340cca7686e7a8
-------------------------------------------------------------------------

Which takes us beyond my capabilities.  Glad to help with some grunt work!

zackees, 2024-12-31:
Thanks for the offer to help.

I'm going to guess that your chipset does not have DMA.

I'm going to guess that the following build define will fix the issue:

`-DFASTLED_RMT_WITH_DMA=0`

Are you building with platformio?

munkerench, 2024-12-31:
Chip is an ESP32 s3.  Building with Visual Studio (Not VScode).  I have another strip of leds that work fine if I do not instantiate the second strip.

Made the recommended changes, builds fine, but still getting failure

E (4748) rmt: rmt_tx_register_to_group(132): no free tx channels
E (4749) rmt: rmt_new_tx_channel(239): register channel failed
E (4749) led_strip: led_strip_refresh_async(42): invalid argument
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42009d51
file: "c:\...\arduino\libraries\fastled\src\platforms\esp\32\led_strip\rmt_strip.cpp" line 269
func: void fastled_rmt51_strip::RmtLedStripNoRecycle::draw_async()
expression: led_strip_refresh_async(mLedStrip)

abort() was called at PC 0x4037dc2f on core 1

Backtrace: 0x40377666:0x3fcebd20 0x4037dc39:0x3fcebd40 0x40383989:0x3fcebd60 0x4037dc2f:0x3fcebde0 0x42009d51:0x3fcebe10 0x4200a076:0x3fcebe30 0x42003355:0x3fcebe50 0x42008ba2:0x3fcebe70 0x42004491:0x3fcebe90 0x4200716d:0x3fcebec0 0x4201b323:0x3fcebf00 0x4038076a:0x3fcebf20

munkerench, 2024-12-31:
After a little more debugging, I get the below.  I think I need to turn DMA back on.

setting up channel 0
setting up channel 1
setting up channel 2
E (5847) rmt: rmt_new_tx_channel(254): intr_priority conflict
E (5848) led_strip_rmt: led_strip_new_rmt_device_with_buffer(266): create RMT channel failed
E (5854) rtm_strip.cpp: construct_led_strip failed because of unexpected error, is DMA not supported on this device?: ESP_ERR_INVALID_ARG
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42009dc7
file: "c:\...\arduino\libraries\fastled\src\platforms\esp\32\led_strip\rmt_strip.cpp" line 219
func: void fastled_rmt51_strip::RmtLedStripNoRecycle::acquire_rmt_if_necessary()
expression: err

abort() was called at PC 0x4037dc2f on core 1

zackees, 2025-01-01:
How did you turn dma off? I didn't know you could do that

munkerench, 2025-01-01:
I set the flag you gave; it had no effect, so I removed it.     Still not able to get the 2nd strip to run.

zackees, 2025-01-04:
Don't use it as an include flag, use it as a build flag.

If you are on ArduinoIDE you can't do this. But you can if you use PlatformIO it's trivial: https://github.com/fastLED/Platformio-starter

munkerench, 2025-01-05:
I added it to the preprocessor defs in Visual studio (Project-> Properties -> C++ - Preprocessor.  It did not work.

zackees, 2025-01-05:
You are the first one that i've known to build with visual studio instead of VSCode, are you using a manual download of our lib?

munkerench, 2025-01-05:
I have a extension from VMicro that sets up an arduino like environment, but with all the visual studio stuff.  Pretty useful.

zackees, 2025-01-06:
Good news is that I was able to reproduce the issue. I'm going to rebase off of the newest arduino strip. We tried to do something fancy involving moving strips to different pins but that experiment didn't work out. So I'm going to rebase and also stick closer to espressifs led strip api.

zackees, 2025-01-06:
You are welcome. I'll ping this thread when the master branch has the changes in it.

zackees, 2025-01-09:
Fixed in master. Will release in the next 24 hours.

closed 2025-01-09 by zackees

munkerench, 2025-01-16:
Verified in the specific failed case.  Thanks!