FastLED/FastLED #1830: Cannot compile example of I2S driver for the ESP32-S3

tresler, 2025-01-05:
I was already running original [LED driver from hpwit](https://github.com/hpwit/I2SClockLessLedDriveresp32s3). 
I tried the insert version directly into FastLED strictly by example, but I can't compile:

```.ino:61:undefined reference to `_ZN2fl17InternalI2SDriver6createEv'```

zackees, 2025-01-06:
Hi, I've been having issues with the cpp file always getting compiled for S3. For some reason, for example, my xiao ESP32S3 wasn't hitting the `CONFIG_IDF_TARGET_ESP32S3` define. I don't know why this is. But right now I'm adding additional boards manually to overcome this, such as:

```C++
#if defined(CONFIG_IDF_TARGET_ESP32S3) || defined(ARDUINO_XIAO_ESP32S3)
```

Let me know your board type (preferably platformio.ini if you have that, ok if you don't) and I'll manually add in your chip into the white list to trigger the driver to be compiled in.

tresler, 2025-01-06:
I use classic ESP32S3-wroom modules, namely N16R8.

lmancini, 2025-01-08:
@zackees coming from this thread on the subreddit (https://www.reddit.com/r/FastLED/comments/1htm0th/comment/m5jkoi1/), I spent some time to try and figure out this issue. Let's start from the g++ command that PIO uses to compile module files:

```
xtensa-esp32s3-elf-g++ -o .pio/build/esp32-s3-devkitm-1/src/main.cpp.o
-c -std=gnu++2b -fexceptions -fno-rtti -Os -mlongcalls -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -freorder-blocks -Wwrite-strings -fstack-protector -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -MMD -DPLATFORMIO=60117 -DARDUINO_ESP32S3_DEV -DARDUINO_USB_MODE=1 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DESP_PLATFORM -DIDF_VER=\"v5.1.4-586-gb6b4727c58-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -DTF_LITE_STATIC_MEMORY -DARDUINO_ARCH_ESP32 -DESP32 -DF_CPU=240000000L -DARDUINO=10812 -DARDUINO_VARIANT=\"esp32s3\" "-DARDUINO_BOARD=\"Espressif ESP32-S3-DevKitM-1\"" -DARDUINO_PARTITION_default -Iinclude -Isrc -I.pio/libdeps/esp32-s3-devkitm-1/FastLED/src -IC:/Users/Utente/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src
(a long list of `-I` follows)
```

That define is not specified on the compilation command line.

So I thought it has to come from some include in the core, and made a couple main.cpp test files:

In the first test file, compiling **doesn't** cause the `#error` to fire.

```cpp
#if defined(CONFIG_IDF_TARGET_ESP32S3)
#error
#endif

#include <Arduino.h>

void setup() {}
void loop() {}
```

In the second test file, compiles **causes** the `#error` to fire.

```cpp
#include <Arduino.h>

#if defined(CONFIG_IDF_TARGET_ESP32S3)
#error
#endif

void setup() {}
void loop() {}
```

Including both Arduino.h or FastLED.h works the same, so I assume that the define is coming from the include chain. I'm going to open a PR that applies this approach in clockless_i2s_esp32s3.cpp in order to use `CONFIG_IDF_TARGET_ESP32S3`.

With that modification I'm able to compile the example. Then the executable crashes because apparently my ESP32 S3 is a model without PSRAM, but that's another issue.

closed 2025-01-09 by zackees

lmancini, 2025-01-09:
@zackees not off the top of my head, but I have the setup ready to test any header very quickly. Are you thinking about any specific header?

zackees, 2025-01-09:
Thank you so much.

Can you try this header:

`#include "eorder.h"`

Thanks again.

lmancini, 2025-01-09:
@zackees including that one doesn't cause `CONFIG_IDF_TARGET_ESP32S3` to be defined.

I tried to set some #warning traps like the ones from my previous comments, and realized that the include that causes the definition seems to be either `cmath` or `math.h`. It's very difficult to follow the include path on the border between FastLED and the IDF framework, though, so I might be missing something obvious.

zackees, 2025-01-10:
I think i got it.

One has to include

`#include "sdkconfig.h"`

Which is what arduino.h is doing. It took me hours to figure this out, saddly.

tresler, 2025-01-18:
I have new knowledge about driver with new arduino-esp32 version 3.1.1. Only the first 8 outputs work with this version. With older arduino-esp32 3.0.7 all 16 outputs work well.

commit aee63ca3 (2025-01-09): Fix #1830: include the main library header file in order to use the CONFIG_IDF_TARGET_* flags