FastLED/FastLED #1833: Remove unnecessary null pointer checks

elfring, 2025-01-08:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [RmtController5](https://github.com/FastLED/FastLED/blob/3fca27c2486e62cfb1b7ef0aab5c8864dac1cea7/src/platforms/esp/32/rmt/idf5_rmt.cpp#L55-L59 "Destructor for RmtController5")
- [led_strip_rmt_del](https://github.com/FastLED/FastLED/blob/3fca27c2486e62cfb1b7ef0aab5c8864dac1cea7/src/platforms/esp/32/led_strip/led_strip_rmt_dev.cpp#L139-L141 "led_strip_rmt_del function")
- [showRGBInternal](https://github.com/FastLED/FastLED/blob/3fca27c2486e62cfb1b7ef0aab5c8864dac1cea7/src/platforms/arm/rp2040/clockless_arm_rp2040.h#L266-L267 "showRGBInternal function")

closed 2025-01-09 by zackees

elfring, 2025-01-09:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability/delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

zackees, 2025-01-09:
We already run cppcheck. High and medium scale issues are already auto-detected. Due to bugs in clang tidy, running it on our source code breaks it. So it's on the exclusion list. Also, FastLED is one of the most tested arduino projects in existence. I'm interested in meaningful issues that can affect stability, which a nullptr check before delete does not do and is probably optimized away by the compiler anyway.

elfring, 2025-01-09:
:thought_balloon: I am curious if another software developer (besides me) can get into the mood to apply a corresponding update suggestion which could be generated by [the software “Coccinelle”](https://coccinelle.gitlabpages.inria.fr/website/ "Source code transformation with Coccinelle") (also with the help of a variant from the following script).

```cocci
@Remove_unnecessary_pointer_checks@
expression x;
@@
-if (\(x != 0 \| x != NULL\))
    free(x);
```

zackees, 2025-01-09:
You are going around other repo's and doing the same thing to them: telling them about nullptr checks before delete. This isn't even useful and wastes time. Your account looks like bot activity so I've reported your account to github.

elfring, 2025-01-09:
:eyes: I dare to point selected remaining improvement possibilities out in various development repositories.

elfring, 2025-01-09:
There are developers who can handle presented change suggestions in more constructive ways. :thinking: