FastLED/FastLED #1835: RMT5 issues on ESP32 and ESP32-S3

RococoN8R, 2025-01-08:
Hi @zackees and everyone working on the FastLED enhancements. I love seeing new momentum given the challenges since we lost @focalintent. I ran a few tests using a project I'm working on and while I can't share the code, I thought it would be helpful to share a few measurements. 

Here's my setup:
Environment: platformio in vscode
Platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10/platform-espressif32.zip
build_flags = -D FASTLED_RMT5=1 (experiments with FASTLED_ESP32_RECYCLE=0 or 1 do not impact results)
lib_deps = 
	mathieucarbou/AsyncTCP@^3.2.12
	mathieucarbou/ESPAsyncWebServer@^3.3.21
	bblanchon/ArduinoJson@^7.2.1
	fastled/FastLED@3.9.9
	knolleary/PubSubClient@^2.8.0
For both modules, my config uses 5 - 10 pins with WS2812B and WS2811 pixels. Up to 8 pins have been used in FastLED versions 3.6.0 and 3.7.8 with the 6.9.0 version of the platform, and FastLED.show() execution times are significantly longer than cited below. Each pin is connected to 100 LEDs, and there is no mixing of WS2812B and WS2811 LEDs across pins (each MCU only uses one type currently). I call FastLED.show() 50 times per second, measure the start to finish time in microseconds for each call and average across the 50 calls.

Observations using ESP-WROOM-32 on NodeMCU-32S module: (WS2811)
Pins Configured      Avg Execution time (uS)
1                              238
2                              426
4                              740
5 or more                RMT errors as reported by others
- First observation, pins are not running in parallel
- Second, this device fails consistently with 5 or more pins
- Third, if I assume 60uS reset time, this indicates ~1.9uS/LED, which seems reasonably close depending on data sent

Using the same setup with an ESP32-S3-WROOM-1 on ESP32-S3-DevKitC-1 v1.1 (using different device pins)
Pins Configured      Avg Execution time (uS)
1                              378
2 or more                RMT errors as reported by others
- Single pin takes significantly more time to show() than on the ESP32 (378 vs 238 uS)
- This device fails with 2 or more pins configure

I'm not suited to debug the code, but I'm hoping this data can bring some visibility and influence your rebasing efforts.

zackees, 2025-01-09:
Fixed. Will be released soon.

It was a bug in the espressif demo, they manually set RMT symbol words to 64, when it should have been 48 on ESP32S3. Surprisingly, this disables half the TX RMT channels.

It will be in the next release, which will happen in the next 24 hours. My local test setup shows all 4 channels working in parallel and it's at the expected speed.

zackees, 2025-01-09:
I want to add, there's no way we are getting 8 channels back with RMT5. Half the channels are RX only. I don't know why this is but it sucks. On the bright side the version enables clockless spi mode, which is always DMA backed and is supposedly extremely stable. This also enables the very cheap ESP32-C2 chipset to now drive clockless LEDS.

closed 2025-01-09 by zackees

RococoN8R, 2025-01-09:
Thanks for the insights into the issues, @zackees , very helpful. My year-round outdoor display uses several ESP32's with 5 - 8 active pins each to maximize refresh rates. Given the 4 ch TX limits on RMT5, it's time I explore the I2S approach. Since it is non-blocking, is there a callback or flag set when FastLED.show() completes sending to all of the pins?

zackees, 2025-01-09:
> Given the 4 ch TX limits on RMT5, it's time I explore the I2S approach. Since it is non-blocking, is there a callback or flag set when FastLED.show() completes sending to all of the pins?

No notification is provided to know when the strip has finished rendering. Also RMT5 is also async, a difference from RMT4. Additionally, FastLED provides two additional channels of super stable async SPI drivers drive WS2812 leds.

You can run both SPI and RMT together. But doing so means you'll have to construct the SPI led driver yourself and add it to the FastLED singleton.

If you add up the I2S, RMT and spi then the theoretical parallel count is 16 + 4 + 2 = 22 strips. However, I've never tried it myself. The only downside to the I2S driver is that it requires a rectangular buffer where the width is the size of the longest strip. But there is so much memory that in practice it's a drop in the bucket.