FastLED/FastLED #1839: Compilation error in v3.9.10 due to unresolved dependency on Arduino IDE

ssr-yuki, 2025-01-11:
Hi, I use this via [M5Unified](https://github.com/m5stack/M5Unified) package with Arduino IDE 2.3.4.

In the the latest release v3.9.10, the following compilation error occurs:

```
In file included from c:\Users\{USER}\Documents\Arduino\libraries\FastLED\src/third_party/yves/I2SClockLessLedDriveresp32s3/driver.h:9,
                 from c:\Users\{USER}\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_i2s_esp32s3.cpp:6:
c:\Users\{USER}\Documents\Arduino\libraries\FastLED\src/third_party/yves/I2SClockLessLedDriveresp32s3/src/I2SClockLessLedDriveresp32s3.h:8:10: fatal error: esp_memory_utils.h: No such file or directory
 #include "esp_memory_utils.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
```

Downgrading to v3.9.10 resolves this problem ad hoc.

The inclusion is added on ebf69454d853848f4582a564b0c69c860dcaa4a6 , which looks to depending on [esp-idf](https://github.com/espressif/esp-idf). For the future release, I suggest you to revert the dependency change or update the dependency list explicitly.

Thank you for maintaining this package.

theelims, 2025-01-11:
I had this issue also when downgrading to v3.9.9. I had to revert back to v3.9.8.

lmancini, 2025-01-12:
@ssr-yuki @R4m1-F1n @theelims I'm not familiar with the M5 devices, but which core version (https://github.com/espressif/arduino-esp32) are you using?

I was getting this same error, and then realised that I was stuck for ages at version 2.0.17. If it's possible for you to upgrade at 3.1.1 (I understand this might not be the case for several reasons), that version ships `esp_memory_utils.h`.

If you don't see that as an available core version, be sure to include `https://espressif.github.io/arduino-esp32/package_esp32_index.json` in Preferences -> Additional boards manager URLs. For some reason it won't fetch by itself without this URL, and the most recent version displayed will be 3.0.7.

Hope this helps!

ssr-yuki, 2025-01-12:
@lmancini Thank you for advising. Following your comments, today I have updated `esp32` from v3.0.7 to v3.1.1 with Arduino IDE boards manager, restarted the IDE updated and updated all libraries accordingly. But unfortunately the same error occurred again.

Downgrading `FastLED` to v3.9.9 is only one promising way for me at this time. I will report you if I find something more.

lmancini, 2025-01-12:
> I have updated esp32 from v3.0.7 to v3.1.1 with Arduino IDE
> But unfortunately the same error occurred again.

That's really weird, can you compare your Arduino IDE Boards Manager sidebar with the one below?

[image]

Queequack, 2025-01-12:
Same problem here. I work with PlatformIO. Tried 2 boards with FastLED today (both have PSRAM, each with main difference in the config): 

- ESP32-S3-Zero (ESP32-S3FH4R2) --> **Works with latest FastLED v3.9.10**
```
board_build.partitions=default.csv
board_build.arduino.memory_type = qio_qspi
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
```
- ESP32-DevKitC-1-N16R8v --> **Only works after downgrade to FastLED v3.9.9**
```
board_build.arduino.memory_type = qio_opi
board_build.prsam_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
```

It took me a while to get the right configuration (platformio.ini) for both boards. If you are interested: [ESP32-S3-Zero](https://github.com/platformio/platform-espressif32/issues/1529), [ESP32-DevKitC-1-N16R8v (Proposal #4 in this Post)](https://community.platformio.org/t/new-esp32-s3-n16r8/41568/5). Both are based on the esp32-s3-devkitc-1 board. Both are clones from Aliexpress.

Error message after the build process:

```
In file included from .pio/libdeps/RoomPanel_esp32-s3-devkitc-1-n16r8v/FastLED/src/third_party/yves/I2SClockLessLedDriveresp32s3/driver.h:9,
                 from .pio/libdeps/RoomPanel_esp32-s3-devkitc-1-n16r8v/FastLED/src/platforms/esp/32/clockless_i2s_esp32s3.cpp:6:
.pio/libdeps/RoomPanel_esp32-s3-devkitc-1-n16r8v/FastLED/src/third_party/yves/I2SClockLessLedDriveresp32s3/src/I2SClockLessLedDriveresp32s3.h:8:10: fatal error: esp_memory_utils.h: No such file or directory

**************************************************************************
* Looking for esp_memory_utils.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_memory_utils.h"
* Web  > https://registry.platformio.org/search?q=header:esp_memory_utils.h
*
**************************************************************************

 #include "esp_memory_utils.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

ssr-yuki, 2025-01-12:
> Have you tried to install core v3.1.1, FastLED v3.9.10, and only compile the standard FastLED Blink example? That should compile without errors.

Yes and no. I found that compilation results depend on the board settings: if I choose `esp32` borads such as "ESP32{*} Dev Module," no error occurred; the reported error appears only with `M5Stack` boards.

I am investigating the possibility that this compilation issue is caused by M5 library dependencies.

[image]
[image]

zackees, 2025-01-12:
Pushing out a fix today.

zackees, 2025-01-12:
For those curious what's going on - a header made it above the include guard for a cpp file and is being compiled on non esp32 chips.

zackees, 2025-01-13:
3.9.11 is pushed to Arduino for release. If you want the fix now, feel free to manually download our library and install it.

closed 2025-01-13 by zackees