FastLED/FastLED #1846: Regression: 3.9.10+, WS2812B, ESP32C3: heavy flickering when using interrupts

osnwt, 2025-01-14:
It seems there is a regression with ESP32C3 RMT driver in FastLED 3.9.10+ when interrupts are used.

I have a simple application which controls a WS2812B LED panel by commands from a BLE remote controller (made for media player, but used to change LED panel modes). It worked well on ESP32C3 until FastLED update 3.9.10 or above. After that update I see heavy flickering when BLE stack is active. As far as I see, ESP32 RMT driver uses interrupts and BLE stack uses them either. So some changes in ISR may be a root cause.

I tried to use `#define FASTLED_ESP32_USE_CLOCKLESS_SPI` but with it I have:

```
CORRUPT HEAP: Bad head at 0x3fca7520. Expected 0xabba1234 got 0x00000003
assert failed: multi_heap_free multi_heap_poisoning.c:279 (head != NULL)
```

I also tried to go back to RMT4 by `#define FASTLED_RMT5 0`, but have no library objects (looks like only RMT5 is compiled):
```
undefined reference to `RmtController::RmtController(int, int, int, int, int, bool)'
...
Using library ArduinoLog at version 1.1.1 in folder: /Users/os/Projects/arduino/libraries/ArduinoLog 
Using library FastLED at version 3.9.11 in folder: /Users/os/Projects/arduino/libraries/FastLED 
Using library BLE at version 3.1.1 in folder: /Users/os/Library/Arduino15/packages/esp32/hardware/esp32/3.1.1/libraries/BLE 
```

I tried to compile the same project for ESP32 core with 3.9.11 and had no issues. The difference is that ESP32 has `#define FASTLED_ESP32_HAS_RMT5 0` but ESP32C3 has `#define FASTLED_ESP32_HAS_RMT5 1`. So they use different drivers. But in 3.9.9 everything worked well.

So the questions are:

1. Has anybody noticed such behavior? It seems to be a regression since 3.9.10. Both 3.9.8 or 3.9.9 work well.
2. Has anybody issues with FASTLED_ESP32_USE_CLOCKLESS_SPI with single-core ESP32C3 module?
3. Can I use old driver with new version, say, RMT4? If yes, how to do that not patching the library code?

I have no currently minimal code at hand to reproduce the issue (lack of time). But may add it later if really necessarily.

zackees, 2025-01-15:
The problem with RMT4 is that if it merely links with RMT5 then the espressif framework will force a boot crash.

Right now the RMT5 is using memory defaults, while RMT4 doubled up on the memory. Can you do me a favor and download this custom branch of FastLED with the RMT5 memory doubled? You may find that it's more stable than RMT4

[FastLED-rmt5-less-flicker.zip](https://github.com/user-attachments/files/18417418/FastLED-rmt5-less-flicker.zip)

Or use this branch:

https://github.com/FastLED/FastLED/tree/rmt5-less-flicker

If this doesn't work, let me know and I'll give you instructions on how to force RMT4. The issue is that you are *probably* trying to disable RMT5 through a header define in your sketch, and that's not going to work.

zackees, 2025-01-15:
Also, upgrade to 3.9.11. There was some cleanup that happened in the RMT driver layer on our end.

osnwt, 2025-01-15:
> Also, upgrade to 3.9.11. There was some cleanup that happened in the RMT driver layer on our end.

I tried both 3.9.10 and 3.9.11, they behave identically with that regard. That's why "since 3.9.10" or "3.9.10+".

As for the custom branch, **I'll try ASAP**, and yes - I tried to do that in sketch, and of course it did not rebuild the library itself. My concern is that if I have to change the library to build flicker-free code, I can also simply stay on 3.9.9. And that's a regression, to my mind...

zackees, 2025-01-15:
Weird that 3.9.9 works for you. Can you take a video if the branch doesn't work for you?

osnwt, 2025-01-15:
Your branch works perfectly!

sutaburosu, 2025-01-15:
I know the rmt5-less-flicker branch is for C3 but I tried it anyway on S3. In case it's useful to know, it crashes on show().

```log
E (112) rmt: rmt_tx_register_to_group(152): no free tx channels
E (112) rmt: rmt_new_tx_channel(288): register channel failed
E (113) led_strip_rmt: led_strip_new_rmt_device(181): create RMT TX channel failed
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42006e8d
  #0  0x42006e8d in IRmtStrip::Create(int, unsigned long, bool, unsigned long, unsigned long, unsigned long, unsigned long, unsigned long, IRmtStrip::DmaMode, unsigned char) at .pio/libdeps/esp32-s3-n16r8/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:63 (discriminator 2)
      (inlined by) RmtStrip at .pio/libdeps/esp32-s3-n16r8/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:76 (discriminator 2)
      (inlined by) IRmtStrip::Create(int, unsigned long, bool, unsigned long, unsigned long, unsigned long, unsigned long, unsigned long, IRmtStrip::DmaMode, unsigned char) at .pio/libdeps/esp32-s3-n16r8/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp:169 (discriminator 2)

file: ".pio/libdeps/esp32-s3-n16r8/FastLED/src/platforms/esp/32/rmt_5/strip_rmt.cpp"WiF line 63
func: led_strip_t* {anonymous}::configure_led_with_timings(int, uint32_t, bool, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, bool, uint8_t)
expression: led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip)

abort() was called at PC 0x4037e5f7 on core 1

Backtrace: 0x40377216:0x3fcae520 0x4037e601:0x3fcae540 0x40384f69:0x3fcae560 0x4037e5f7:0x3fcae5e0 0x42006e8d:0x3fcae610 0x42006bd0:0x3fcae680 0x42007f12:0x3fcae6d0 0x420137d9:0x3fcae750 0x4200340a:0x3fcae7a0 0x42013d2d:0x3fcae7c0 0x42020b54:0x3fcae860 0x4037f316:0x3fcae880
  #0  0x42013d2d in CFastLED::show() at .pio/libdeps/esp32-s3-n16r8/FastLED/src/FastLED.h:735
      (inlined by) loop() at src/main.cpp:128
  #1  0x42020b54 in loopTask(void*) at ~/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74
  #2  0x4037f316 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

zackees, 2025-01-16:
Yeah, this is not surprising. I double the memory S3.

I will only double the memory for the C3 then.

sutaburosu, 2025-01-16:
My understanding is that the 48 RMT words (on S3) per channel are the total. If you want to ping-pong buffer it, each half must be 24 words. There is a dedicated area of RAM on the die for this, so it's not configurable.

closed 2025-01-17 by zackees

alex2501, 2025-01-19:
@zackees I am having the same issue with all versions after 3.9.9, but on a ESP32-S3.
Even tried latest master, but not working, only going back to 3.9.9 works.

zackees, 2025-01-20:
> @zackees I am having the same issue with all versions after 3.9.9, but on a ESP32-S3.
> Even tried latest master, but not working, only going back to 3.9.9 works.

You can double the memory for S3 but you will only get two RMT channels.

This will need to be a build define and not an include define.

As an alternative, I recommend using the I2C protocol which was added in 3.9.10, it will give you 16 channels and is very stable.

You can also use the new SPI protocol. It's also super stable and uses DMA backed memory.

Sorry about this, S3 has this weird thing with the RMT channels. We added these protocols to give you a work around to these issues.

See the examples list to see an example of the I2S, or the readme to find an image that explains how to do it.

kokospalme, 2025-04-23:
The less flicker version didn't work on my C3. my ```platformio.ini```:
```
[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
board_build.mcu = esp32c3
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
board_build.partitions = no_ota.csv
board_upload.flash_size = 4MB
board_build.filesystem = SPIFFS
framework = arduino
build_flags =
    -DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
    ; -DFASTLED_ALLOW_INTERRUPTS=0
    ; -DFASTLED_INTERRUPT_RETRY_COUNT=3
    ; -DINTERRUPT_THRESHOLD=0
upload_port = /dev/cu.usbmodem114401
monitor_port = /dev/cu.usbmodem114401
monitor_speed = 115200
lib_deps = 
	https://github.com/kokospalme/Artnet
    https://github.com/kokospalme/DMXnow
    Preferences @ 2.0.0
    ; fastled/FastLED @ 3.9.9
    https://github.com/FastLED/FastLED.git
```

Could downgrading the whole esp32 core be the answer?

kokospalme, 2025-04-23:
[image]

This are lines in ```src/third_party/espressif/led_strip/src/enabled.h```. Shouldn't ```FASTLED_RMT5``` be 1?
Why is that?

kokospalme, 2025-04-23:
> > [@zackees](https://github.com/zackees) 
> As an alternative, I recommend using the I2S protocol which was added in 3.9.10, it will give you 16 channels and is very stable.
> 
> You can also use the new SPI protocol. It's also super stable and uses DMA backed memory.

Is this also the case for the C3?
When I try to Use the I2S driver with```#define FASTLED_ESP32_I2S``` I get various compiler errors:
```
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:75:50: error: 'DR_REG_I2S_BASE' was not declared in this scope
 #define REG_I2S_BASE(i)                         (DR_REG_I2S_BASE + (i) * 0x1E000)
                                                  ^~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:81:34: note: in definition of macro 'ETS_UNCACHED_ADDR'
 #define ETS_UNCACHED_ADDR(addr) (addr)
                                  ^~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:182:14: note: in expansion of macro 'WRITE_PERI_REG'
             (WRITE_PERI_REG((reg),(READ_PERI_REG(reg)&(~((bit_map)<<(shift))))|(((value) & bit_map)<<(shift)) ));      \
              ^~~~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:5: note: in expansion of macro 'SET_PERI_REG_BITS'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
     ^~~~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/i2s_reg.h:74:38: note: in expansion of macro 'REG_I2S_BASE'
 #define I2S_INT_ENA_REG(i)          (REG_I2S_BASE(i) + 0x0014)
                                      ^~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:23: note: in expansion of macro 'I2S_INT_ENA_REG'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
                       ^~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:75:50: note: suggested alternative: 'DR_REG_I2S0_BASE'
 #define REG_I2S_BASE(i)                         (DR_REG_I2S_BASE + (i) * 0x1E000)
                                                  ^~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:81:34: note: in definition of macro 'ETS_UNCACHED_ADDR'
 #define ETS_UNCACHED_ADDR(addr) (addr)
                                  ^~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:182:14: note: in expansion of macro 'WRITE_PERI_REG'
             (WRITE_PERI_REG((reg),(READ_PERI_REG(reg)&(~((bit_map)<<(shift))))|(((value) & bit_map)<<(shift)) ));      \
              ^~~~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:5: note: in expansion of macro 'SET_PERI_REG_BITS'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
     ^~~~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/i2s_reg.h:74:38: note: in expansion of macro 'REG_I2S_BASE'
 #define I2S_INT_ENA_REG(i)          (REG_I2S_BASE(i) + 0x0014)
                                      ^~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:23: note: in expansion of macro 'I2S_INT_ENA_REG'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
                       ^~~~~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:52: error: 'I2S_OUT_EOF_INT_ENA_V' was not declared in this scope
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
                                                    ^~~~~~~~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:157:76: note: in definition of macro 'WRITE_PERI_REG'
             (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
                                                                            ^~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:5: note: in expansion of macro 'SET_PERI_REG_BITS'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
     ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:413:52: note: suggested alternative: 'I2S_TX_DONE_INT_ENA_V'
     SET_PERI_REG_BITS(I2S_INT_ENA_REG(i2s_device), I2S_OUT_EOF_INT_ENA_V, 1,
                                                    ^~~~~~~~~~~~~~~~~~~~~
/Users/UserXY/.platformio/packages/framework-arduinoespressif32@3.20006.221224/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/soc.h:157:76: note: in definition of macro 'WRITE_PERI_REG'
             (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
                                                                            ^~~
[... cut ...]
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:495:10: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
     i2s->conf.rx_start = 0;
          ^~~~
          rx_conf
.pio/libdeps/esp32-c3-devkitm-1/FastLED/src/platforms/esp/32/i2s.hpp:496:10: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
     i2s->conf.tx_start = 0;
          ^~~~
          rx_conf
```

Have I overlooked something?

Thank you very much!

zackees, 2025-04-23:
`FASTLED_ESP32_I2S` only works on the ESP32, the S3 variant is called `FASTLED_USES_ESP32S3_I2S`

The I2S feature is not implemented for any other esp platform than these two.

The reason that there is a regression has to do with the RMT5 driver that came with the forced idf upgrade to arduino core. 

The RMT5 driver is based on the esp led_strip component library that I rebased off of. Unfortunately this library has a bunch of issues like performance.

You can force FastLED to bind to the RMT4 driver is way more stable thanks to a lot of low level custom code that @samguyer put in. There is a compiler switch to force it back to RMT4. You'll want to use `FASTLED_RMT5=0` as a **build define** and not as just an include define, or else it won't work.

Essentially, my strategy for the ESP at this point is to concentrate on the bulk drivers so that users have an escape hatch. The I2S driver can drive 16 strips in parallel and users are reporting it as very stable.

Unfortunately the C3 and C6 have a lot of flicker and there is nothing that will be done about it in the near time frame. Just spend the extra $2 and get the S3 variable. It's the top of the line chip from espressif and you really shouldn't be doing serious LED work on any other chip. If you are cost sensitive then go with the C2 and use the SPI controller. Btw, you should try the SPI controller for the C3, it might give you better results.

kokospalme, 2025-04-24:
Thank you a lot for this tipps and getting this whole project step by step further! I'm using Fastled for about 5 years now and it's impressive how it evolved...

Unfortunately my usecase is indeed a little cost and size critical, so I'll get some ESP32 C2 boards and see what these can do. But can you explain why the ESP8684 (this is the C2, right?) should have less problems than the C3?

kokospalme, 2025-04-25:
I'm very sorry, my ATX powersupply was faulty. It worked for a couple of month but seems to be broken now, as I can now draw only about 20% of the max. current. Everything above let drop the voltage significantly. This caused the flickering. I just tested the code with a bench power supply and it works great not...
Thank you anyways for your support

zackees, 2025-04-26:
Glad everything is working great for you. Yeah the older chips are really great. The new flagship s3 can't drive as many leds as the good old esp32dev