FastLED/FastLED #1855: Feature Request: support WS2816 LEDs

kbob, 2025-01-20:
The World Semi WS2816 family LEDs are high dynamic range, high refresh rate RGB pixels.  They receive 16 bits each of red, green, and blue, 48 bits total.  They have hardware gamma correction as well.  They have a high refresh rate at 10,000 Hz.  And they use a clockless daisy-chained communication protocol similar to the WS2812, just with twice as many bits per pixel.

http://world-semi.com/ws2816-family/

It would be wonderful if FastLED could support these chips.

kbob, 2025-01-20:
It would be a big architectural change to add more pixel datatypes to FastLED such as  RGB16.  FastLED already supports RGBW by expanding RGB into RGBW at pixel refresh time, and it supports APA102-HD's I+RGB the same way.

So it would be possible to do the same with WS2816's RGB16.  Multiplying 8 bits of brightness, 8 bits of RGB, and 8 bits of color correction gives 24 bits, which can be scaled down to 16 bits.  That would give less resolution than true 16 bit color, but it would allow a high dynamic range -- even with brightness == 1, RGB would have 8 bit precision.

In terms of hardware support, a scheme similar to RGBWEmulated should work.  A "virtual" WS2816 controller layers on top of and calls into the platform-specific WS2812 controller.  It converts each 48 bit pixel into two 24 bit pixels and lets the WS2812 driver deliver 24 bit pixels.  That should work across the whole range of platforms(?)  (Actually, I'm kind of sketchy on how FastLED drives parallel outputs like Octo2811.)

zackees, 2025-01-21:
This is good code and exactly the way it should be done. I'll start integrating this.

zackees, 2025-01-21:
can you upload a copy of their data sheet for the WS2816? It looks like it wants me to enter in my email to get it but it's all in chinese.

zackees, 2025-01-21:
Okay I got it. Your algorithm is not complete however. You are doing a linear power ramp from 8->16 bit. Then there is a 4-bit gamma correction inside the chipset. However, this can be improved by doing a partial gamma correction from 8->16 that is then complimented by the gamma correction from the chipset.

However, the math on this is going to be a little bit insane on this and will need unit tests. So I'm going to take your code, fix it up a little, then mark the controller as "beta" and come back to it later.

kbob, 2025-01-22:
Um...  I started testing this on Teensy 4, and it doesn't work with the ObjectFLED driver.  I'm not sure why yet, but ObjectFLED outputs all zero bits.  So you might want to hold off on integrating this.  I am debugging.

I don't understand how you hope to improve on the hardware gamma, so I guess I will watch and learn. (-:

kbob, 2025-01-22:
The data sheet is here.
https://www.lcsc.com/datasheet/lcsc_datasheet_2412021803_Worldsemi-WS2816C-2121_C965561.pdf

zackees, 2025-01-22:
> started testing this on Teensy 4, and it doesn't work with the ObjectFLED driver. I'm not sure why yet, but ObjectFLED outputs all zero bits. So you might want to hold off on integrating this. I am [...]

Oh no, WS2816 just got pushed.

kbob, 2025-01-22:
Wow, you're fast.  Okay, it's a new feature, and it works on some platforms.  Not ideal, but not a regression.

Meanwhile, the many versions and implementations of show, showLeds, showLedsInternal, et al are a twisty little maze.

zackees, 2025-01-23:
> Wow, you're fast. Okay, it's a new feature, and it works on some platforms. Not ideal, but not a regression.
> 
> Meanwhile, the many versions and implementations of show, showLeds, showLedsInternal, et al are a twisty little maze.

Yup and before they were all the same overloaded name.

mcer12, 2025-01-23:
@zackees From a quick glance over the PR, do I understand it correctly that at this point, that the implementation works with 8bit color values and just mapping those to 16bit so it's not taking advantage of the full 16bit resolution?

kbob, 2025-01-23:
It multiplies the 8 bit color channel, the 8 bit brightness, and the 8 bit color correction, and converts that to a 16 bit value.  So yes, hue is limited to 8 bit resolution.  But intensity is effectively 16 bits.  (actually 15-ish because of quantization from calling `scale16by8` twice)

kbob, 2025-01-23:
Maybe I could do better than that.  I could compute this.

    uint16_t color16 = ((uint32_t)color8 * (uint32_t)brightness * (uint32_t)correction) >> 8;

I didn't do that because it dims slightly. (255 * 255 * 255) >> 8 == 64770, not 65535.  What you want is this.

    uint16_t color16 = ((uint32_t)color8 * (uint32_t)brightness * (uint32_t)correction) / 255;

But that's an expensive division, and the corresponding multiplication requires more than 8 bits to be accurate, and would therefore overflow a `uint32_t`.

Maybe instead of "16 bit color" we should advertise "quite a bit better than 8 bit color". (-:

sutaburosu, 2025-01-23:
Rather than `a * b * c` you could `(a * b + a + b) * c + c` 

```cpp
uint16_t color16 = ((uint32_t)color8 * (uint32_t)brightness + color8 + brightness) * \
                    (uint32_t)correction + correction) >> 8;
```

gives `((255 * 255 + 255 + 255) * 255 + 255 ) / 256 == 65280`. 65280 is 255 << 8.

This is probably unnecessary, but to extend the range to 65535 you could also add `color16 += color16 >> 8`.

zackees, 2025-01-23:
Correct, and the chances of the core library going to a full CRGB16 format is very slim, it's simply too much code in the codebase. Other attempts to just add a +W component have never been completed.

The simplest and most impactful way to address this 16-bit resolution is to figure out how to do partial gamma correction on the software side so that when it get's passed to the 16 bit format, the 4-bit gamma correction native to the chipset doesn't overcorrect.

Mathematically it makes sense, but I think best results will come from eye-ing and adjusting it.

My strategy with FastLED is to have it the best possible library for the majority of use cases. And if you are an absolute performance freak, then you'll need to drop to a slightly lower level in FastLED to achieve it. And we hope of course, that you merge your changes back.

closed 2025-01-24 by zackees