FastLED/FastLED #1856: ESP32 RMT4 driver block FastLED.show()

Jueff, 2025-01-21:
Having that FastLED controller configuration

CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0,230); 
CLEDController& controller1 = FastLED.addLeds<NEOPIXEL, 32>(leds+230,  1); 
CLEDController& controller2 = FastLED.addLeds<NEOPIXEL, 16>(leds+231,  3); 
CLEDController& controller3 = FastLED.addLeds<NEOPIXEL, 14>(leds+234,  32); 
CLEDController& controller4 = FastLED.addLeds<NEOPIXEL, 18>(leds+266,  2); 

FastLED.show() block after some seconds.

Platform: ESP32 Arduino
Board: 1.0.4
FastLED: 3.9.9

When removing controller4 no more blocks occur. 
When downgrading to 3.7.0 no more blocks occur.

The sample code below reproduces the problem.
It also runs an extra task on other core to "revives' FastLED by calling rmt drivers GiveGTX_sem().

```
#include <FastLED.h>

#define NUM_LEDS 268

// Define the array of leds
CRGB leds[NUM_LEDS];
TaskHandle_t  taskHnd;
uint32_t lastShow = 0;
uint32_t lastBlink = 0;
uint32_t lastLog = 0;
uint32_t cnt = 0;
uint32_t failCount = 0;
bool displayLedStartMsg = true;
bool displayCheckStartMsg = true;
bool ledIsOn;

extern void IRAM_ATTR GiveGTX_sem() __attribute__((weak));
void setup() { 
  Serial.begin(115200);

  CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0,230); 
  CLEDController& controller1 = FastLED.addLeds<NEOPIXEL, 32>(leds+230,  1); 
  CLEDController& controller2 = FastLED.addLeds<NEOPIXEL, 16>(leds+231,  3); 
  CLEDController& controller3 = FastLED.addLeds<NEOPIXEL, 14>(leds+234,  32); 

  // **********************************************************
  // when commenting out the controller4 FastLED.show() won't block
  // **********************************************************

  CLEDController& controller4 = FastLED.addLeds<NEOPIXEL, 18>(leds+266,  2); 
  
  // create a task that monitor the alive state of the ESP
  // commenting out the next line displayed the alive check, FastLED.show() will then block until reboot
  xTaskCreatePinnedToCore(OtherCoreTask,"Task",4096,NULL,1,&taskHnd,0);

  delay (1);
}

void loopFastLED() 
{
    if (displayLedStartMsg)
    {
      displayLedStartMsg = false;
      Serial.printf("loopFastLED is running on core %d\n", xPortGetCoreID());
    }
    lastShow = millis();
    if ((millis()-lastBlink)>500)
    {
        ledIsOn = !ledIsOn;
        lastBlink = millis();
        for (int i=0;i<NUM_LEDS;i++)
        { 
          leds[i].r = ledIsOn ? 10 : 0;
        }
        Serial.println();
    }
    FastLED.show();
    Serial.print('.');
    cnt++;
}

void loopCheck() 
{ 
  if (displayCheckStartMsg)
  {
      displayCheckStartMsg = false;
      Serial.printf("loopCheck is running on core %d\n", xPortGetCoreID());
  }

  if (lastShow!=0 && (millis()-lastShow)>500)         // FASTLed is blocked
  {
      failCount++;
      if (GiveGTX_sem!=NULL) GiveGTX_sem();
      Serial.printf("\r\n**** FASTLed hang detected: failure Time %4d:%02d:%02d, FASTLed failures %d *******\r\n", (int)(millis()/(1000*60*60*24)), (int)(millis()/(1000*60*60) % 24), (int)(millis()/(1000*60) % 60), failCount);
      lastShow = 0;
  }
  if ((millis()-lastLog)>=10000)
  {
       lastLog=millis();
       Serial.printf("\r\nUptime %4d:%02d:%02d, FASTLed failures %d, updates in 10 seconds: %d\r\n", lastLog/(1000*60*60*24), lastLog/(1000*60*60) % 24, lastLog/(1000*60) % 60, failCount, cnt);
       cnt=0;
  }
}

void loop() 
{
    loopFastLED();
}

void OtherCoreTask(void * parameter)
{
  while(1) 
  {
    yield();
    loopCheck();
    delay (1);
  }
}

```

Jueff, 2025-01-21:
tried compiling the code with

Board: 3.1.1
FastLED: 3.9.11

also doesn't work, but different to previous test it shows repeated error messages

```
E (91677) rmt: rmt_new_tx_channel(288): register channel failed
.E (91683) rmt: rmt_tx_register_to_group(152): no free tx channels
E (91688) rmt: rmt_new_tx_channel(288): register channel failed
.E (91694) rmt: rmt_tx_register_to_group(152): no free tx channels
E (91700) rmt: rmt_new_tx_channel(288): register channel failed
.E (91706) rmt: rmt_tx_register_to_group(152): no free tx channels
E (91711) rmt: rmt_new_tx_channel(288): register channel failed
.E (91717) rmt: rmt_tx_register_to_group(152): no free tx channels
E (91723) rmt: rmt_new_tx_channel(288): register channel failed
.E (91729) rmt: rmt_tx_register_to_group(152): no free tx channels

```

Jueff, 2025-01-21:
Problem seems to be that startNext is called from ISR. startNext (and called functions) don't have IRAM_ATTR.
I'm working on a fix, where startNext is called from main thread, not from ISR.
could also fix #1438.

zackees, 2025-01-21:
RMT sucks.

We have an I2S driver for both ESP32 and ESP32-S3 that is much more stable and drives a *lot* of leds. Try using this.

Jueff, 2025-01-21:
In my project a cannot ensure that all controllers use the same chipset.

Jueff, 2025-01-22:
[idf4_rmt_impl.zip](https://github.com/user-attachments/files/18498640/idf4_rmt_impl.zip)

Attached you find an improved code, that works good here. Long-run testing will start tomorrow.

Changes:
1.) gTxSem is now triggered every time a channel has finished sending data. Adding a waiting controller to the freed channel is no long done within the interrupt.
Now inside showPixel method a loop takes care about waiting for all controllers to be finished and/or assigning controllers to channel that got free. 
2.) I protect access to RMT registers with a lock

adding only 1.) wasn't enough, still had stability issues
on the other hand adding 2.) to the original code where channels are counted/assigned in ISR also didn't work.

zackees, 2025-01-22:
Can you send a pull request?

Jueff, 2025-01-24:
I also did some long term tests with ESP_IDF_VERSION 4. 
Without my changes the ESP still blocks at least 2 time in 12h.
So I would suggest to enable the new behavior by default.

Should I change the pull request?

zackees, 2025-01-24:
> Should I change the pull request?

Yes. Your previous changes have all gone in, I believe. Please rebase to master and then push your new changes.

Jueff, 2025-01-25:
Just saw that ESP32RMTController::showPixel has the IRAM_ATTR attribute. 
ESP32RMTController::showPixel is called from RmtController::showPixels, which doesn't have IRAM_ATTR  set.
Looking at the other ESP showPixel implementations they all don't have that attribute. 

Q: Is showPixel intended to be called from within ISR?
I think not, and so I suggest to remove IRAM_ATTR  from ESP32RMTController::showPixel

zackees, 2025-01-25:
I agree. Can you send me a PR for this?

closed 2025-01-25 by zackees