FastLED/FastLED #1863: Add pin definitions for ATtiny13

zackees, 2025-01-24:
I was gifted partial pin defintions, but these are missing the third parameter which should be A,B,C etc..

I'm looking for help on what these should be

```
#define MAX_PIN 5

_FL_DEFPIN(0, 0); // PB0 (MOSI/AIN0/OC0A/PCINT0)
_FL_DEFPIN(1, 1); // PB1 (MISO/AIN1/OC0B/INT0/PCINT1)
_FL_DEFPIN(2, 2); // PB2 (SCK/ADC1/T0/PCINT2)
_FL_DEFPIN(3, 3); // PB3 (PCINT3/CLKI/ADC3)
_FL_DEFPIN(4, 4); // PB4 (PCINT4/ADC2)
_FL_DEFPIN(5, 5); // PB5 (PCINT5/RESET/ADC0/dW)

#define HAS_HARDWARE_PIN_SUPPORT 1
```

sutaburosu, 2025-01-24:
The comments in that snippet suggest all the pins are on IO port B: PB0 is port B bit 0. A quick check of the datasheet confirms that the ATtiny13 only has pins on IO port B. 

So the third parameter should be "B" for all of them.

zackees, 2025-01-27:
Okay, pin definitions are in, but the platform still won't link. I'm getting an error message about a bad opcode. But at least we made some progress.

closed 2025-01-27 by zackees