FastLED/FastLED #1864: Add pin definitions for ATtiny4313, we have the partial pin definitions here

zackees, 2025-01-24:
ATtiny4313 Pin Definitions

```
#define MAX_PIN 19

_FL_DEFPIN(0, 0); // PA0 (ADC0/PCINT0)
_FL_DEFPIN(1, 1); // PA1 (ADC1/PCINT1)
_FL_DEFPIN(2, 2); // PA2 (ADC2/PCINT2)
_FL_DEFPIN(3, 3); // PD0 (RXD/PCINT16)
_FL_DEFPIN(4, 4); // PD1 (TXD/PCINT17)
_FL_DEFPIN(5, 5); // PD2 (INT0/PCINT18)
_FL_DEFPIN(6, 6); // PD3 (INT1/PCINT19)
_FL_DEFPIN(7, 7); // PD4 (T0/XCK/PCINT20)
_FL_DEFPIN(8, 8); // PD5 (T1/PCINT21)
_FL_DEFPIN(9, 9); // PD6 (AIN0/PCINT22)
_FL_DEFPIN(10, 10); // GND (Ground)
_FL_DEFPIN(11, 11); // PB0 (ICP/PCINT8)
_FL_DEFPIN(12, 12); // PB1 (OC0A/PCINT9)
_FL_DEFPIN(13, 13); // PB2 (SS/OC0B/PCINT10)
_FL_DEFPIN(14, 14); // PB3 (MOSI/OC1A/PCINT11)
_FL_DEFPIN(15, 15); // PB4 (MISO/OC1B/PCINT12)
_FL_DEFPIN(16, 16); // PB5 (SCK/PCINT13)
_FL_DEFPIN(17, 17); // PB6 (XTAL1/PCINT14)
_FL_DEFPIN(18, 18); // PB7 (XTAL2/PCINT15)
_FL_DEFPIN(19, 19); // VCC (Power Supply)
```

sutaburosu, 2025-01-24:
I have checked this list against the datasheet, and it appears to be correct.

The third parameter can be taken from the comments. "PBx" is port B, and so on. VCC and GND need to be removed though; they are not IO pins.

sutaburosu, 2025-01-25:
Reflecting on this, I don't think I was adequately clear. The 2nd parameter is the bit number, so can't be more than 7. The list should look like this:

```cpp
_FL_DEFPIN(0, 0, A); // PA0 (ADC0/PCINT0)
_FL_DEFPIN(1, 1, A); // PA1 (ADC1/PCINT1)
_FL_DEFPIN(2, 2, A); // PA2 (ADC2/PCINT2)

_FL_DEFPIN(3, 0, D); // PD0 (RXD/PCINT16)
_FL_DEFPIN(4, 1, D); // PD1 (TXD/PCINT17)
_FL_DEFPIN(5, 2, D); // PD2 (INT0/PCINT18)
_FL_DEFPIN(6, 3, D); // PD3 (INT1/PCINT19)
_FL_DEFPIN(7, 4, D); // PD4 (T0/XCK/PCINT20)
_FL_DEFPIN(8, 5, D); // PD5 (T1/PCINT21)
_FL_DEFPIN(9, 6, D); // PD6 (AIN0/PCINT22)

_FL_DEFPIN(11, 0, B); // PB0 (ICP/PCINT8)
_FL_DEFPIN(12, 1, B); // PB1 (OC0A/PCINT9)
_FL_DEFPIN(13, 2, B); // PB2 (SS/OC0B/PCINT10)
_FL_DEFPIN(14, 3, B); // PB3 (MOSI/OC1A/PCINT11)
_FL_DEFPIN(15, 4, B); // PB4 (MISO/OC1B/PCINT12)
_FL_DEFPIN(16, 5, B); // PB5 (SCK/PCINT13)
_FL_DEFPIN(17, 6, B); // PB6 (XTAL1/PCINT14)
_FL_DEFPIN(18, 7, B); // PB7 (XTAL2/PCINT15)
```

zackees, 2025-01-25:
You rock!

closed 2025-01-27 by zackees