FastLED/FastLED #1865: 1624 with clock speed 10Mhz instead 20Mhz wrong color

DietmarM, 2025-01-24:
on the attiny with arduino using 20Mhz led work fine.
Changing the setting to 10Mhz the Leds shine white only.

Any idea what is wrong, how I can fix it?

sutaburosu, 2025-01-24:
It [was reported in 2020](https://github.com/FastLED/FastLED/issues/716#issuecomment-647146670) that less than 16MHz clock on AVR can be problematic.

If I were in your position, I would start by investigating what values are coming from the macros which calculate the clock ticks for T1,T2,T3. Are they sane? Do they leave enough ticks for the other instructions that are interleaved with the pin twiddling in show() on AVR?

zackees, 2025-01-24:
Nothing we can do about it. The AVR stuff is hand coded assembly to squeeze as much performance out of the AVR chipset as possible. You are going below the threshold of stability and now your colors are changing because 0's and 1's are starting to flip.

If you want to investigate further:

  * See if one of the previous FastLED versions works. In this case, this is our regression.
  * You discover that the adafruit neopixel library works, in which case we may copy them.

Daniel, the person who wrote that assembly, died tragically. I have near zero skills with assembly, so there is nothing I can do about it if the current code is working as intended.

closed 2025-07-29 by zackees