FastLED/FastLED #1867: D1 Mini Clone crashes

ohanahandyman, 2025-01-24:
D1 Mini Clone crashes when pubsubclient establishes a mqtt connection, when fastled library 3.9.12 is loaded. No code is necessary for fastled just the library needs to be included.  You can use the example code for pubsubclient and mqtt.  Fastled 3.7.8 works NP.

zackees, 2025-01-24:
Yes, we've had issues with this board.

I have no idea why a recent version will fail. Though it's possible you are running out of memory.

Believe it or not, most of the support for these boards comes externally from our user base now.

Here is how you can help me help you:

  * Try to find the first version of FastLED that starts failing.
  * Better yet, bisect the commit where it starts failing, this will give us all the information we need to figure out the culprit
  * Upload an example sketch
  * Bonus: convert your sketch to PlatformIO, as this has total build information, unlike ArduinoIDE which hides everything.
    * We have a guide to do this, it's very easy and everyone who does it thanks me profusely for increasing their coding velocity. I promise you you will never go back. It's 1-1 compatible with ArduinoIDE so once you convert, you can compile your sketch any way you want. https://github.com/FastLED/PlatformIO-Starter

goddade, 2025-02-11:
Same problem.

yelvieman, 2025-02-15:
Hi - it appears to be this commit https://github.com/FastLED/FastLED/commit/03bc3e3e75d3556082eb679fce28829ea8c0307b where it starts failing.

zackees, 2025-02-15:
Wow thanks!

Any log files you can share?

yelvieman, 2025-02-16:
[fl.zip](https://github.com/user-attachments/files/18814666/fl.zip)

Attached is a zip file with the main.cpp & platformio.ini for a minimal app that shows the crash and the 2 commits either side of 1 working and 1 failing. I've also included the output.

Hope this helps

zackees, 2025-03-03:
Okay

yelvieman, 2025-03-08:
Yeah, release 3.9.14 has fixed the issue. Thank you

closed 2025-03-08 by zackees