FastLED/FastLED #1868: WS2816 LED not working with ESP32S3 I2S controller

kbob, 2025-01-25:
When I drive WS2816 LEDs with the ESP32S3 I2S controller, the LEDs usually come on solid white.

Looking at the generated waveforms, it looks like the timing is far enough out of spec that the chip reads every bit as  '1'.

This is a sequence 01000... as transmitted by the I2S controller.  The 0 bit is on for just over 400 ns.  The WS2816 spec calls for 200-320 ns on time.

[image]

For reference, here is the ESP32S3 IDF5 RMT controller.  The 0 bit pulse is shorter, and the LED does the right thing.

[image]

nomakewan, 2025-01-25:
Is this based on 3.9.12 or the current code in `main`?

kbob, 2025-01-25:
Current code in main.

kbob, 2025-01-25:
I should mention that the WS2816 seems finickier than most clockless LEDs.  It needs a higher signal voltage too.

zackees, 2025-01-25:
People have been using this driver with success. Can you try the previous version of FastLED and confirm the bug is also present before current version on your platform?

zackees, 2025-01-25:
Right, the I2S Yves driver has baked in timings for the WS2812 and those timings are different from what FastLED uses. Ours timings are optimized and battle tested. Yves choose less optimal timings.

With a little tweaking it can be made to work with WS2816. I don't have this led chipset so I can't test.

zackees, 2025-01-27:
Wow this code is hard to read.

However overclocking the chipset is actually very easy, and this might push the 1 bit closer to 0 enough to actually work.

If you want to test, adjust this:

`#define CLOCKLESS_PIXEL_CLOCK_HZ (24 * 100 * 1000)`

zackees, 2025-01-27:
I've asked Yves, the creator of the I2S driver, for advice on how to tighten up T1H

kbob, 2025-01-28:
Interesting results.

Setting CLOCKLESS_PIXEL_CLOCK_HZ to 3.1 MHz seems to work.  On my LED strip, with my wiring.

At that speed, the timings are

* T1 = 312 ns  (in spec)
* T1 + T2 = 626 ns (in spec)
* T1 + T2 + t3 = 936 ns (too fast: spec is 1000-1520 ns)

But there's a caveat.  In src/third_party/yves/I2SClocklessLedDriveresp32s3.h, the color order is hardcoded.

    #define COLOR_ORDER_RBG

(*N.B.* that's RBG, not GRB, not RGB.)  And that breaks the WS2816 which wants RGB.  I changed the \#define to hardcode RGB, and that maps the colors correctly.  (Breaks every other LED type, though.)

So that's information.  Not information that points toward a fix, though.

zackees, 2025-01-28:
Breaks every other LED type? FastLED should be doing the remapping just in time when it sends the pixels.

kbob, 2025-01-28:
I mean that to get WS2816 to work, I hardcoded the I2S driver's color order to RGB.  It was hardcoded to RBG (sic) for other LEDs.

See https://github.com/FastLED/FastLED/commit/13e0d82b9805a9f11fb4193994f39332be3c1608

The s4 i2s driver needs to be parameterized on color order, I think.

zackees, 2025-03-11:
Any update on this?

kbob, 2025-03-11:
Not from me.  I am not planning to work on the i2s driver.

zackees, 2025-03-11:
> Not from me.  I am not planning to work on the i2s driver.

I don't understand how the RGB order can be bad. Obviously this was tested and you are the first one to report otherwise.

zackees, 2025-05-10:
The FASTLED_OVERCLOCK has been ported to work with the I2S driver for ESP32S3.

You will need to define it as a build define, not an include define.

If someone could try this and tell me if this works for then then I can issue a general fix which won't need this hack.

zackees, 2026-04-18:
Update: the new ESP32-S3 channels-API driver with **wave8** encoding (`src/platforms/esp/32/drivers/i2s/wave8_encoder_i2s.h`) should now handle WS2816 timing correctly. Key difference vs. the old Yves I2S driver: **8 pulses per data bit instead of 3**, which gives ~125 ns resolution at the default 8 MHz PCLK instead of the ~400 ns steps that were putting the 0-bit out of spec.

### Timing breakdown — wave8 at 8 MHz PCLK

| Bit | Pattern (MSB→LSB) | T_H | T_L | Total | WS2816 spec |
|-----|------------------|-----|-----|-------|-------------|
| 0   | `0b11000000`     | 250 ns | 750 ns | 1000 ns | T0H **200–320 ns** ✓ |
| 1   | `0b11111100`     | 750 ns | 250 ns | 1000 ns | T1H 580–1000 ns ✓ |

### Comparison to previous measurements in this thread

@kbob measured the out-of-spec driver producing a 400+ ns low-bit-high time (too long). The overclock-to-3.1-MHz workaround got you to T1=312 ns, T1+T2=626 ns, T1+T2+T3=936 ns (period slightly short). Wave8 at default PCLK hits **T0H=250 ns, T1H=750 ns, period=1000 ns** without overclocking — all three numbers land inside the WS2816 window.

### Resolution headroom

Each pulse is 125 ns at 8 MHz, so if a given chipset variant needs different pulse counts (e.g. tighter T0H), you can tune `ChipsetTiming` and `calculateI2sClockHz()` will pick the matching PCLK; the 8-pulse granularity means ~125 ns timing steps instead of the old ~400 ns steps.

If anyone on this thread still has WS2816 hardware available, a re-test against current `main` using the channels-API driver would be great — based on the timing math above it should just work at default timings.

closed 2026-04-18 by zackees