FastLED/FastLED #1873: Multiple strip led declaration would trigger reboot of device on ESP32-C3 (LOLIN and XIAO)

Jf7Dee, 2025-01-26:
Using XIAO ESP32-C3 and LOLIN ESP32-C3 PICO, having one single strip led declared works fine.
But adding another one makes it so ESP32 would keep on rebooting

1) works fine
 ` FastLED.addLeds<LED_TYPE, ONBOARD_LED_PIN, ONBOARD_COLOR_ORDER>(led_onBoard, ON_BOARD_NUM_LEDS);  
`
2) will reboot constantly
```
  FastLED.addLeds<LED_TYPE, ONBOARD_LED_PIN, ONBOARD_COLOR_ORDER>(led_onBoard, ON_BOARD_NUM_LEDS);  
  FastLED.addLeds<LED_TYPE, OUTSIDE_DATA_PIN, OUTSIDE_COLOR_ORDER>(leds, ON_BOARD_NUM_LEDS);

```

Behaviour was not found on Arduino Uno, ESP32 DevKitC V4 nor XIAO ESP32-S3
Rolling bak to Fastled 3.9.11 "fixed" the issue

Reboot was observed using serial monitor: 

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381894
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1144
load:0x403cc710,len:0xad8
load:0x403ce710,len:0x2f80
entry 0x403cc710
 readyE (1104) rmt: rmt_tx_register_to_group(152): no free tx channels
E (1105) rmt: rmt_new_tx_channel(288): register channel failed
E (1105) led_strip_rmt: led_strip_new_rmt_device(189): create RMT TX channel failed
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x42001ee0
file: "c:\Users\Admin\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\rmt_5\strip_rmt.cpp" line 63
func: led_strip_t* {anonymous}::configure_led_with_timings(int, uint32_t, bool, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, bool, uint8_t)
expression: led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip)

abort() was called at PC 0x40385b25 on core 0
Core  0 register dump:
MEPC    : 0x40381346  RA      : 0x40385b30  SP      : 0x3fc99460  GP      : 0x3fc8de00  
TP      : 0x3fc996d0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fc9948c  S1      : 0x3fc9948c  A0      : 0x3fc99498  A1      : 0x3fc9947a  
A2      : 0x00000000  A3      : 0x3fc994c5  A4      : 0x00000001  A5      : 0x3fc90000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000001  S3      : 0x00000118  
S4      : 0x00000177  S5      : 0x000003e8  S6      : 0x0000036b  S7      : 0x000000fa  
S8      : 0x00000009  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000 
```

zackees, 2025-01-26:
Are you building with platformio? Confirm the issue is happening because of 

```C++
#ifndef LED_STRIP_RMT_DEFAULT_MEM_BLOCK_SYMBOLS_MULTIPLIER
#if CONFIG_IDF_TARGET_ESP32C3
// Fixes https://github.com/FastLED/FastLED/issues/1846
#define LED_STRIP_RMT_DEFAULT_MEM_BLOCK_SYMBOLS_MULTIPLIER 2 <----
#else
#define LED_STRIP_RMT_DEFAULT_MEM_BLOCK_SYMBOLS_MULTIPLIER 1
#endif
#endif
```

Set this 1 and see if your problem goes away. If so, then guru meditation is required.

Jf7Dee, 2025-01-28:
I am not using platformio, but Arduino v2.3.4
And yes, your fix is working fine, both strips are declared and running smoothly, on all my C3 devices.  
And yes, I believe guru must start his meditation on issue 1846... sorry for that

zackees, 2025-01-28:
So with the stock configuration (multiplier = 2) what are the maximum number of strips you can run? 2?

Jf7Dee, 2025-01-28:
With the fix you provided, applied to 3.9.12, i can run 2, but it's the max.  If I add a 3th one, the reboot is triggered. 
Stock 3.9.12 is  "broken" as well as Stock 3.9.13 : only one strip can work.
reverting to 3.9.11 : 2 strips are fine, and 3 strips are on reboot from hell sequence  ;)
So yes, on 3.9.11, maximum are 2 strips

zackees, 2025-01-29:
What happens if you halve the memory (set multiplier to 1, set the base memory to half of what it is now). Does that mean all 4 rmt channels now works?

zackees, 2025-01-29:
If you want another strip, a good temporary work around is to instantiate the new SPI WS2812 driver that we now provide and pass it in manually through the non template FastLED.add(...) function.

I'm pretty much at my wits end for the RMT5 driver for Esp32C3. It's very buggy. And we can't go back to RMT4 because espressif forces a boot crash if RMT4 and RMT5 are merely **linked** into the same binary. ESP32C3 just isn't a good board for driving LEDs. I recommend you switch to the ESP32-S3. It will give you 16 channels using it's I2S driver alone.

However, I'll use this bug report and forward it to Espressif. They maintain the led_strip lib that we depend on.

You might be able to do 16 (i2s) + 4 (rmt) + 1-2(spi)

Jf7Dee, 2025-01-30:
If only I can remember which file to change, I'll do it glady (or try to figure out what to change haha).
And how do you downgrade to 4.4 ??

New SPI WS2812 ? yoiu mean via `#define FASTLED_ESP32_USE_CLOCKLESS_SPI `

As for using ESP32S3, I do agree with you. I intended to use the C3s for small tasks, and my S3s for more demanding stuff.
I guess I might have to adapt ;)
Thnaks for it all, appreciated.  I

zackees, 2025-01-30:
You have two choices. If you are using Arduino IDE then set the arduino core to anything before 3.0. This advice comes from others and i have *not* done it myself.

If you are using platformio then you will be able to do a build define via the `build_flags` argument. In this case you don't need to downgrade anything as the following build define switch will work with IDF 5.1+

In this case, you will set it like this:

`-DFASTLED_RMT5=0`

zackees, 2025-02-03:
I'm filing a bug with espressif. Can I confirm that you saw this problem without explicitly instantiating an RMT device yourself? Can you confirm that there isn't a built in WS2812 status pixel which could secretly be allocating a RMT device inplicitely?

I'll share with you the link to the bug report that I'm filing when it's done.

Jf7Dee, 2025-02-03:
Hello sir.  Thanks for the linbk, and your  perseverance!
I can assure you it's not the case. I ran the tests with 4 different WS2812 strips, including one built in the board itself ( the Lolin ESP32 C3 pico) .
I also tested it with different combinations of those strips.
Adding to it, even if no strips are attached, the problem persists.: it is really while calling the 2nd "fastLED.AddLeds" attachement.

Included is a test sketch.  If only first AddLeds is there(and thus the 2nd is commented), no problem.  If 2nd is present, the C3 will reboot.

[ESP32_Lolin_TwoStrip_BugReport.ino.txt](https://github.com/user-attachments/files/18637069/ESP32_Lolin_TwoStrip_BugReport.ino.txt)

Jf7Dee, 2025-02-03:
I forgot to mention that I encountered the same behaviour with the SeeedStudio XIAO ESP32-C3, which does NOT have an onboard WS2812, nor a regular led.
Both Lolin and XIAO chips would reboot while having 2 strips added.

zackees, 2025-02-03:
Can you reply to the espressif link?

Diex, 2025-03-20:
Hi guys. I can confirm that using Arduino's 2.0.18 ESP32 boards versions it works as expected (as seen using other boards, NODEMCU, WEMOS, etc)
I successfully added 8 strips (using io 5-10, 20 and 21) and at least 300 leds per stripe.

something like this:

```
#include <FastLED.h>
#include <limits.h>
#include <algorithm>

#define NUM_STRIPS 8
#define NUM_LEDS_PER_STRIP 300

CRGB leds[NUM_STRIPS][NUM_LEDS_PER_STRIP];

void setup() {
  Serial.begin(115200);

  FastLED.addLeds<NEOPIXEL, 5>(leds[0], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 6>(leds[1], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 7>(leds[2], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 8>(leds[3], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 9>(leds[4], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 10>(leds[5], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 20>(leds[6], NUM_LEDS_PER_STRIP);
  FastLED.addLeds<NEOPIXEL, 21>(leds[7], NUM_LEDS_PER_STRIP);

  FastLED.setBrightness(127);
}

void loop() {

// blabla... your led data here...

  FastLED.show();

  delay(33);
}
```

Jf7Dee, 2025-03-21:
Thanks Diex, it's appreciated taht you took some time to look it up.
I just would like to metion that the problem occurs ONLY with the ESP32 C3, using FastLED 3.9.12 and above.
Arduino version is 2.3.4, even if it does not matter much, here.
According to what we have seen on a response from Espressif ( see https://github.com/espressif/idf-extra-components/issues/473#issuecomment-2724544075), it is related to the structure of the ESP32-C3

zackees, 2025-03-21:
...it only happens with 3.9.-2 and above?

JFClaeys, 2025-03-21:
Indeed, Zackees.  It's what I wrote in the initial bug report : 
> Rolling back to Fastled 3.9.11 "fixed" the issue

I systematically tested on every release afterward, still no fix (didn't expected one soon, by following this thread)

Griiis, 2025-04-07:
Good day! Run in this issue today. Any news on fixes or workaround? ESP32_S3R2 Board... Undable to connect more then 4 led strips

zackees, 2025-04-07:
No work around yet. 3.9.0 uses RMT4 - that's why it works.

A deep refactor needs to happen with RMT5 to support a worker pool again.

zackees, 2025-05-16:
Are you aware of the I2S for S3?

This is our bulk driver. It can drive a lot of pins.

4wheeljive, 2025-05-16:
@zackees Thanks so much for the suggestion. Yes, I have looked briefly into the I2S for S3 driver, but I'm not sure it's a good solution for me right now. Reading through the Esp32S3I2SDemo file (https://github.com/FastLED/FastLED/blob/master/examples/Esp32S3I2SDemo/Esp32S3I2SDemo.ino), several things jumped out at me:
- It requires using PlatformIO to build. I'm a real noob to pretty much everything here (and I mean everything: GitHub, Arduino, coding, electronics, etc.). I installed PlatformIO today, and I'm sure I can figure out how to use it, but I wasn't sure how much time I wanted to spend going down that path right now, especially given some other things, such as:
- It appears that for the I2S route you MUST use exactly 16 pins. However:
  - My current project involves twelve 8x32 tiles; so I'd have to completely redesign the layout and purchase all new strips, and
  - I've been using XIAO controllers, which don't have that many pins; so I'd need to purchase some additional new hardware.

On their own, none of these issues are huge or insurmountable. But taken together, my inclination for now is to look/hold out for another solution.

But thanks once again for responding to my post. (Among the many things I had never even heard of just forty-eight hours ago are: RMT, rolling back FastLED versions, PlatformIO, or Zachary Vorhies. Since then, I've discovered what a rock star you are in the FastLED world, and I was excited to realize what you rolled out with 3.9.16 and are working on for 3.9.17. I don't mean to get all fanboy here; I just really appreciate that you took the time to offer a suggestion on a minor issue for a novice like me!)

zackees, 2025-05-16:
Ah thanks! You are too nice!

A lot of people are hesitant to try platformio. But then I insist they try it and then five minutes later they have everything working.

It's really that simple.

I've created a guide to gently help you: https://github.com/FastLED/PlatformIO-Starter

About the 16 pins: it turns out you don't need all of them. You can have as few pins active as you want

4wheeljive, 2025-05-16:
Oh, that's huge! If I don't need to use 16 pins (and therefore don't have to start all over with a new controller and LEDs), then it will definitely be worth diving into platformio (which it sounds like I should be doing anyway!) to give I2S a try. Thanks again!

4wheeljive, 2025-05-17:
Okay...I did it. I got an S3 going in 3.9.16 with I2S driving 6 pins controlling 512 LEDs each. (And the pins were non-contiguous and somewhat arbitrary: 2,3,4,9,8,7.)

I used the Esp32S3I2SDemo as my starting point: 
https://github.com/FastLED/FastLED/blob/master/examples/Esp32S3I2SDemo/Esp32S3I2SDemo.ino

I learned enough about platformio to make this happen, and I have two thoughts I'd like to share on this:

> First, the transition from Arduino IDE was a lot trickier (at least for me) than some sources might suggest. With all due respect to zackees, I suspect that the folks who are able to get everything working in just 5 minutes must have a much stronger foundation in computer science and coding than I do. It took me several hours of trial and error to figure out some basic differences in how platformio compiles compared to Arduino IDE. For example, it took me at least 15 minutes just to realize why the .ino sketches I copied into the platformio project /src folders were not functional. (I sort of knew that .ino scripts aren't true C++, but I didn't know the implications of this. It took me some time to realize that the .ino code needed to be moved to the main.cpp file.) It alto took me chunks of time to figure out numerous other differences such as:
> - How to include/reference external header files
> - Proper/valid C++ structure and syntax for various things that Arduino is a little more flexible with (Again, for someone with pretty much any background in C++, this would not be an issue. But for someone like whose first real coding began in the Arduino IDE, there's a bit more of a learning curve.)
> 
> Second, despite the numerous things I'm needing to learn or relearn with platformio, I'm really glad I've begun the transition. I'm really appreciating a lot of the additional features and functionality, speedier processing, etc. This investment will no doubt pay big dividends over time.
> 
> I have numerous follow-up comments and questions that I'll save for a more appropriate forum/thread. But I though I'd share at least the above in case someone else who reads this thread trying to solve their multi-strip issues is also a relative novice like myself.

I do have at least one question, though, that relates directly to the I2S solution to the multi-pin issue. The Esp32S3I2SDemo referenced above indicates that the platformio.ini file should include the following build flag:

`platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip`
    
It looked to me like that particular release (51.03.04) might be a little old; and I wasn't (and am still not) sure whether to use that or something more recent. I initially tried various platform-espressif32 releases that were more recent, but couldn't get things working until I used 51.03.04 (although that could have been due to other transition factors I had not yet figured out). But now **I'm worried that if I'm locked into an older release just to utilize I2S, I won't have access to all the good stuff included in other forks/releases/whatever. I'd love any guidance anyone could provide on this.**

Thanks,
Jeff

zackees, 2025-05-18:
About the pinned version of platform.zip

Esp has been naughty and just breaks everyone with a sloppy update.

For the example... RMT! Which they just forced on everyone in arduino. They also just broke I2S with the last release. Because Arduino doesn't use pinned deps a lot of n00bs are now broken and don't know how to fix it because they can't or it's hard.

By using a pinned dep, this becomes not your problem. If the product builds today it builds tomorrow.

You are in a much better position now.

zackees, 2025-05-18:
And don't worry about missing out on new features for esp idf. This isn't a windows update.

You need to use whatever esp-idf your libraries depend on working. I think the most complicated thing here is I2S.

If in the future you want to update then you have the absolutely freedom to do this and find out if a new version breaks you

4wheeljive, 2025-05-19:
Thanks zackees for your thoughts on the benefits of using a pinned version of platform.zip. That's a helpful perspective to keep in mind. 

As far as "missing out" on new features, I wasn't really thinking about stuff like a splashy new OS rollout. I was thinking specifically about being able to use what I thought might be relatively new elements of FastLED.

In my case, I got I2S working for a sketch that involves my somewhat modified versions of (pre-fx/ScreenMap) Pride2015 and ColorWavesWithPalettes. I then tried to test out one the newer examples: https://github.com/FastLED/FastLED/blob/master/examples/Fire2023/Fire2023.ino. However, I'm getting errors along the lines of the following:

>    'vector' is not a member of 'fl'
>    namespace "fl" has no member "vector"
>    'vec2f' was not declared in this scope
>    identifier "vec2f" is undefined
>    'lut' was not declared in this scope
>    identifier "lut" is undefined

After various unsuccessful troubleshooting efforts, I started wondering if using the older pinned platform.zip meant I couldn't use some of this newer FastLED stuff.  After further fumbling, I noticed the following:

There is no vector.cpp.d or vector.cpp.o file in the following build path:
\Projects\Fire_I2S_64x48\.pio\build\seeed_xiao_esp32s3\lib661\FastLED\fl\

There is a vector.h file in the following source path: \Projects\Fire_I2S_64x48\.pio\libdeps\seeed_xiao_esp32s3\FastLED\src\fl\vector.h
However, there is no corresponding vector.cpp file in that src\fl\ path.

I'm in way over my head here, but this seems like it could be significant. I looked in several "normal"/current FastLED repos, and I couldn't find a vector.cpp anywhere (although I do see a test_vector.cpp). I notice that many .h files have a corresponding .cpp file, and many don't. In the Arduino IDE, "vector" is successfully added to the fl namespace with just the .h file. But it appears that this does not work when compiling in platformio.

The more I've flailed around trying various ways force "vector" into the fl namespace in the platformio build, the more I've started to wonder if there's something special about that particular function that makes it difficult to use outside the Arduino IDE. I've noticed that the vector fl function doesn't seem to appear in many (if any) of the other FastLED examples, and I wonder if I just picked an unfortunate sketch to try as my first test of some "newer" sample code.

4wheeljive, 2025-05-20:
BTW, I don’t mean to hijack this thread with off-topic matters, and I realized after posting the message above that, at least on the surface, it had little to do with the OP. I’ll be happy to migrate my comments/questions elsewhere. That said:
- The underlying issue here involves alternatives to RMT for driving multiple strips with multiple pins
- One of the options, I2S, generally requires using platformio (or another non-Arduino IDE)
- If certain of the published FastLED example sketches won’t work with platformio, or if special steps need to be taken to get them to work, then that’s something that would be good for people (at least neophytes like me) to be aware of

zackees, 2025-05-20:
> In my case, I got I2S working for a sketch that involves my somewhat modified versions of (pre-fx/ScreenMap) Pride2015 and ColorWavesWithPalettes. I then tried to test out one the newer examples:

The fresh example you just grabbed is the master copy and references new code, in your case the fl::vector class, which is a drop replacement for the standard std::vector class that's new for the next version of FastLED.

Fortunately, there's an easy fix because your are on platformio. Have your fastled dependency use the github master version.

[platformio]
default_envs = my_env

[env:my_env]
platform = espressif32
board = esp32dev
framework = arduino

lib_deps =
    https://github.com/FastLED/

4wheeljive, 2025-05-20:
Thank you @zackees! I REALLY appreciate your responsiveness in helping me work through all of this. (I hope this thread will be useful to others, which is why I've been elaborating on my progress/process a bit.)

Your suggestion worked...eventually. (The term "easy fix" is always relative!)

The first thing I needed to work out was how best to access the fastled master version. When I first tried the direct link you posted, platformio prompted me to download a Git client (which I found a bit intimidating, as the install process required answering about a dozen config questions for things I know nothing about). After installing that, it took me a few minutes to figure out that I needed to use the Source Control panel in VSCode to make use of it. That was intimidating too, as it seemed to involve creating and uploading to a new repository, and it wasn't clear to me why this should be necessary or what the implications of doing so might be. (Since what I was trying to access was the master fastled repository, I was afraid that if I proceeded I might create some sort of "two-way entanglement" with the repository, instead of just a one-way pull of the code into my sketch.( So **_I ended up just downloading a zip of the fastled master and having the lib_deps point to that_**.

When I finally got the sketch running, I enjoyed a nice LED display of fire in lovely shades of...green. I realized right away that I probably just didn't have the correct GRB color order set for my WS2812 panels, but figuring out how to do this with the Esp32S3I2SDemo code proved a bit challenging. That code provides an option to set up the LED strips using either "FastLED bindings" or the "raw driver." It was obvious how to set the GRB order if were using the FastLED API:

`FastLED.addLeds<WS2812, DATA_PIN_1, GRB>(leds + (0 * NUM_LEDS_PER_STRIP), NUM_LEDS_PER_STRIP)      `

However, I have not yet been able to get things working using the FastLED setup. So I've been using the raw driver, and it took quite a bit of code searching and trial and error before I figuring out how and where to set the color order for that. Specifically, **I changed `#define COLOR_ORDER_RGB` to `#define COLOR_ORDER_GRB` in the driver.h file located in the following path:**

**\Projects\Fire_I2S_64x48\.pio\libdeps\seeed_xiao_esp32s3\FastLED\src\third_party\yves\I2SClockLessLedDriveresp32s3\driver.h**

One final comment for anyone else who reads this thread trying to figure out how to get going with I2S. In the post immediately above showing how to use the github master version for the fastled dependency, there were some additional lines shown for the platformio.ini file:

> [env:my_env]
> platform = espressif32
> board = esp32dev
> framework = arduino

I wasn't sure if those settings would work for my I2S use case, so I left those as I had them:

> [env:seeed_xiao_esp32s3]
> platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
> board = seeed_xiao_esp32s3
> framework = arduino    

Thanks again zackees for all your help!

4wheeljive, 2025-05-20:
After all of my efforts to get the fl::vector function working so I could compile the Fire2023 sketch in platformio, I just realized something. Vector is called only as part of the screenMap function, which is apparently used "for FastLED Web to map the 1D strip to a 2D screen" for display in a browser. So, if I'm not mistaken, vector and screenMap may be complete non-issues in the context of using I2S to drive multiple LED strips. Is that correct?

zackees, 2025-05-22:
Correct

zackees, 2025-11-06:
try it now, it might be fixed. I brought back RMT worker pools

closed 2026-04-19 by zackees