FastLED/FastLED #1875: ESP32 Dev - I2S - allow multiple includes of FastLED without symbol collision

zackees, 2025-01-27:
Reported by https://github.com/RococoN8R

Split off from issue:

https://github.com/FastLED/FastLED/pull/1866

> Yes, I'm using FastLED.h in two .cpp files.I have a few years of custom effects and sequences built around the legacy architecture, and while I'm interested in adopting your 4.0 vision, I can't begin that exploration for 6 or more months. Unfortunately I can't tell you how many other developers will encounter similar constraints. Let me know your plans on the cpp file so I can determine how to best proceed.Sent from my phone... please excuse the topys.On Jan 25, 2025 12:06 PM, Zachary Vorhies ***@***.***> wrote:
I don't think that's possible, but I may go back to doing a CPP file for the ESP i2s stuff.
You are trying to include FastLED in multiple cpp files yes?
You should know that if you are trying to hide implementation that I've put a lot of work into making this easy. While one cpp file will need to include FastLED.h, the rest should only have to rely on crgb.h in order to get the CRGB struct/class so you can do your pixel writing.
The convention I use is that gfx.cpp in my sketch includes the FastLED.h header, everywhere else includes crgb.h because typically pixel painting is done in multiple visualizer cpp's.
Before the 3.9.X series, crgb.h included FastLED.h. Now it does not and can be included anywhere you want.
Good luck.

closed 2025-01-27 by zackees

RococoN8R, 2025-01-27:
That was quick, thanks! 
I confirmed it compiles and links properly. 
Will load and test on device tonight.

RococoN8R, 2025-01-28:
Hey @zackees - I used the FASTLED_ESP32_I2S build flag with 5 strings/pins of 100 LEDs and had a lot of noise. When I only run 3 strings, it is better. I wanted to test increasing the DMA buffers using FASTLED_ESP32_I2S_NUM_DMA_BUFFERS=5, but there's a code issue. Specifically, clockless_i2s_esp32.h is needs access to gCntBuffer which is defined in i2s.cpp.

zackees, 2025-01-28:
Can you fix it and paste in a patch?

RococoN8R, 2025-01-28:
Agreed, but in full transparency, I'm a hobbyist with minimal experience publishing changes in github. But I'm hoping you can confirm a couple of things so I "learn to fish". First, I should fork https://github.com/fastled/FastLED.git#6fdf262, correct? The two changes are minor, as you said. How do I issue the fix when done?

Also, I'm having a medical procedure done this afternoon, so I may not get this done until tomorrow night. But I'll take care of it!

zackees, 2025-01-29:
ah ok, this is new to you. yeah let me fix

zackees, 2025-01-29:
Should be fixed now.

RococoN8R, 2025-01-31:
Apologies for the delayed response. I tested it. Seems ok with 2-3 strips, but at 5 strips there's more noise than with RMT. Will explore if other networking protocols are interfering this weekend.

zackees, 2025-01-31:
Jeez okay.

Thanks for the feedback.

zackees, 2025-01-31:
We have a three different WS2812 controllers for esp.

If I do the work and make it easy to instantiate all three types so it's dead simple, will you commit to testing it?

You will be able to instantiate I2S, SPI and RMT in different controllers with an optional enum to FastLED.add(...). Right now it's super hard in the api and you have to do a bunch of low level boiler plate to get it to work.

If you agree to simply be the first beneficiary, then I will agree to do the work to help your lights get more stable.

RococoN8R, 2025-01-31:
That would be awesome! I'm happy to take some time testing this and providing feedback.