FastLED/FastLED #1876: Blink example doesn't compile since 3.9.13

Jueff, 2025-01-28:
Board: 1.0.4
FastLED: 3.9.13

compiling the Blink example fails with error: 'cpu_hal_get_cycle_count' was not declared in this scope

```
In file included from C:\Users\user\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\rmt_4\idf4_rmt_impl.h:15:0,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\rmt_4\idf4_rmt.cpp:14:
C:\Users\user\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clock_cycles.h: In function 'uint32_t __cpu_hal_get_cycle_count()':
C:\Users\user\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clock_cycles.h:14:56: error: 'cpu_hal_get_cycle_count' was not declared in this scope
   return static_cast<uint32_t>(cpu_hal_get_cycle_count());
```

Jueff, 2025-01-28:
the board version has the include file <esp32-hal.h>, but cpu_hal_get_cycle_count isn't declared in there.

maybe change line 11 to 

`#elif __has_include(<esp32-hal.h>) && ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(3, 0, 0)
`

then it compiles in my environment.

zackees, 2025-01-29:
Thanks for the patch!

closed 2025-01-29 by zackees